OpenIntelWireless/itlwm #292: [AirPort Itlwm] Handoff and AirDrop not working

RGarrido03, 2020-09-16:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: AirPortItlwm 1.1.0
- WiFi Card Model: Intel AX200
- Product ID: 0x8086, 0x84
- macOS Version: macOS Big Sur 11.0

**Description**
I created an issue since Gitter is just for personal support. 
The AirPort version of itlwm works very nice and connecting the PC to Wi-Fi is really Broadcom-like!
However, Handoff and AirDrop don't work. AirDrop is expected, however, Handoff should be working according to the release notes, and it's not.
When I open a Pages file or a Safari page on my iPhone, there's no Pages / Safari icon (with iPhone badge) in the macOS Dock. The same happens if I open a Safari page from my Mac.
Thank you so much for everything!

**Bug Report Archive**
As there'a no HeliPort, how can I get a log? Maybe through the "Getting Kext Logs For Runtime issues" method on your docs? Please tell me so I can help you 😁 
[image]
[image]

**Upload your Kext**
Just the latest AirPortItlwm from GitHub Releases.

williambj1, 2020-09-17:
1. Make sure you have the same Apple ID signed in on both of your devices
1. Make sure your Apple ID has two-factor authentication enabled
1. Disable VPN on both of your devices
1. Try to re-sign in on both of your devices
1. Try universal clipboard first, it may take a while after boot for service initialization.

tommyvct, 2020-09-17:
> 1. Make sure you have the same Apple ID signed in on both of your devices
> 2. Make sure your Apple ID has two-factor authentication enabled
> 3. Disable VPN on both of your devices
> 4. Make sure Bluetooth is able to connect to iPhone, iPad, etc
> 5. Try to re-sign in on both of your devices
> 6. Try universal clipboard first, it may take a while after boot for service initialization.

Tried, and nothing worked as I expected.

I don't know is this expected or not, the AirDrop channel indicated in the System Information is 0:
```
  Software Versions:
  CoreWLAN:	13.0 ((null))
  CoreWLANKit:	13.0 ((null))
  Menu Extra:	13.0 ((null))
  System Information:	12.0 (1500)
  IO80211 Family:	12.0 (1200.12.2b1)
  Diagnostics:	1.0 (910)
  AirPort Utility:	Not Installed
  Interfaces:
en1:
  Card Type:	AirPort Extreme  (0x8086, 0x2110)
  Firmware Version:	itlwm: 1.1.0 fw: 29.1654887522.0
  MAC Address:	88:b1:11:22:53:26
  Locale:	FCC
  Country Code:	ZZ
  Supported PHY Modes:	802.11 a/b/g/n/ac
  Supported Channels:	1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 36, 40, 44, 48, 52, 56, 60, 64, 100, 104, 108, 112, 116, 120, 124, 128, 132, 136, 140, 144, 149, 153, 157, 161, 165
  Wake On Wireless:	Supported
  AirDrop:	Supported
  AirDrop Channel:	0
  Status:	Network Service Inactive
```

williambj1, 2020-09-17:
Airdrop is not expected to work, still needs further investigation.

tommyvct, 2020-09-17:
I'm actually amazed how you guys reversed `Apple80211` and made the whole thing possible.

seungjunProgramming, 2020-09-17:
Wifi Sharing(Hotspot) Doesn't work...

williambj1, 2020-09-17:
> Wifi Sharing(Hotspot) Doesn't work...

Known issue

lgariv, 2020-09-17:
continue from [here](https://github.com/OpenIntelWireless/itlwm/issues/290#issuecomment-694182802).

Tried all of the steps again, waited for almost an hour, still nothing seems to work.

RGarrido03, 2020-09-17:
I did nothing and it started working 😂 it's very good! AirDrop now recognizes my iPhone but won't send the file, which is normal due to the development of the kext. Thank you!!

williambj1, 2020-09-17:
@lgariv What’s your card model, system version, and other info?

Have you tried resetting network interface configuration?

Many people have reported that they got handoff working so it’s likely to be an user related issue instead of the code.

enPause96, 2020-09-17:
tried all these things and i still dont get Handoff and Shared Clipboard to work
im on Catalina 10.15.4 and Opencore 0.6.1
my wifi card is Intel wireless N 7260

Zackary-Lykos, 2020-09-18:
Also unable to use handoff and universal clipboards.
----
Already checked #291, order is right.
Already tried  https://github.com/OpenIntelWireless/itlwm/issues/292#issuecomment-693728787
Now my iPad can find PC through Airdrop(unable to send file) , and PC can find iPad through sidebar(unable to connect due to time out)
----
Opencore 0.6.1
AirPortItlwm 1.1.0
Catalina 10.15.6
Intel® Dual Band Wireless-AC 8275  (0x8086, 0x3E02)

lgariv, 2020-09-18:
> @lgariv What’s your card model, system version, and other info?
> 
> 
> 
> Have you tried resetting network interface configuration?
> 
> 
> 
> Many people have reported that they got handoff working so it’s likely to be an user related issue instead of the code.

Catalina 10.15.6
OpenCore 0.6.1
Intel 9462

Will try to reset network configuration and report the result.

lgariv, 2020-09-19:
> Catalina 10.15.6
> OpenCore 0.6.1
> Intel 9462
>
> Will try to reset network configuration and report the result.

tried to reset by removing "location" in network preferences, and by deleting configs from SystemConfiguration and reboot. 
Still not working.
Compiled from source latest version of AirportItlwm, tried everything previously suggested, still not working.

keke-cute, 2020-09-19:
universal clipboard worked once, airdrop show my iPhone but can't send file, safari handoff work, sidecar show my iPad but not work

eric2877, 2020-09-29:
deleting configs from SystemConfiguration and reboot re_login every device ， still not work， reisntall system still not work， 
Catalina 10.15.7/6
opencore 0.6.1
intel 8265

phu54321, 2020-10-05:
I'm checking with [Stats](https://github.com/exelban/stats). (FOSS iStats Menu alternative)

eric2877, 2020-10-09:
After open the two steps auth, it works，but can only send data to iPad , can't receive , then I re-open the handoff in preferences, after all, it works well.
 by the way , I have  changed my SN code, and delete the old iMac in iCloud count.
Thank you guys!!!

phucnguyen2411, 2020-10-10:
> i'm using the latest version 1.2.0 pre-release for big sur and handoff/ Airdrop not working for AX200.

after logout and trying to login again, airdrop is able to scan the device but can't send. Handoff is not working.

eric2877, 2020-10-11:
You can re-open your handoff in preference, then , it will work, good luck

phucnguyen2411, 2020-10-12:
> You can re-open your handoff in preference, then , it will work, good luck
> […](#)

yes, handoff is working but airdrop is not working then. :(

eric2877, 2020-10-13:
> … <x-msg://1/#>

In deed, AirDrop is still in developing，It never worked!

phucnguyen2411, 2020-10-13:
> In deed, AirDrop is still in developing，It never worked!
> […](#)

thanks, i'm looking that code, might be can contribute with you guys. Thanks

RGarrido03, 2020-10-18:
I edited the title so there’s no misunderstanding about Handoff and Clipboard not being available (when it is). Great work devs!

cavazzatommaso, 2020-10-21:
Some news about sidecar?

Theredmc, 2020-11-10:
just wanted to say that airdrop is now working 
[image]

the MacBook is the one with the intel card, I can drop files too

phu54321, 2020-11-10:
Yeah [there had been massive commits referring to AWDLs](https://github.com/OpenIntelWireless/itlwm/commits/master). Will wait until features are stabilied. Thanks for the massive contribution, zxystd!

If there's a patreon or paypal I could pledge to, I'm really willing to contribute.

gtrooper, 2020-11-15:
> just wanted to say that airdrop is now working
> [image]
> 
> the MacBook is the one with the intel card, I can drop files too
> 
> It just doesn't work from my intel card notebook to my and pc, it doesn't show up anything
> 
> Intel card: Intel Corporation Wireless 3165
> Bluetooth Dongle: LogiLink, Cambridge Silicon

I am using an INTEL NUC8 with latest itlwm and IntelBluetooth and still I cannot see AirDrop show up in the finder... How did it appear to you ?

M69W, 2020-11-15:
[image]

[image]

while use [the newest v1.2.0-alpha](https://github.com/OpenIntelWireless/itlwm/releases/tag/v1.2.0-alpha)
I hava same case like yours, when I confirm to shared button of iPhone(the iPhone have connected the net), latop's wifi is still no working.

aryankantal01, 2020-11-18:
I m  running On bigSur with opencore 0.6.3
Now both My handoff and Universal clpbrd are working perfectly-
My pc config-
ram-4gb
i3 6100
intel hd 520 graphics
Hp laptop
-------------
How I got both to work?
I installed airportitlwm and inserted in Kext/Others folder, and deleted itlwm,airportblwmfix was already there in my kext folder, after inserting I opened plist editor and did snapshot and rebooted my pc..after that I could see airdrop in finder but it didn't seem to work as it is still in development stage...for testing handoff..I just reseted my network preferences on Hackintosh,and resigned to iCloud...both devices must be on same wifi +same iCloud +bluetooth onn..and u shld turn on handoff in Mac as well as ipn...after that u both works seamlessly;)
I hope it helps:)

statuspace, 2020-11-20:
> ## I m running On bigSur with opencore 0.6.3
> Now both My handoff and Universal clpbrd are working perfectly-
> My pc config-
> ram-4gb
> i3 6100
> intel hd 520 graphics
> Hp laptop
> How I got both to work?
> I [...]
**excellent** 
Can you share your efi folder ?