OpenIntelWireless/itlwm #294: [AirportItlwm] 无法连接隐藏的WiFi

JunkuiZhang, 2020-09-17:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: AirportItlwm 1.1.0
- WiFi Card Model:  intel ac 8265
- Product ID: 8086
- macOS Version:  Catalina 10.15

**Description**
WiFi隐藏了ssid，且有密码，但无论是2.4g的频道还是5g的频道都无法连接上。连接WiFi时提示“找不到该网络”。
WiFi取消隐藏后可以连上，所以我猜这可能是一个bug。

**Bug Report Archive**
由于使用的是AirportItlwm，因此没有安装HeliPort，请问怎么抓取AirportItlwm的log？

**Upload your Kext**
使用的是[最新release](https://github.com/OpenIntelWireless/itlwm/releases/tag/v1.1.0)的版本

williambj1, 2020-09-17:
HeliPort + itlwm 是否正常？

JunkuiZhang, 2020-09-17:
> HeliPort + itlwm 是否正常？

使用Heliport+itelwm时是正常的，可以连接隐藏Wi-Fi。
使用AirportItlwm时，出现下图的提示，不能正常使用。
[image]

johnnync13, 2020-09-17:
this guy has native control menu? https://github.com/usr-sse2/Black80211-Catalina

zxystd, 2020-09-18:
Confirm it is a bug.

hezeniu, 2021-01-07:
这个不能连接隐藏，在文档中说了。