OpenIntelWireless/itlwm #297: Shutdown during sleep

phucnguyen2411, 2020-09-18:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version:  Airportltlwmv1.0_Beta_big_sur
- WiFi Card Model: Ax200
- Product ID: 0x2723
- macOS Version: 

**Description**
After sleep around 1800sec, the system getting issue with panic USB, Bluetooth and shutdown

**Bug Report Archive**
Not installed Heli in this case

**Upload your Kext**
[AirportItlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/5242784/AirportItlwm.kext.zip)
[itlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/5242786/itlwm.kext.zip)

williambj1, 2020-09-18:
Please attach the panic log

phucnguyen2411, 2020-09-19:
just facing one time, will try to export the log in the next time.

williambj1, 2020-09-26:
Will reopen once the log is provided.

closed 2020-09-26 by williambj1