OpenIntelWireless/itlwm #301: Intel AC 7265 - kext doesn't load after itlwm 1.0.0

maxhartung, 2020-09-19:
**Have You Read Our Docs**
YES

**Are You Reporting A Bug**
YES

**Environment**

- Kext Version: 1.1.0
- WiFi Card Model: Intel AC 7265 (0x095A)
- Product ID: 0x095A
- macOS Version: 10.15.6

**Description**

I tried using itlwm 1.1.0 both stable and beta but the kext doesn't load at all on my wi-fi card. OpenCore is setup properly. The only kext loading is the first release of itlwm 1.0.0

**Upload your Kext**

[itlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/5249349/itlwm.kext.zip) - 1.1.0

williambj1, 2020-09-19:
Logs please

alessiomatricardi, 2020-09-19:
I've the same card (same Device id 0x095A) and everything work properly with AirportItlwm.
- Check your EFI (from this version you must change name and path of the kext in config.plist -> Kernel -> Add -> Item X
- Check that AirportItlwm is loaded before Bluetooth and Injector
- Check that you haven't other Wifi kexts

Hope you will solve!! 💪🏼

maxhartung, 2020-09-19:
> Logs please

What kind of logs you need ? OpenCore debug build logs are okay or you need anything else ?

Thanks,
Max

maxhartung, 2020-09-19:
> I've the same card (same Device id 0x095A) and everything work properly with AirportItlwm.
>
> * Check your EFI (from this version you must change name and path of the kext in config.plist -> Kernel -> [...]

I checked and everything looks fine to me

zxystd, 2020-09-19:
i don't think it is a driver issue, personal hackintosh related help would not be supportef here. you should fix it by yourself.

closed 2020-09-19 by zxystd

maxhartung, 2020-09-19:
> i don't think it is a driver issue, personal hackintosh related help would not be supported here. you should fix it by yourself.

85:166 00:013 OCAK: Dependency com.apple.iokit.IO80211Family was not found for kext com.zxystd.AirportItlwm
85:214 00:048 OCAK: Vtable patching failed for kext com.zxystd.AirportItlwm
85:219 00:005 OC: Prelinked injection AirportItlwm.kext () - Invalid Parameter

I think it's something

williambj1, 2020-09-19:
Enable Apple Secure Boot