OpenIntelWireless/itlwm #302: [Airpotitlwm]

PhantomR, 2020-09-19:
**Have You Read Our Docs**
A small part to be honest, but I am sure this is a problem.

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 1.1.0 stable, Big Sur version
- WiFi Card Model: Intel AC 9560
- Product ID: 8086:a370:8086:0034
- macOS Version: 11.0 Public Beta (20A5364e)

**Description**
Network: a mobile hotspot from a Galaxy S8 phone (sharing the phone's mobile data)
Initially it showed up in the available networks list no problem, I connected and it was working well. It disconnected after restart (I had not checked the automatic join option), so I checked the automatic join and connected. I am not sure it disconnected by itself or it was me joining a normal (non-hotspot) network, but after it disconnected it would no longer show up in the available networks. I tried restarting the hotspot to no success. However, connecting via manual name ("Other" in the Wireless menu) worked like a charm.
[AirportItlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/5250013/AirportItlwm.kext.zip)

**Bug Report Archive**
Can't do this since I am using AIrportitlwm.. Heliport does not work, says itlwm is not running.

**Upload your Kext**
[AirportItlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/5250015/AirportItlwm.kext.zip)

williambj1, 2020-09-19:
Please follow those instructions, replace `Black80211` with `AirportItlwm`, obtain the logs, and compress them. Thanks.

https://github.com/usr-sse2/Black80211-Catalina/issues/42#issuecomment-695041876

PhantomR, 2020-09-19:
Not sure I did it right, but I ran the commands in succession and also re-ran the first two again at the end. There definitely is a problem.. after I disconnected (to try and connect while third command is running), it took a few tries to get it running again (between which I turned the hotspot on/off and at times stopped the command & ran it again and also tried connecting to other networks so I could try and connect to the hotspot again. Hotspot network AndroidAPC950).

williambj1, 2020-09-20:
Could you specify how did you manage to disconnect? It's not supposed to work by clicking the menu right now.

PhantomR, 2020-09-20:
I turned Wi-fi off. Other than that, I swapped to another connection in the list.

williambj1, 2020-10-16:
Please reopen if this still exists using the latest alpha.

closed 2020-10-16 by williambj1