OpenIntelWireless/itlwm #303: [Request] Possible to get Killer 1435 Wireless-AC working?

dimitrijmalcev, 2020-09-20:
Hello zxystd,

would it be possible to get Intel® Chip Based WiFi Chip Killer™ Wireless-AC 1435 working?
Many modern systems desktops, laptops have this chip, because of good performance.

The Killer Wifi is based on modules:
	Intel® Wireless-AC 9560
	Intel® Wireless-AC 9260

detailed information:
VEN_168C&DEV_003E

Sources:
https://www.killernetworking.com/products/killer-wireless-ac-1435/#1538447498979-c399b5f7-a928
https://www.intel.com/content/www/us/en/support/articles/000026140/network-and-i-o/wireless.html

If your project makes it possible which help/information will you need from me, to get this issue rolling?
I am extremely impressed by the ton of work done so far.

williambj1, 2020-09-20:
Vendor `0x168C` is Qualcomm Atheros, it has nothing to do with Intel.

closed 2020-09-20 by williambj1

dimitrijmalcev, 2020-09-20:
Thank you for the information. Last question: Do you know about a possible project to port on mac regarding Killer Wireless-AC 1435  from BSD?

williambj1, 2020-09-20:
I don't think there are existing projects that port code to support Ath cards, you can start a project yourself based on FreeBSD or OpenBSD and use our existing 80211 stack ported from OpenBSD.