OpenIntelWireless/itlwm #305: AirportItlwm not Loading

ImSaaam, 2020-09-20:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 
- WiFi Card Model: AC 8265
- Product ID: 
- macOS Version: 

**Description**
AirportItlwm delay the boot for around 20s and then it doesn't load at all, Black80211 works fine. Logs are blank and real time airport logs just errors out with "Unable to retrieve AirPort interface list"

**Kext Download Source**
1.1.0 stable release

alessiomatricardi, 2020-09-20:
I'm sure it is a user mistake.
- Check if SecureBootModel is set to "Default" (or a valid value according your SMBIOS) in Misc->Security.
- Delete/disable all others wifi kexts such as Black82011 and other versions of itlwm.

ImSaaam, 2020-09-20:
> I'm sure it is a user mistake.
> 
> * Check if SecureBootModel is set to "Default" (or a valid value according your SMBIOS) in Misc->Security.
> * Delete/disable all others wifi kexts such as Black82011 and other versions of itlwm.
> 
> If the problem persists, upload logs so kext's moderators/owner can help you!!

SecureBootModel is set to Default, I’ve used both airportitlwm and itlwm 1.1.0.
I can’t upload logs because they’re blank/kext hasn’t loaded.

ErrorErrorError, 2020-09-20:
Make sure you're using the AirportItlwm.kext Catalina and using OpenCore 0.6.1 or higher. And if you're using itlwm.kext you must disable it and any other related kext if you're using AirportItlwm.kext.

Also, you can get logs if you follow this comment. https://github.com/usr-sse2/Black80211-Catalina/issues/42#issuecomment-695041876

ImSaaam, 2020-09-20:
> And if you're using itlwm.kext you must disable it and any other related kext if you're using AirportItlwm.kext.
> 
well this solved it, I was using both airport and itlwm when I should've used the latter only.
thank you for pointing it out.

closed 2020-09-20 by ImSaaam

ethans65, 2025-02-15:
i'm having the same issue where it won't load but only using one at a time, i'm wondering now if it's because i'm using a couple kexts to enable ethernet on my hackintosh. would these being there matter?