OpenIntelWireless/itlwm #306: Adopting MacKernelSDK

vit9696, 2020-09-21:
Hello, we made a new product designed to help all the kext developers to easily target different macOS versions — [MacKernelSDK](https://github.com/acidanthera/MacKernelSDK). This is a modified Kernel.framework SDK, which contains private headers, removed deprecations, availability checks in many places, and several other fixes.

In Acidanthera we migrated all our kexts to use MacKernelSDK. At this point we believe we can offer the new SDK to let itlwm compile much easier in any Xcode version. In addition to that we would like to include IO80211Family headers in MacKernelSDK.

zxystd, 2020-09-21:
ok, I agree it is a good way, I will comeback tomorrow from trip and will replace the project build SDK to MacKernelSDK soon. also, IO80211Family reverse headers is not copyright, everyone can use it for any purpose. thanks to Acidanthera.

vit9696, 2020-09-21:
Well, with IO80211Family it needs some effort to merge. Currently the headers in https://github.com/OpenIntelWireless/itlwm/tree/airport_impl and https://github.com/usr-sse2/Black80211-Catalina/tree/master/Black80211/apple80211 have their own missing pieces, and I am not too happy that we have entirely separate directories for each operating system. We will need to:

- Decide whether we merge them into one directory and guard with availability macros or leave them in separate directories. I would root for the former.
- Decide who handles the merge. Since that would require updating itlwm we will at least want you to check this if not to handle yourself.

zxystd, 2020-09-21:
Yeah, do you have any idea or plan?

vit9696, 2020-09-21:
I think we should include `<Availability.h>`in the headers and then use `#if __MAC_OS_X_VERSION_MIN_REQUIRED >= __MAC_10_14` and similar for each supported version. Once this is done and the headers are merged I could just grab the updated headers from `airport_impl` and commit them to MacKernelSDK.

zxystd, 2020-09-21:
got it, will do it soon.

zxystd, 2020-09-21:
can I merge Black80211's headers? or  @usr-sse2 will do it? and if you put them into the SDK, how can I add the new reversed things?

vit9696, 2020-09-21:
I could put Black80211 headers to MacKernelSDK right away, but they only contain 10.15 and 11.0. Perhaps it will be best if you update them later with 10.14 and new reversed things?

zxystd, 2020-09-21:
yeah, that would be more fine.

vit9696, 2020-09-21:
Pushed. In addition to Mojave it will be great to have [High Sierra and Sierra](https://github.com/rpeshkov/black80211/tree/master/Black80211/apple80211) imported as well eventually.

zxystd, 2020-10-12:
Now we have adopted to `MacKernelSDK `, thanks for your help.

closed 2020-10-12 by zxystd