OpenIntelWireless/itlwm #308: [Airportitwlm] Scanning shows only my connected network on Big Sur

tikscr, 2020-09-21:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: Airportitwlm beta
- WiFi Card Model: Intel AC-9560
- Product ID: 0x8086, 0x4030
- macOS Version: 11.0 beta 7

**Description**
After installation of Airportitwlm i see wi-fi working, but in list of networks there's only one network
**Bug Report Archive**
[Log_2020-09-21_10-10-28.log](https://github.com/OpenIntelWireless/itlwm/files/5253869/Log_2020-09-21_10-10-28.log)

**Kext Download Source**
Airportitwlm downloaded from 1.1.0 Release page

zxystd, 2020-09-21:
attach is not a full log, can you wait some minutes for the network list display again?

tikscr, 2020-09-21:
> attach is not a full log, can you wait some minutes for the network list display again?

I've waited for a 10 minutes

unhoo28, 2020-09-21:
My card is BCM943602CD, it also shows only connected network on Big Sur, so it looks like normal

williambj1, 2020-10-03:
@tikscr Is this still reproducible with the kext built from master https://github.com/OpenIntelWireless/itlwm/commit/9b60103c3369e950027286b2a0639a067ac93de0?

tikscr, 2020-10-03:
> @tikscr Is this still reproducible with the kext built from master https://github.com/OpenIntelWireless/itlwm/commit/9b60103c3369e950027286b2a0639a067ac93de0?

How can i build airportitwlm from master? I see no manual for it.

williambj1, 2020-10-03:
> > @tikscr Is this still reproducible with the kext built from master [9b60103](https://github.com/OpenIntelWireless/itlwm/commit/9b60103c3369e950027286b2a0639a067ac93de0)?
> 
> How can i build airportitwlm from master? I see no manual for it. Or xcodebuild would produce both itwlm and airportitwlm?

I haven't updated the docs, you can use our Ci config as a reference.

https://github.com/OpenIntelWireless/itlwm/blob/0120e171f6424a461d3c5973c3e55b5af8cd8603/.github/workflows/main.yml#L46

Jh0817, 2020-10-10:
My notebook's WI-FI and bluetooth is working!
Thank you!
But AirDrop and Handoff is not working.

Jh0817, 2020-10-16:
Airdrop and Handoff not working!

My device info :Samsung notebook 9 lite M58S,macOS 10.15.7, OpenCore 0.6.1

Jh0817, 2020-10-16:
Unlock Mac with Apple Watch is not working too

williambj1, 2020-10-31:
Feel free to request a reopen if this issue is still present using the latest alpha.

closed 2020-10-31 by williambj1