OpenIntelWireless/itlwm #310: 曲线救国连接iPhone热点

xPlutow, 2020-09-22:
用线连接iPhone--开热点--拔掉iPhone

或者先点iPhone的热点,等提示类似“连接失败不在范围内”之后再线连手机就会连上.

[image]

nn-666, 2020-09-22:
网卡慢慢有解，完美hackintosh指日可待

M69W, 2020-10-08:
> 用线连接iPhone--开热点--拔掉iPhone
> 
> 或者先点iPhone的热点,等提示类似“连接失败不在范围内”之后再线连手机就会连上.
> 
> [image]

另外，
使用 `AirportItlwm.kext` ，先通过蓝牙连接，再连接iPhone 热点(`智能热点`)；
这种情况不稳定，可能十次只有一两次成功。
[image]

win1010525, 2020-10-10:
对，是这样。。。问题出在不能p2p上，不能直连iPhone，手机先开电脑再联可以

asd45183, 2020-10-11:
请教下，加载驱动的时候有提示签名无效么？ 还是说高版本系统才会出现

win1010525, 2020-10-11:
不会的，你通过OC加载就好了

asd45183, 2020-10-12:
好的，我通过OC 开机加载试试。需要先更改无线密码么？

closed 2022-03-03 by williambj1

atomxing, 2025-06-19:
关掉电脑的蓝牙，再链接热点，成功率高