OpenIntelWireless/itlwm #313: [Bug/Feature request] Wrong country code

alessiomatricardi, 2020-09-23:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Not properly

**Environment**

- Kext Version: AirportItlwm 1.1.0
- WiFi Card Model: AC7265
- Product ID: 0x095A
- macOS Version: macOS Catalina 10.15

**Description**
Country code is "ZZ".
In my case, it should be "IT" in Italy.

**Bug Report Archive**
I think it is not necessary

**Kext Download Source**
Github 1.1.0 stable release

eric2877, 2020-09-29:
I have the same problem, my card is 8265, Country code is "ZZ",  it should be "CN" , is there any way to change the code by hand?

ViPiMP, 2020-10-03:
> I have the same problem, my card is 8265, Country code is "ZZ", it should be "CN" , is there any way to change the code by hand?

Sure!!
Did it by myself today.
[image]

Look inside the Kext, and find the plist.
In the Plist-file, you will find the country code (devREGION) very near at top!! 
greetings

alessiomatricardi, 2020-10-03:
Other available networks have the right country code (image below) so i think it's not complicated to resolve this problem. I've also found the line where "ZZ" is copied into a specific variable, but I don't know where is stored the right country code information
[image]

eric2877, 2020-10-09:
so sad it doesn’t work， I think u miss understanding， “Localization native development region” can’t change the “ZZ” in last line。

zxystd, 2020-10-18:
@eric2877 @alessiomatricardi  It is just a display issue, the country code is correct while working, Airport wants the country code when it is in initial state but the really country code is parsing from the card later, and Airport will never update it. I think we will not fix it. 
Also, If you think it is necessary to implement the function of customizing the country code, such as adding parameters to the startup parameters to solve the problem of some national frequency band conflicts, it is also OK.
Thanks.

alessiomatricardi, 2020-10-18:
@zxystd mm okey, i tought it was possible to display the correct country code as @usr-sse2 done with his old project here: https://github.com/usr-sse2/Black80211-Catalina/commit/7e16ac84a1c38af370135759a291331ef59d1565 and https://github.com/usr-sse2/itlwm/commit/50634943cc6f579897cbdf98faf9cd5f0f305d35
Maybe he has more info about it

usr-sse2, 2020-10-18:
> @zxystd mm okey, i tought it was possible to display the correct country code as @usr-sse2 done with his old project here: [...]

Did it work? I did it blindly, because my card is 8265 with LAR disabled, so country code requests are ignored. Anyway, I already began implementing getCOUNTRY_CODE and setCOUNTRY_CODE in AirportItlwm, will make a PR soon.

alessiomatricardi, 2020-10-18:
> > @zxystd mm okey, i tought it was possible to display the correct country code as @usr-sse2 done with his old project here: [usr-sse2/Black80211-Catalina@7e16ac8](https://github.com/usr-sse2/Black80211-Catalina/commit/7e16ac84a1c38af370135759a291331ef59d1565) and [usr-sse2@5063494](https://github.com/usr-sse2/itlwm/commit/50634943cc6f579897cbdf98faf9cd5f0f305d35)
> > Maybe he has more info about it
> 
> Did it work? I did it blindly, because my card is 8265 with LAR disabled, so country code requests are ignored. Anyway, I already began implementing getCOUNTRY_CODE and setCOUNTRY_CODE in AirportItlwm, will make a PR soon.

Also my card has LAR disabled, so it didn't work. But maybe there is a workaround to make it works

zxystd, 2020-10-18:
> Did it work? I did it blindly, because my card is 8265 with LAR disabled, so country code requests are ignored. Anyway, I already began implementing getCOUNTRY_CODE and setCOUNTRY_CODE in [...]

Delete the LAR condition code, and hardcode a country code to see if it works. I have tried to implement by self in gen1 and gen2 card, but it seems not work, works like above what I said, I think there is a notification about country code changing?

zxystd, 2020-11-03:
[image]
[image]
[image]
confirm that the country code can be changed.

MichaelDevC, 2020-11-08:
> Other available networks have the right country code (image below) so i think it's not complicated to resolve this problem. I've also found the line where "ZZ" is copied into a specific variable, but I don't know where is stored the right country code information
> [image]

Hi, also i'm from Italy and the country code is "ZZ", how you fixed? 

Because the correct country code is the system that changes the Wi-Fi frequency and make faster connection based on the router that you use...

alessiomatricardi, 2020-11-08:
> Hi, also i'm from Italy and the country code is "ZZ", how you fixed?
>
> Because the correct country code is the system that changes the Wi-Fi frequency and make faster connection based on the router [...]

I didn't fix it, that photo refers to one of the available networks (not the one you're connected).

zxystd, 2020-11-08:
please help to check whether the country code recognize correctly.

[归档.zip](https://github.com/OpenIntelWireless/itlwm/files/5506676/default.zip)

pytason, 2020-11-08:
@zxystd 
I can confirm that it is recognized correctly now

alessiomatricardi, 2020-11-08:
> please help to check whether the country code recognize correctly.
> 
> [归档.zip](https://github.com/OpenIntelWireless/itlwm/files/5506676/default.zip)

Unfortunately I changed my wifi card some days ago with a Broadcom one, so I can't provide any feedback

MichaelDevC, 2020-11-08:
> please help to check whether the country code recognize correctly.
> 
> [归档.zip](https://github.com/OpenIntelWireless/itlwm/files/5506676/default.zip)

Same problem... Country code still 'ZZ'; i reseted also the network settings -.-

usr-sse2, 2020-11-09:
>please help to check whether the country code recognize correctly. ...

@zxystd does it only display the country code or actually use only the allowed channels?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

usr-sse2, 2020-11-09:
>please help to check whether the country code recognize correctly. ...

@zxystd does it only display the country code or actually use the allowed channels?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

williambj1, 2020-11-09:
@usr-sse2 Allowed channels can be retrieved after setting the country code. There seems to some extra configuration going on with continuity regarding 802.11d. Currently, itlwm manages scanning on its own.

zxystd, 2020-11-09:
@usr-sse2 yes, only display, but the IO80211Family internal doing the channel filtering/validating according to the country code.

williambj1, 2020-11-09:
Should be fixed in c0554bc194e40d58dd4e9202eed907ff17f8244f.

Since we are planning on adding a custom option for cards that don't report country code, this issue will remain opened.

MichaelDevC, 2020-11-09:
Same problem... downloaded latest alpha build [ a6e8dea ] , but the country code in the Wi-Fi Card Info still 'ZZ'
I also deleted the network configuration with the guide in the ITLWM website -.-

closed 2020-11-15 by zxystd

startergo, 2021-07-18:
> Since we are planning on adding a custom option for cards that don't report country code, this issue will remain opened.

So this issue is closed, but It is still not resolved I guess? My card still reports zz as country code.

williambj1, 2021-07-19:
> > Since we are planning on adding a custom option for cards that don't report country code, this issue will remain opened.
> 
> So this issue is closed, but It is still not resolved I guess? My card still reports zz as a country code.

I strongly encourage you to read the docs, it only takes 5 minutes. There's a boot-args that controls the country code.