OpenIntelWireless/itlwm #314: Every network show as "perfectly received" / "very near"

moutorde, 2020-09-23:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: AirportItlwm 1.1.0
- WiFi Card Model: AC8265
- Product ID: 0x24FD
- macOS Version: 10.15.6

**Description**
Every network show as "perfectly received" / "very near", including networks that are far and show properly under Windows.

**Bug Report Archive**
[AirportItlwm.log](https://github.com/OpenIntelWireless/itlwm/files/5269342/AirportItlwm.log)

**Kext Download Source**
Github 1.1.0 stable release

juniorzhou, 2020-09-23:
I also had this bug once. Does not know what trigger it.

hybrid72, 2020-10-02:
Same issue here.

Kext Version: AirportItlwm 1.1.0
WiFi Card Model: AC7260
macOS Version: 10.15.7

Wifi Explorer Lite is showing all weak (neighbour) networks as 100%. Best Networks are below. My connected network is 64% (Should be about 80%).

williambj1, 2020-10-07:
Fixed in b919444

closed 2020-10-07 by williambj1