OpenIntelWireless/itlwm #315: Slow speeds on both 2.4 and 5 GHz networks - AirportItlwm

FaneCH, 2020-09-23:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: AirportItlwm_v1.0_Beta
- WiFi Card Model: Intel AC-3165
- Product ID: 0x8086
- macOS Version: 10.15.6

**Description**
Slow speeds on 2.4/5 GHz networks.
Happens on both AirportItlwm and itlwm (latest versions).

[image]

**Bug Report Archive**

[AirportItlwm.log](https://github.com/OpenIntelWireless/itlwm/files/5269468/AirportItlwm.log)

**Kext Download Source**
https://github.com/OpenIntelWireless/itlwm/releases

moutorde, 2020-09-23:
https://openintelwireless.github.io/itlwm/FAQ.html#performance-sucks

FaneCH, 2020-09-23:
If i'll have the speeds from the website you gave me, i'll be more then pleased. But 35 mb/s it's not ideal.

moutorde, 2020-09-23:
Probably due to your wifi card having 1x1 antenna compatibility.
The card used for the example is an AC9560 which is, at first, better on all points, and it also has 2x2 antennas compatibility.

FaneCH, 2020-09-23:
It is what it is then. Thanks man for the answers. It is really stable on the other hand, works like a charm! :)

closed 2020-09-23 by williambj1