OpenIntelWireless/itlwm #316: [AirportItlwm]WiFi does not work from a cold boot. Works fine if reboot from windows.

juniorzhou, 2020-09-23:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: AirportItlwm_v1.0_Beta_Catalina.kext
- WiFi Card Model: AX200
- Product ID: 
- macOS Version: 10.15.6

**Description**
The wifi card does not work from a cold boot. It works fine when boot from Windows. The kext was shown as "loaded" from the **kextstat**. I have attached both log files, one is the cold boot, another is the reboot from windows. This also happened when I used to use the itlwm.kext. 

**Bug Report Archive**
[reboot_from_win.txt](https://github.com/OpenIntelWireless/itlwm/files/5272453/reboot_from_win.txt)
[cold_boot_log_1.txt](https://github.com/OpenIntelWireless/itlwm/files/5272454/cold_boot_log_1.txt)

**Kext Download Source**
The 1.1.0 release.

williambj1, 2020-10-03:
Please attach your ioreg from both reboot and cold boot。

johnwtt, 2020-10-11:
Try using itlwm and Black80211

jihooyoon, 2020-10-22:
Have you tried disabling fast shutdown/startup on Windows? 
With this option enabled, "shutdown" is not truly shutdown, it just a kind of hibernate. I think this is the problem, btw, disabling something that is default is not a good solution

tommyvct, 2020-11-12:
This happens not only on macOS, but also on Linux. 

Try to turn off the computer from Windows, then boot it up to any other OSs other than Windows, the WiFi will not work at all. If you restart the computer from Windows, such problem will not happen.

Fast start-up is the culprit, it let the Windows version of firmware creep inside the wireless card and confuses other OSs which didn't expect such behaviour from Windows.

closed 2021-02-25 by williambj1

zejian-li, 2025-02-12:
Turning off WiFi in Windows before hibernating seems to allow AirportItlwm to work properly in MacOS.