OpenIntelWireless/itlwm #318: Wifi cannot work after wake up from sleep

liuyue0016, 2020-09-24:
**Have You Read Our Docs**
<Yes>

**Are You Reporting A Bug**
<Yes>

**Environment**

- Kext Version: 1.1.0 stable Airportitlwm
- WiFi Card Model: AC9560 2nd Generation
- Product ID: 
- macOS Version: 20A5374i
**Description**
<Wifi cannot work after wake up from sleep>

**Kext Download Source**
<Github>

williambj1, 2020-09-24:
Will open again once logs are provided.

closed 2020-09-24 by williambj1

liuyue0016, 2020-09-24:
不确定上传的对不对
[wifi-09-24-2020__18:43:47.740.log](https://github.com/OpenIntelWireless/itlwm/files/5277025/wifi-09-24-2020__18.43.47.740.log)
[wifi-09-24-2020__18:44:36.130.log](https://github.com/OpenIntelWireless/itlwm/files/5277027/wifi-09-24-2020__18.44.36.130.log)
[wifi-09-24-2020__18:47:16.621.log](https://github.com/OpenIntelWireless/itlwm/files/5277028/wifi-09-24-2020__18.47.16.621.log)
[wifi-09-24-2020__18:48:07.009.log](https://github.com/OpenIntelWireless/itlwm/files/5277030/wifi-09-24-2020__18.48.07.009.log)
[wifi.log](https://github.com/OpenIntelWireless/itlwm/files/5277032/wifi.log)
[wifivelocity_logarchive.log](https://github.com/OpenIntelWireless/itlwm/files/5277034/wifivelocity_logarchive.log)

[Uploading wifi-09-24-2020__18:43:47.740.log…]

[wifi-09-24-2020__18:44:36.130.log](https://github.com/OpenIntelWireless/itlwm/files/5276998/wifi-09-24-2020__18.44.36.130.log)

[wifivelocity_logarchive.log](https://github.com/OpenIntelWireless/itlwm/files/5276993/wifivelocity_logarchive.log)
[wifi.log](https://github.com/OpenIntelWireless/itlwm/files/5276994/wifi.log)
[wifi-09-24-2020__18:48:07.009.log](https://github.com/OpenIntelWireless/itlwm/files/5276995/wifi-09-24-2020__18.48.07.009.log)
[Uploading wifi-09-24-2020__18:47:16.621.log…