OpenIntelWireless/itlwm #326: [AirPort] Can't connect to hotspots

GGorAA, 2020-09-28:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes
**Environment**

- Kext Version: 1.1.0
- WiFi Card Model: Intel Wireless 8260
- Product ID: idk where is is, but here is Hackintool screenshots
[image]
[image]
- macOS Version: Big Sur 11.0 20A5364e

**Description**
I can't connect to networks. Well, WiFi entry is created, I can see the hotspots, but I can't connect to them. When I was on `itlwm.kext`% everything worked flawlessly. But when I switched to AirPort release, I lost my ability to connect to hotspots.
**Bug Report Archive**
Well, they are blank

**Kext Download Source**
This repo

GGorAA, 2020-09-28:
found out Product ID. it was 0x24f3

ErrorErrorError, 2020-09-28:
From @zxystd release notes for AiportIltmw.

`` Instant HotSpot from iPhone can be recognized but may likely fail to connect. Workaround: use itlwm.kext with HeliPort.``

GGorAA, 2020-09-28:
Nope, my network is my normal home WiFi

williambj1, 2020-09-28:
Will reopen once logs are provided. Read our docs for more info.

closed 2020-09-28 by williambj1

GGorAA, 2020-09-29:
There are no logs, when I pasted the commands nothing was in output. That's why I said that this is blank

GGorAA, 2020-09-29:
Nope, found logs. here they are
[AirportItlwm.log](https://github.com/OpenIntelWireless/itlwm/files/5297339/AirportItlwm.log)

williambj1, 2020-09-29:
> Kext Download Source
> This repo

Sorry that I did not mention this earlier, could you please be specific about where did you get this kext? There are multiple versions of "Airport" in this repo. Please provide the URL.

The logs you provided are incomplete and not useful to identify your issue. Please follow our docs *strictly*.