OpenIntelWireless/itlwm #329: [Intel AC-7260] Wifi is not working on Windows after using itlwm or AirPortItlwm

vaczi001, 2020-09-30:
**Have You Read Our Docs**
- Yes

**Are You Reporting A Bug**
- Yes

**Environment**

- Kext Version: **AirPortItlwm v1.0 Beta, itlwm v1.1.0 + HeliPort v1.0.1**
- WiFi Card Model: **Intel AC-7260**
- Product ID: **Wifi:** 7d0, **Bluetooth:** 7dc  
- macOS Version: **10.11 Beta 2 and 6 - Windows 10 Dual-Boot with OC 0.6.1** 

**Description**
- **First on macOS:**
  - AirportItlwm worked fine on macOS Big Sur Beta 2, but not on Big Sur Beta 6. Wifi is cannot be started from Network settings nor from control center. However with itlwm and HeliPort it's working. Bluetooth is working all the time.
  - USB is properly patched
- **Second on Windows:**
  - Wifi is not working after I used AirportItlwm or itlwm + HeliPort (on macOS Beta 2 and 6 too)
  - In Device Manager it says "This device cannot start. (Code 10), {Operation Failed} - The requested operation was unsuccessful."
[image]

  - Still not working after removing itlwm completely, it's stuck with that state.
- What I've tried:
  - Cold boot
  - Reset NVRAM
  - Reinstalling Windows drivers

Nothing helped unfortunately.

- On Catalina 10.15.6 it's worked with itlwm + HeliPort, BUT if I disabled the wifi inside macOS, then I had no Wifi on Windows (same error, also Bluetooth didn't work). After re-enabling it worked again after rebooting to Windows.

**Bug Report Archive**
[itlwm_debug.zip](https://github.com/OpenIntelWireless/itlwm/files/5306103/itlwm_debug.zip)
- the zip contains bootlog, IOReg, OC EFI (and the pictures)
- Inside the bootlog at line 2382 says "itlwm: fatal firmware error", after that it starts spamming:
```
2020-09-26 11:25:14.783448+0200 0x159      Default     0x0                  0      0    kernel: (IOUSBHostFamily) 000021.671534 IOUSBHostDevice@14500000: IOUSBHostDevice::updateIdlePolicyGated: IOUSBHostInterface@0 [<private>] has an idle policy of 500ms

2020-09-26 11:25:14.793510+0200 0x159      Default     0x0                  0      0    kernel: (IOUSBHostFamily) 000021.681610 IOUSBHostDevice@14500000: IOUSBHostDevice::updateIdlePolicyGated: IOUSBHostInterface@1 [<private>] has an idle policy of 500ms

2020-09-26 11:25:14.803814+0200 0x159      Default     0x0                  0      0    kernel: (IOUSBHostFamily) 000021.691917 IOUSBHostDevice@14500000: IOUSBHostDevice::updateIdlePolicyGated: consensus idle policy timeout 500ms

2020-09-26 11:25:15.371240+0200 0xe9       Default     0x0                  0      0    kernel: (IOUSBHostFamily) 000022.259334 IOUSBHostDevice@14500000: IOUSBHostDevice::forcePowerGated: kPowerStateSuspended timeout 1000ms

2020-09-26 11:25:15.384485+0200 0x159      Default     0x0                  0      0    kernel: (IOUSBHostFamily) 000022.272584 IOUSBHostDevice@14500000: IOUSBHostDevice::powerStateWillChangeToGated: kPowerStateSuspended

2020-09-26 11:25:15.394805+0200 0x159      Default     0x0                  0      0    kernel: (IOUSBHostFamily) 000022.282900 IOUSBHostDevice@14500000: IOUSBHostDevice::setPowerStateGated: kPowerStateOn -> kPowerStateSuspended

2020-09-26 11:25:15.408088+0200 0x159      Default     0x0                  0      0    kernel: (IOUSBHostFamily) 000022.296179 IOUSBHostDevice@14500000: IOUSBHostDevice::setPowerStateGated: suspending pipes

2020-09-26 11:25:15.417802+0200 0xe9       Default     0x0                  0      0    kernel: (IOUSBHostFamily) 000022.305900 IOUSBHostDevice@14500000: IOUSBHostDevice::forcePowerGated: allowing power level to float

2020-09-26 11:25:15.426889+0200 0xe9       Default     0x0                  0      0    kernel: (IOUSBHostFamily) 000022.314991 IOUSBHostDevice@14500000: IOUSBHostDevice::forcePowerGated: kPowerStateSuspended completed with 0x00000000 after 33ms

2020-09-26 11:25:15.437064+0200 0x159      Default     0x0                  0      0    kernel: (IOUSBHostFamily) 000022.325164 IOUSBHostDevice@14500000: IOUSBHostDevice::powerStateWillChangeToGated: kPowerStateOn

2020-09-26 11:25:15.446150+0200 0x159      Default     0x0                  0      0    kernel: (IOUSBHostFamily) 000022.334247 IOUSBHostDevice@14500000: IOUSBHostDevice::setPowerStateGated: kPowerStateSuspended -> kPowerStateOn

2020-09-26 11:25:15.455587+0200 0x159      Default     0x0                  0      0    kernel: (IOUSBHostFamily) 000022.343674 IOUSBHostDevice@14500000: IOUSBHostDevice::setPowerStateGated: resuming pipes

2020-09-26 11:25:15.464684+0200 0x74       Default     0x0                  0      0    kernel: (IOUSBHostFamily) 000022.352786 IOUSBHostDevice@14500000: IOUSBHostDevice::powerChangeDone:
```
[image]

- It's going on infinitely until I shut down the computer
- The IOReg shows that "IOUSBHostDevice@14500000" is the Bluetooth device.

**Kext Download Source**
- From OpenIntelWireless's GitHub release page.

J-o-s-h-S-i-m-s, 2020-10-01:
This is happening with my Intel AC 8265 as well.

vaczi001, 2020-10-01:
> This is happening with my Intel AC 8265 as well.

One thing I got in my mind, try a CMOS reset then boot right into Windows and let's see if it's working or not.
still not a convenient way to use Wifi on Windows tho.

J-o-s-h-S-i-m-s, 2020-10-01:
> > This is happening with my Intel AC 8265 as well.
> 
> One thing I got in my mind, try a CMOS reset then boot right into Windows and let's see if it's working or not.
> still not a convenient way to use Wifi on Windows tho.

Okay, I reset the CMOS and now I can use wifi on Windows 10 as well as Catalina. How long did this solution work for you?

vaczi001, 2020-10-01:
I didn't actually try that, was just an idea. Glad it's worked. Now I'm using a TP-Link USB wifi, so I'll do the same soon.
That is my 2nd notebook and I rarely use it (usually only for betas like Big Sur, etc., my father uses it more often on Windows) so I don't really tell you how long will it last.

seungjunProgramming, 2020-10-02:
My laptop works fine

vaczi001, 2020-10-06:
Okay so here is some news:

- Installed Big Sur Beta 9 (complete reinstall), reinstalled AirPortItlwm to L/E and updated persmissions & kextcache, wifi is working fine. 

- Checked whether Wifi is working or not on Windows
  - Wifi still not working so I decided to do a CMOS reset on my Lenovo ThinkPad T440
    - Wifi still not working after that, same error and same spam on macOS

williambj1, 2020-10-06:
@vaczi001 

> reinstalled AirPortItlwm to L/E and updated persmissions & kextcache

You should never do that.

---

> I double-checked, moving to OpenCore and booting with that to macOS breaks Windows wifi

We don't have any Bootloader specific code. Based on your description, the issue is our of our control. If you have solid evidence (Bootloader Logs, Windows Driver Debug Logs, etc) of OpenCore putting your card into a bad state, you should report them to Acidanthera.

closed 2020-10-06 by williambj1

drinvincibilis, 2021-01-10:
Hi, this is quite a serious behavior. I am another one that set up dual boot with macOS and Windows. After using AirPortItlwm the device is reported as broken in Windows.
[image]

vaczi001, 2021-01-10:
@drinvincibilis 
I looked a bit into it and what I learned is that OpenCore is injecting all the SSDT/DSDT/other modified data to Windows too while Clover is running Windows native with native unmodified DSDT/other UEFI data, so maybe that's why it's not working with OpenCore and working on Clover.

One thing I still don't understand is that why did it work before using AirPortItlwm, and why not after using it. 
Somehow it's related to OpenCore, but also to AirportItlwm too, idk, I collected all the logs and attached, I can't do anything about that (I can do tests tho if someone ask me).

Boot Windows directly (not with OpenCore, but with Windows's own loader, you have to disable OpenCore's bootstrap for that), it'll start working again. It's pretty annoying tho.

drinvincibilis, 2021-01-10:
Hi @vaczi001 thanks for the feedback. I did a step further and mixed the hard drive EFI from both macOS/OpenCore and Windows so it does dual boot. I will revert to Windows original EFI contents and boot macOS with the installer USB stick EFI.

drinvincibilis, 2021-01-10:
@vaczi001 I can confirm. Reverting to original Windows EFI partition and booting restored the Wi-Fi huh

vaczi001, 2021-01-20:
**SOLVED!**
 So I was right, OpenCore is injecting SSDTs and all other patches for Windows too, so that's why it isn't working with OpenCore. So we need OSI patches to prevent injecting data for Windows by checking what is the OS. If Windows, do not inject patches. Here, just put this file to ACPI folder and add to your config.plist 1st column:
https://github.com/tylernguyen/x1c6-hackintosh/blob/master/EFI-OpenCore/EFI/OC/ACPI/SSDT-Darwin.aml

zen3gr, 2021-08-08:
Hello, link for the solution is dead? Would you be kind enough to update it and maybe write the basic steps of making opencore not inject data to Windows? FYI, I've got a thinkpad x240 with intel 7260 wireless; wifi is working fine on hackintosh (catalina) but got error "this device cannot start (code 10)" on windows 10.

Thanks!

> **EDIT: Still not solved**, after some restarts it's just stopped working again. No idea why OpenCore is injecting SSDTs and all other data to Win too
> The bluetooth still spamming me the same errors I wrote in the first post in macOS.
> I gave up and changed the wifi card to BCM94352Z.
> 
> _**(SOLVED!** So I was right, OpenCore is injecting SSDTs and all other patches for Windows too, so that's why it isn't working with OpenCore. So we need OSI patches to prevent injecting data for Windows by checking what is the OS. If Windows, do not inject patches. Here, just put this file to ACPI folder and add to your config.plist 1st column: https://github.com/tylernguyen/x1c6-hackintosh/blob/master/EFI-OpenCore/EFI/OC/ACPI/SSDT-Darwin.aml)_

kushwavez, 2021-08-08:
I forgot to update on this, but it didn't work even with that SSDT-Darwin.aml. 
You cannot make OC to disable patching on Windows, so you have to workaround it by using [Refind](https://www.rodsbooks.com/refind/) or [RefindPlus](https://github.com/dakanji/RefindPlus) as a 2nd bootloader. So you have to boot OC first, then Refind, then Windows. 

It is really quirky "solution" so I just decided to not use OC and use Clover instead.

Or my next idea is to select Windows Boot Manager with F12 when booting, that will boot Windows directly if I'm correct.

zen3gr, 2021-08-08:
Thank you very much!

Stevemoretz, 2021-12-07:
Still happens, migrating to clover is not really a solution :)

kushwavez, 2021-12-08:
I did not say that, read again. The solution is to use Refindplus