OpenIntelWireless/itlwm #333: Intel Dual Band Wireless-AC 3165 loses connection after a few minutes

ciccio-90, 2020-10-04:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: itlwm v1.1.0 stable
- WiFi Card Model: Intel Corporation Dual Band Wireless-AC 3165 Plus Bluetooth
- Product ID: Vendor 0x8086 Device 0x3155
- macOS Version: Catalina 10.15.7

**Description**
After a few minutes that I use Wi-Fi, the connection loses. So I must turn Wi-Fi off and then turn Wi-Fi on to reconnect, but after a cupple of minutes the connection loses again and I must repete the operation.

**Bug Report Archive**

**Kext Download Source**
https://github.com/OpenIntelWireless/itlwm/releases/tag/v1.1.0

williambj1, 2020-10-04:
Will reopen once logs are provided.

ciccio-90, 2020-10-06:
[bugreport_4653.zip](https://github.com/OpenIntelWireless/itlwm/files/5335668/bugreport_4653.zip)

williambj1, 2020-10-07:
Looks like itlwm is sending deauth to the router. CC @zxystd

ciccio-90, 2020-10-07:
How can I look that? 
However I'm having the same beaviour using Airportitlwm.

enbyautumn, 2020-10-11:
Same thing is happening to me

ciccio-90, 2020-10-11:
In my home network the 2.4GHz and 5GHz have the same SSID and password, so I've disabled the 5GHz and Hackintosh Wi-Fi seems to work in a correct way using Airportitlwm kext. I do not know if this info can help you **@williambj1** and **@zxystd** .

asd45183, 2020-10-13:
@ciccio-90  hi brother，i use the same wifi card AC3165 (0x8086, 0x4010)  and OC 0.6.1 whit MACOS 10.5.7 ,but it looks like my wifi card does not work  with this driver , may you give me some suggestion ?

ciccio-90, 2020-10-13:
> @ciccio-90 hi brother，i use the same wifi card AC3165 (0x8086, 0x4010) and OC 0.6.1 whit MACOS 10.5.7 ,but it looks like my wifi card does not work with this driver , may you give me some suggestion ?

Hi bro, can you tell me the steps u followed to configure the kext?

asd45183, 2020-10-13:
@ciccio-90  thanks bro, I had used  itlwm with  HeliPort ,that does not work ;
 now I use airpotItwm  that also not work ...
i put my config in the zip file ,thank you.

[log&ocCOnfig.zip](https://github.com/OpenIntelWireless/itlwm/files/5371576/log.ocCOnfig.zip)

ciccio-90, 2020-10-13:
**@asd45183** Hey man, try to use this injection order in **config.plist**
**- AirportItlwm
- Bluetooth Firmware
- Bluetooth Injector**

Then to **enable AirportItlwm support with OpenCore**, you'll need to add in your **config.plist** under Misc -> Security -> **SecureBootModel** by setting it as **Default**.

Let me know!

zxystd, 2020-11-15:
I think it is supported?

closed 2022-01-15 by williambj1