OpenIntelWireless/itlwm #335: [Airportitlwm] When I connect to my home wifi, it's connected but the web pages and MAC APPS don't load (but through iphone's hotspot hotspot works)

festano92, 2020-10-07:
hello, it's a strange issue that i've had with itlwm+heliport too. Sometimes this issue doesn't show up and I have internet connection at start. 
When it's connected to my home wifi, the web pages keep loading and Apps don't open or open slowly, they seems stucked. If I turn wifi off (when home wifi connected) the apps suddenly open. Moreover, I can't create the real-time airports log for the reason explained. 
[AirportItlwm.log](https://github.com/OpenIntelWireless/itlwm/files/5339418/AirportItlwm.log)

williambj1, 2020-10-07:
Please use the issue format. Will reopen once it's done.

williambj1, 2020-10-07:
What's the memory and CPU usage of `kernel task`?

TBH, we can't tell what's wrong based on your limited information and it's not reproducible on our hacks.

I would suggest try installing OpenBSD and see if you face the same issue to help us identify the factors.

festano92, 2020-10-07:
`**What's the memory and CPU usage of kernel task?**`
Right now Cpu and memory Usage is extremely low, now it's working though. The next restart, when the issue probably occurs again, I'll tell you if there's something wrong about CPU and memory.

`**I would suggest try installing OpenBSD and see if you face the same issue to help us identify the factors.**`
Ok. Anyhow I'll try to gather as much information as I can, thanks.

festano92, 2020-10-07:
> Wed Oct  7 13:08:28: <kernel> Unexpected payload found for message 4, dataLen 0
Wed Oct  7 13:08:28: <kernel> Error storing postMessage 4
Wed Oct  7 13:08:28: <kernel> postMessage::1349 APPLE80211_M_BSSID_CHANGED received
Wed Oct  7 13:08:28:  <kernel> Unexpected payload found for message 4, dataLen 0
Wed Oct  7 13:08:28: <kernel> Error storing postMessage 4
Wed Oct  7 13:08:28: <kernel> Unexpected payload found for message 4, dataLen 0
Wed Oct  7 13:08:28: <kernel> Error storing postMessage 4
Wed Oct  7 13:08:28: <kernel> Unexpected payload found for message 4, dataLen 0
Wed Oct  7 13:08:28: <kernel> Error storing postMessage 4
Wed Oct  7 13:08:28: <kernel> AirportItlwm::getSSIDData(): Get failure: APPLE80211_IOC_SSID: -536870212
Wed Oct  7 13:08:28: <airportd[208]> _handleLinkEvent: WiFi is not powered. Resetting state variables.
Wed Oct  7 13:08:28: <kernel> IO80211VirtualInterface::handleSIOCSIFFLAGS : Source handleIoctlGated calling peerManager->disable
Wed Oct  7 13:08:28: <airportd[208]> _processIPv4Changes: ARP/NDP offloads disabled, not programming the offload
Wed Oct  7 13:08:28: <airportd[208]> _processIPv4Changes: ARP/NDP offloads disabled, not programming the offload
Wed Oct  7 13:08:29: <kernel> en1: setting diversity to: -1
Wed Oct  7 13:08:29: <kernel> en1: Error configuring antenna diversity (index = -1).
Wed Oct  7 13:08:29: <kernel> en1: setting tx antenna: -1
Wed Oct  7 13:08:29: <kernel> en1: Error configuring transmit antenna (index = -1).
Wed Oct  7 13:08:29: <airportd[208]> airportdProcessDriverEvent: configure Apple device IE failed on en1 err = 5 (Input/output error)
Wed Oct  7 13:08:29: <kernel> Setting BTCoex Profile: band:8
Wed Oct  7 13:08:29: <kernel> Profile[0]: mode:7; desense:0; desense_level:0; chain_power_offset:<kernel> 0,<kernel> 0,<kernel> 0,<kernel> 0,<kernel> 
Wed Oct  7 13:08:29: <kernel> Profile[1]: mode:7; desense:0; desense_level:0; chain_power_offset:<kernel> 0,<kernel> 0,<kernel> 0,<kernel> 0,<kernel> 
Wed Oct  7 13:08:29: <kernel> Profile[2]: mode:7; desense:0; desense_level:0; chain_power_offset:<kernel> 0,<kernel> 0,<kernel> 0,<kernel> 0,<kernel> 
Wed Oct  7 13:08:29: <kernel> Profile[3]: mode:7; desense:0; desense_level:0; chain_power_offset:<kernel> 0,<kernel> 0,<kernel> 0,<kernel> 0,<kernel> 
Wed Oct  7 13:08:29: <kernel> Setting BTCoex Profile: band:8
Wed Oct  7 13:08:29: <kernel> Profile[0]: mode:7; desense:0; desense_level:0; chain_power_offset:<kernel> 0,<kernel> 0,<kernel> 0,<kernel> 0,<kernel> 
Wed Oct  7 13:08:29: <kernel> Profile[1]: mode:7; desense:0; desense_level:0; chain_power_offset:<kernel> 0,<kernel> 0,<kernel> 0,<kernel> 0,<kernel> 
Wed Oct  7 13:08:29: <kernel> Profile[2]: mode:7; desense:0; desense_level:0; chain_power_offset:<kernel> 0,<kernel> 0,<kernel> 0,<kernel> 0,<kernel> 
Wed Oct  7 13:08:29: <kernel> Profile[3]: mode:7; desense:0; desense_level:0; chain_power_offset:<kernel> 0,<kernel> 0,<kernel> 0,<kernel> 0,<kernel> 
Wed Oct  7 13:08:29: <airportd[208]> systemWokenByWiFi: System Wake Reason not found
Wed Oct  7 13:08:30: <airportd[208]> ERROR: rapportd (383) is not entitled for com.apple.wifi.join_history, will not allow request
Wed Oct  7 13:08:30: <airportd[208]> ERROR: sharingd (416) is not entitled for com.apple.wifi.join_history, will not allow request
Wed Oct  7 13:08:30: <kernel> Unexpected payload found for message 4, dataLen 0
Wed Oct  7 13:08:30: <kernel> Error storing postMessage 4
Wed Oct  7 13:08:30: <kernel> postMessage::1349 APPLE80211_M_BSSID_CHANGED received
Wed Oct  7 13:08:30:  <kernel> Unexpected payload found for message 4, dataLen 0
Wed Oct  7 13:08:30: <kernel> Error storing postMessage 4
Wed Oct  7 13:08:30: <airportd[208]> ERROR: rapportd (383) is not entitled for com.apple.wifi.join_history, will not allow request
Wed Oct  7 13:08:30: <airportd[208]> ERROR: sharingd (416) is not entitled for com.apple.wifi.join_history, will not allow request
Wed Oct  7 13:08:30: <airportd[208]> ERROR: rapportd (383) is not entitled for com.apple.wifi.join_history, will not allow request
Wed Oct  7 13:08:30: <airportd[208]> ERROR: sharingd (416) is not entitled for com.apple.wifi.join_history, will not allow request
Wed Oct  7 13:08:30: <kernel> Setting BTCoex Profile: band:8
Wed Oct  7 13:08:30: <kernel> Profile[0]: mode:7; desense:0; desense_level:0; chain_power_offset:<kernel> 0,<kernel> 0,<kernel> 0,<kernel> 0,<kernel> 
Wed Oct  7 13:08:30: <kernel> Profile[1]: mode:7; desense:0; desense_level:0; chain_power_offset:<kernel> 0,<kernel> 0,<kernel> 0,<kernel> 0,<kernel> 
Wed Oct  7 13:08:30: <kernel> Profile[2]: mode:7; desense:0; desense_level:0; chain_power_offset:<kernel> 0,<kernel> 0,<kernel> 0,<kernel> 0,<kernel> 
Wed Oct  7 13:08:30: <kernel> Profile[3]: mode:7; desense:0; desense_level:0; chain_power_offset:<kernel> 0,<kernel> 0,<kernel> 0,<kernel> 0,<kernel> 
Wed Oct  7 13:08:30: <airportd[208]> _processIPv4Changes: ARP/NDP offloads disabled, not programming the offload
Wed Oct  7 13:08:30: <airportd[208]> _processIPv4Changes: ARP/NDP offloads disabled, not programming the offload
Wed Oct  7 13:08:30: <kernel> AirportItlwm::apple80211RequestIoctl type APPLE80211_IOC_LINK_CHANGED_EVENT_DATA cmd GET
Wed Oct  7 13:08:30: <kernel> Unexpected payload found for message 4, dataLen 0
Wed Oct  7 13:08:30: <kernel> Error storing postMessage 4
Wed Oct  7 13:08:30: <kernel> AirportItlwm::apple80211RequestIoctl type APPLE80211_IOC_SSID cmd GET
[... cut ...]
Wed Oct  7 13:08:33: <kernel> Profile[1]: mode:7; desense:0; desense_level:0; chain_power_offset:<kernel> 0,<kernel> 0,<kernel> 0,<kernel> 0,<kernel> 
Wed Oct  7 13:08:33: <kernel> Profile[2]: mode:7; desense:0; desense_level:0; chain_power_offset:<kernel> 0,<kernel> 0,<kernel> 0,<kernel> 0,<kernel> 
Wed Oct  7 13:08:33: <kernel> Profile[3]: mode:7; desense:0; desense_level:0; chain_power_offset:<kernel> 0,<kernel> 0,<kernel> 0,<kernel> 0,<kernel> 
Wed Oct  7 13:08:33: <airportd[208]> _processIPv4Changes: ARP/NDP offloads disabled, not programming the offload<

festano92, 2020-10-07:
I removed SSDT-UIAC.aml and remapped the ports. Now I use Usbmap.kext, and the wifi is working, but there was one of the 5/6 boot where wifi didn't work and at the same time I couldn't turn off the bluetooth. Can my problem be related to bluetooth issue?

alessiomatricardi, 2020-10-07:
Try to use this injection order in config.plist
- AirportItlwm
- Bluetooth Injector
- Bluetooth Firmware

festano92, 2020-10-08:
yeah, I read somewhere when I first use itlwm, and I've used that order from then on

festano92, 2020-10-10:
From IOReg:
Works when:
IO80211Band=2.4GHz
IO80211Channell=0x06

Doesn't work when:
IO80211Band=5GHz
IO80211Channell=0x90

williambj1, 2020-10-10:
@festano92 Does your router share the same SSID with different bands?

festano92, 2020-10-10:
@williambj1 Yeah 2.4Ghz and 5Ghz share the same SSID!

festano92, 2020-10-10:
i changed the SSIDs of the networks and clearly now it works, is there an alternative, without changing the SSID? It's quite common having the same SSID for different bands unfortunately. Thanks!

williambj1, 2020-10-10:
> is it normal that using 5Ghz band I have a slower DL and UP speed than 2.4Ghz?

Should be the opposite.

---

>  is there an alternative, without changing the SSID?

We'll try to reproduce this issue and determine whether it's a bug.

alessiomatricardi, 2020-10-10:
> i changed the SSIDs of the networks and clearly now it works, is there an alternative, without changing the SSID? It's quite common having the same SSID for different bands unfortunately. Thanks!

Check that your router supports band steering and it's enabled. If you only call both network with same SSID, the driver is unstable. I had same problem.

> Ps: is it normal that using 5Ghz band I have a slower DL and UP speed than 2.4Ghz?

I've also this problem. 2.4Ghz is more stable & DL/UP is better than 5Ghz. First of all the 5Ghz's upload is very poor

festano92, 2020-10-10:
For me: 
2.4Ghz    DL=40/45 Mbps
                   UL=10/15 Mbps

5Ghz       DL=5/7Mbps
                  UL=1/2Mbps

@alessiomatricardi are you italian? Maybe it's an Italian problem lol

alessiomatricardi, 2020-10-10:
> For me:
> 2.4Ghz DL=40/45 Mbps
> UL=10/15 Mbps
> 
> 5Ghz DL=5/7Mbps
> UL=1/2Mbps
> 
> @alessiomatricardi are you italian? Maybe it's an Italian problem lol

Yes lol :)
I've a 200 Mbps FTTC connection

2.4Ghz
ping 25ms
DL 11 Mbps
UP 9 Mbps (while i was testing, connection goes down...)

5Ghz
ping 23ms
DL 17 Mbps
UP 8 Mbps (but unable to do a Facetime face-to-face call)

williambj1, 2020-10-10:
> Yes lol :)
> I've a 200 Mbps FTTC connection
>
> 2.4Ghz
> ping 25ms
> DL 11 Mbps
> UP 9 Mbps (while i was testing, connection goes down...)
>
> 5Ghz
> ping 23ms
> DL 17 Mbps
> UP 8 Mbps (but unable to do a Facetime [...]

Try an iperf3 test, should be location independent.

alessiomatricardi, 2020-10-10:
idk if i made it right
[iperf3.zip](https://github.com/OpenIntelWireless/itlwm/files/5359218/iperf3.zip)

williambj1, 2020-10-16:
Duplicate of #333

closed 2020-10-16 by williambj1