OpenIntelWireless/itlwm #337: Handoff and shared clipboard partially work

enPause96, 2020-10-07:
**Have You Read Our Docs**

Yes
**Are You Reporting A Bug**

Yes
**Environment**

- Kext Version: AirportItlwm 1.2.0
- WiFi Card Model: Intel Wireless N 7260
- Product ID: 0x08B1
- macOS Version: Big Sur 11.0 beta 8

**Description**

Handoff and shared clopboard work only in one direction. Handoff icon doesn't show on dock when i use Safari, Message, Mail on my iPhone but when i use them on my PC i can see them on iPhone. also i can access to the clipboard of my PC on my iPhone but can't access of clipboard of my iPhone from PC
i tried resetting network preferences and log out then relog in on both my devices but didn't work
**Bug Report Archive**

[AirportItlwm.log](https://github.com/OpenIntelWireless/itlwm/files/5342415/AirportItlwm.log)
real time airport log https://pastebin.pl/view/d577d059
**Kext Download Source**

from official OpenIntelWireless github

[image]
[image]

williambj1, 2020-10-08:
Still sounds like a userspace issue to me. The logs didn't contain useful information for debugging continuity. Maybe you can contact people that own the same card as yours to help us identify whether it's a card-specific or a userspace issue.

williambj1, 2020-10-08:
Duplicate of #292

closed 2020-10-08 by williambj1