OpenIntelWireless/itlwm #338: Why is the LINK SPEED is 100Mbit/s

xqf9725, 2020-10-08:
Why is the LINK SPEED is 100Mbit/s,Can it be 1Gbit/s? Thinks!

M69W, 2020-10-08:
Your system?
Your network card?
Which kexts you use?

I think of the first problem is to make it work stable and reliable. And then make it work better.
you are always welcome to join the developer team and make it better.

ErrorErrorError, 2020-10-08:
> Why is the LINK SPEED is 100Mbit/s,Can it be 1Gbit/s? Thinks!

I currently does not reflect the real speed of the wireless speed. More info here: https://github.com/OpenIntelWireless/itlwm/issues/198

williambj1, 2020-10-08:
Makes no sense, you can set it to 1Tbit/s or 1Ebit/s if you want. Doesn’t affect any performance. We decided to set it to 100Mbit/s since it’s the closest to the maximum performance.

closed 2020-10-08 by williambj1

xqf9725, 2020-10-09:
> Makes no sense, you can set it to 1Tbit/s or 1Ebit/s if you want. Doesn’t affect any performance. We decided to set it to 100Mbit/s since it’s the closest value to the current maximum performance.

Thanks a lot! How set it to 1Tbit/s or 1Ebit/s?