OpenIntelWireless/itlwm #339: Apple Watch解锁不稳定（Apple Watch Unlock function not stable)

lunjielee, 2020-10-10:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: AirportItlwm-Big_Sur-v1.2.0-DEBUG-alpha-2f302c6
- WiFi Card Model: AX201
- Product ID: 8086, 06f0
- macOS Version: 11.0 Beta (20A5384c)

**Description**
新增的apple watch解锁功能在开启后有时能正常解锁多次，之后apple watch便会在一段时间后断开与mac的连接，必须手动再次重新勾选使用apple watch解锁的选项才能再次使用apple watch解锁
The new apple watch unlocking function may work for some time after it is turned on, then the apple watch will be disconnected from the mac after a while.
_Before_
[image]
_After some time_
[image]

和DW1820A的apple watch解锁表现一致（可能无关）
Like the DW1820A's apple watch unlocking (possibly irrelevant)

**Bug Report Archive**
[Archive.zip](https://github.com/OpenIntelWireless/itlwm/files/5358874/Archive.zip)

**Kext Download Source**
https://github.com/OpenIntelWireless/itlwm/releases/download/v1.2.0-alpha/AirportItlwm-Big_Sur-v1.2.0-DEBUG-alpha-2f302c6.zip

zxystd, 2020-10-15:
emmm，没有苹果表，短时间内不会修，看看能不能玄学修复吧。

lunjielee, 2020-10-15:
> emmm，没有苹果表，短时间内不会修，看看能不能玄学修复吧。

哈哈哈，没事，不着急，感谢你的时间与付出