OpenIntelWireless/itlwm #340: AC3165 使用1.1版本，查找不到网络

asd45183, 2020-10-10:
ma c:10.15.4
cpu i7-7700
蓝牙驱动 load 后，识别不出蓝牙。。
mac@macdeMacBook-Pro driver % log show --last boot | grep AirportItlwm      
2020-10-10 23:17:13.618716+0800 0x25a      Error       0x0                  60     0    kextd: (IOKit) [com.apple.kext:kextlog] Kext id com.zxystd.AirportItlwm not found; removing personalities from kernel.
2020-10-10 23:27:06.680286+0800 0x2e28     Default     0x0                  920    0    sudo:      mac : TTY=ttys000 ; PWD=/Users/mac/Desktop/driver ; USER=root ; COMMAND=/bin/chmod -R 755 AirportItlwm.kext IntelBluetoothFirmware.kext IntelBluetoothInjector.kext itlwm.kext
2020-10-10 23:27:12.867583+0800 0x2e74     Default     0x0                  923    0    sudo:      mac : TTY=ttys000 ; PWD=/Users/mac/Desktop/driver ; USER=root ; COMMAND=/usr/sbin/chown -R root AirportItlwm.kext IntelBluetoothFirmware.kext IntelBluetoothInjector.kext itlwm.kext
2020-10-10 23:27:20.679081+0800 0x2ec1     Default     0x0                  927    0    sudo:      mac : TTY=ttys000 ; PWD=/Users/mac/Desktop/driver ; USER=root ; COMMAND=/usr/sbin/chown -R wheel AirportItlwm.kext IntelBluetoothFirmware.kext IntelBluetoothInjector.kext itlwm.kext
2020-10-10 23:27:28.289323+0800 0x2eff     Default     0x0                  930    0    sudo:      mac : TTY=ttys000 ; PWD=/Users/mac/Desktop/driver ; USER=root ; COMMAND=/usr/bin/chgrp -R wheel AirportItlwm.kext IntelBluetoothFirmware.kext IntelBluetoothInjector.kext itlwm.kext
2020-10-10 23:27:42.400950+0800 0x2f82     Default     0x0                  937    0    sudo:      mac : TTY=ttys000 ; PWD=/Users/mac/Desktop/driver ; USER=root ; COMMAND=/sbin/kextload AirportItlwm.kext
2020-10-10 23:27:42.484739+0800 0x25a      Error       0x0                  60     0    kextd: (IOKit) [com.apple.kext:kextlog] Kext with invalid signature (-67062) allowed: <OSKext 0x600003c64900 [0x7fff9030f8c0]> { URL = "file:///Users/mac/Desktop/driver/AirportItlwm.kext/", ID = "com.zxystd.AirportItlwm" }
2020-10-10 23:27:42.777759+0800 0x25a      Default     0x0                  0      0    kernel: OSMetaClass: Kext com.zxystd.AirportItlwm class IOTaskQueue is a duplicate;kext com.zxystd.itlwm already has a class by that name.
2020-10-10 23:27:42.777762+0800 0x25a      Default     0x0                  0      0    kernel: Kext com.zxystd.AirportItlwm start failed (result 0xdc00400a).
2020-10-10 23:27:42.777764+0800 0x25a      Default     0x0                  0      0    kernel: Kext com.zxystd.AirportItlwm failed to load (0xdc008017).
2020-10-10 23:27:42.777767+0800 0x25a      Default     0x0                  0      0    kernel: Failed to load kext com.zxystd.AirportItlwm (error 0xdc008017).
2020-10-10 23:27:42.781503+0800 0x25a      Error       0x0                  60     0    kextd: (IOKit) [com.apple.kext:kextlog] Failed to load /Users/mac/Desktop/driver/AirportItlwm.kext - (libkern/kext) kext (kmod) start/stop routine failed.
2020-10-10 23:27:46.934038+0800 0x2fc4     Default     0x0                  940    0    sudo:      mac : TTY=ttys000 ; PWD=/Users/mac/Desktop/driver ; USER=root ; COMMAND=/sbin/kextload AirportItlwm.kext
2020-10-10 23:27:46.987691+0800 0x25a      Error       0x0                  60     0    kextd: (IOKit) [com.apple.kext:kextlog] Kext with invalid signature (-67062) allowed: <OSKext 0x600003c5bcc0 [0x7fff9030f8c0]> { URL = "file:///Users/mac/Desktop/driver/AirportItlwm.kext/", ID = "com.zxystd.AirportItlwm" }
2020-10-10 23:27:47.120935+0800 0x25a      Default     0x0                  0      0    kernel: OSMetaClass: Kext com.zxystd.AirportItlwm class IOTaskQueue is a duplicate;kext com.zxystd.itlwm already has a class by that name.
2020-10-10 23:27:47.120939+0800 0x25a      Default     0x0                  0      0    kernel: Kext com.zxystd.AirportItlwm start failed (result 0xdc00400a).
2020-10-10 23:27:47.120941+0800 0x25a      Default     0x0                  0      0    kernel: Kext com.zxystd.AirportItlwm failed to load (0xdc008017).
2020-10-10 23:27:47.120944+0800 0x25a      Default     0x0                  0      0    kernel: Failed to load kext com.zxystd.AirportItlwm (error 0xdc008017).
2020-10-10 23:27:47.124802+0800 0x25a      Error       0x0                  60     0    kextd: (IOKit) [com.apple.kext:kextlog] Failed to load /Users/mac/Desktop/driver/AirportItlwm.kext - (libkern/kext) kext (kmod) start/stop routine failed.
2020-10-10 23:27:47.808178+0800 0x2fc9     Default     0x0                  942    0    sudo:      mac : TTY=ttys000 ; PWD=/Users/mac/Desktop/driver ; USER=root ; COMMAND=/sbin/kextload AirportItlwm.kext
2020-10-10 23:27:47.859323+0800 0x25a      Error       0x0                  60     0    kextd: (IOKit) [com.apple.kext:kextlog] Kext with invalid signature (-67062) allowed: <OSKext 0x600003c723a0 [0x7fff9030f8c0]> { URL = "file:///Users/mac/Desktop/driver/AirportItlwm.kext/", ID = "com.zxystd.AirportItlwm" }
2020-10-10 23:27:47.985975+0800 0x25a      Default     0x0                  0      0    kernel: OSMetaClass: Kext com.zxystd.AirportItlwm class IOTaskQueue is a duplicate;kext com.zxystd.itlwm already has a class by that name.
2020-10-10 23:27:47.985978+0800 0x25a      Default     0x0                  0      0    kernel: Kext com.zxystd.AirportItlwm start failed (result 0xdc00400a).
2020-10-10 23:27:47.985981+0800 0x25a      Default     0x0                  0      0    kernel: Kext com.zxystd.AirportItlwm failed to load (0xdc008017).
2020-10-10 23:27:47.985992+0800 0x25a      Default     0x0                  0      0    kernel: Failed to load kext com.zxystd.AirportItlwm (error 0xdc008017).
2020-10-10 23:27:47.989858+0800 0x25a      Error       0x0                  60     0    kextd: (IOKit) [com.apple.kext:kextlog] Failed to load /Users/mac/Desktop/driver/AirportItlwm.kext - (libkern/kext) kext (kmod) start/stop routine failed.
2020-10-10 23:27:48.608162+0800 0x2fcd     Default     0x0                  944    0    sudo:      mac : TTY=ttys000 ; PWD=/Users/mac/Desktop/driver ; USER=root ; COMMAND=/sbin/kextload AirportItlwm.kext
2020-10-10 23:27:48.661583+0800 0x25a      Error       0x0                  60     0    kextd: (IOKit) [com.apple.kext:kextlog] Kext with invalid signature (-67062) allowed: <OSKext 0x600003c5bf00 [0x7fff9030f8c0]> { URL = "file:///Users/mac/Desktop/driver/AirportItlwm.kext/", ID = "com.zxystd.AirportItlwm" }
2020-10-10 23:27:48.788817+0800 0x25a      Default     0x0                  0      0    kernel: OSMetaClass: Kext com.zxystd.AirportItlwm class IOTaskQueue is a duplicate;kext com.zxystd.itlwm already has a class by that name.
2020-10-10 23:27:48.788820+0800 0x25a      Default     0x0                  0      0    kernel: Kext com.zxystd.AirportItlwm start failed (result 0xdc00400a).
2020-10-10 23:27:48.788823+0800 0x25a      Default     0x0                  0      0    kernel: Kext com.zxystd.AirportItlwm failed to load (0xdc008017).
2020-10-10 23:27:48.788825+0800 0x25a      Default     0x0                  0      0    kernel: Failed to load kext com.zxystd.AirportItlwm (error 0xdc008017).
2020-10-10 23:27:48.792766+0800 0x25a      Error       0x0                  60     0    kextd: (IOKit) [com.apple.kext:kextlog] Failed to load /Users/mac/Desktop/driver/AirportItlwm.kext - (libkern/kext) kext (kmod) start/stop routine failed.
2020-10-10 23:27:49.493216+0800 0x2fd3     Default     0x0                  946    0    sudo:      mac : TTY=ttys000 ; PWD=/Users/mac/Desktop/driver ; USER=root ; COMMAND=/sbin/kextload AirportItlwm.kext
2020-10-10 23:27:49.545357+0800 0x25a      Error       0x0                  60     0    kextd: (IOKit) [com.apple.kext:kextlog] Kext with invalid signature (-67062) allowed: <OSKext 0x600003c46520 [0x7fff9030f8c0]> { URL = "file:///Users/mac/Desktop/driver/AirportItlwm.kext/", ID = "com.zxystd.AirportItlwm" }
2020-10-10 23:27:49.675754+0800 0x25a      Default     0x0                  0      0    kernel: OSMetaClass: Kext com.zxystd.AirportItlwm class IOTaskQueue is a duplicate;kext com.zxystd.itlwm already has a class by that name.
2020-10-10 23:27:49.675758+0800 0x25a      Default     0x0                  0      0    kernel: Kext com.zxystd.AirportItlwm start failed (result 0xdc00400a).
2020-10-10 23:27:49.675761+0800 0x25a      Default     0x0                  0      0    kernel: Kext com.zxystd.AirportItlwm failed to load (0xdc008017).
2020-10-10 23:27:49.675763+0800 0x25a      Default     0x0                  0      0    kernel: Failed to load kext com.zxystd.AirportItlwm (error 0xdc008017).
[... cut ...]
  支持的频段：	1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 36, 40, 44, 48, 52, 56, 60, 64, 100, 104, 108, 112, 116, 120, 124, 128, 132, 136, 140, 144, 149, 153, 157, 161, 165
  无线网络唤醒：	支持
  隔空投送：	支持
  “隔空投送”频段：	0
  状态：	未关联

驱动表：
WhateverGreen.kext
AirportItlwm.kext
AppleALC.kext
HoRNDIS.kext
IntelBluetoothFirmware.kext
IntelBluetoothInjector.kext
Lilu.kext
RealtekRTL8111.kext
SMCBatteryManager.kext
SMCLightSensor.kext
SMCProcessor.kext
SMCSuperIO.kext
USBInjectAll.kext
VirtualSMC.kext
VoodooHDA.kext
VoodooPS2Controller.kext

zxystd, 2020-10-15:
我也有3165，没这种情况。另外你按照issue规范上传日志文件吧。

asd45183, 2020-10-17:
@zxystd 老铁，我是来反馈的，我今天用oc重做了15.4，之后换成oc6.2搭配强制启动io80211那个设置，开机无线上的蓝牙自动识别了。还没用上你的驱动。。。下一步准备用驱动折腾下无线。有进展了再来反馈。
[image]

asd45183, 2020-10-17:
感谢，我这里可以了

closed 2020-10-17 by williambj1

JonathanSum, 2022-07-08:
一樣問題,如何解決?