OpenIntelWireless/itlwm #341: Auto disconnection from Open WiFi and unable to connect to Windows Hotspot

huhanwj, 2020-10-10:
**Have You Read Our Docs**
Yep

**Are You Reporting A Bug**
Yep

**Environment**

- Kext Version: v1.1.0 stable
- WiFi Card Model: AX201 (embedded in NUC 10, but identified as AC9462 both in Hackintool and other linux OS)
- Product ID: 8086, 02F0
- macOS Version: 10.15.7

**Description**

Auto disconnection and reconnection (looping) from my campus wifi (authentication via webpage) after establishing connection for a few seconds.
Unable to connect to a hotspot created from a windows PC (AX200 installed).
**Bug Report Archive**

[bugreport_57021.zip](https://github.com/OpenIntelWireless/itlwm/files/5359528/bugreport_57021.zip) auto disconnection
[bugreport_8019.zip](https://github.com/OpenIntelWireless/itlwm/files/5359530/bugreport_8019.zip) Windows hotspot

**Kext Download Source**

https://github.com/OpenIntelWireless/itlwm/releases/download/v1.1.0/itlwm_v1.1.0_Stable.kext.zip

williambj1, 2020-10-11:
- 1st bug report: deauth
- 2nd bug report: deauth + fw crash

CC @zxystd

williambj1, 2020-10-11:
@huhanwj Did you try AirportItlwm 1.0 & 1.2.0-alpha?

huhanwj, 2020-10-11:
> 
> 
> @huhanwj Have you tried AirportItlwm 1.0 & 1.2.0-alpha?

not yet
For Airportltlwm, I may first upgrade my oc from 0.6.0 to 0.6.2, but I'm busy these days so may try it later.
For 1.2.0-alpha, I will try today.

huhanwj, 2020-10-11:
> @huhanwj Have you tried AirportItlwm 1.0 & 1.2.0-alpha?

Updating OC seems to be much easier than what I thought, tried both and still having same issue.

zxystd, 2020-10-11:
@huhanwj 
the first disconnect problem log indicate that the wifi signal you are connected is weak and then trigger roam to another wifi, but it roam fail, so it disconnected.
the second log indicate that the driver can not associate to the AP, error code is 45(Invalid RSN information element capabilities: If WPA/RSN IE is malformed, such as incorrect length etc, you will see this code.), what's the version of Windows and how do you enable the hotspot?

huhanwj, 2020-10-11:
> @huhanwj
> the first disconnect problem log indicate that the wifi signal you are connected is weak and then trigger roam to another wifi, but it roam fail, so it disconnected.
> the second log indicate [...]

It's a bit strange for the first problem (I got around -60dbm for the RSSI), but I may try it in a different environment tomorrow. 
Are there any methods to force 2.4G connection instead of 5G? Because it will always use 5G WiFi and may have problem? (not sure)

For the second question, 20H2 ver Windows 10 Pro for workstation (19042.546), hotspot created from the setting panel and is useable for my Huawei Phone, iPad Pro and laptop (2004 Win10 Pro)

zxystd, 2020-10-11:
@huhanwj let 2.4G and 5G wifi separated will fix your problem.
ok, I will try.

huhanwj, 2020-11-02:
It's a bit strange but after I switch to AirportItlwm for a few days (version before this issue), the auto connection problem never appears. It seems that the driver now sticks to one frequency (5G in my case).
About the hotspot, it also appears to my other Wi-Fi adapters (ath9k based ones under Ubuntu 14.04), so maybe the hotspot does not support 802.11n?

williambj1, 2021-02-25:
Closed since this issue is no longer active, feel free to add more comments.

closed 2021-02-25 by williambj1