OpenIntelWireless/itlwm #343: [Airportitlwm] 1.1版本,偶尔断线重连,睡眠唤醒后,几率变大

CodeApePro, 2020-10-12:
硬件为nuc8i5beh,1.1版本,oc 0.6.1 偶尔断线重连,睡眠唤醒后,几率变大,日志如下.辛苦.
[itlwm_log.txt](https://github.com/OpenIntelWireless/itlwm/files/5363402/itlwm_log.txt)

CodeApePro, 2020-10-15:
补充:
网卡型号:Intel® Wireless-AC 9560 + Bluetooth 5.0
系统型号:10.15.7
oc版本:0.6.1
驱动版本:Airportitlwm 1.1

CodeApePro, 2020-10-15:
@zxystd

CodeApePro, 2020-10-15:
谢谢.辛苦.

zxystd, 2020-10-15:
换个路由器吧。。。

closed 2020-10-15 by williambj1

zxystd, 2020-10-15:
@CodeApePro 对了，你试试1.2.0 alpha版本，可能对你的现在的路由环境会好一点。

CodeApePro, 2020-10-15:
收到,谢谢,麻烦了.我试试1.2.0.这个产生的原因是什么呢? @zxystd

CodeApePro, 2020-10-15:
另外使用OC引导,更新驱动版本之后需要重建缓存和清理nave吗?新人不是很懂,麻烦您也帮忙解答下了.

williambj1, 2020-10-15:
> 另外使用OC引导,更新驱动版本之后需要重建缓存和清理nave吗?新人不是很懂,麻烦您也帮忙解答下了.

如果用引导注入就不需要

CodeApePro, 2020-10-15:
好的,谢谢您解答,我换1.2.0测试版试试. @williambj1

CodeApePro, 2020-10-15:
试用了1.2.0测试版驱动,还是会偶尔的时候断开重连.麻烦请教下这个问题出在哪里呢?我这边是办公网环境,路由器哪些设置影响了网络连接吗?麻烦了. @zxystd @williambj1

M69W, 2020-10-15:
如果条件允许，不妨把【网络】界面【左边的网络列表】删除（-），然后再新建（+）连接试试？

CodeApePro, 2020-10-15:
> 如果条件允许，不妨把【网络】界面【左边的网络列表】删除（-），然后再新建（+）连接试试？

好的.谢谢.我照着试试.

CodeApePro, 2020-10-15:
> 如果条件允许，不妨把【网络】界面【左边的网络列表】删除（-），然后再新建（+）连接试试？

一样的,刚才断了,并不算频繁,但是每天总要出个四五次.尤其是睡眠刚唤醒的时候.O(∩_∩)O

M69W, 2020-10-15:
睡眠断连，
我那边（ Airportitlwm1.2.0 alpha+ AX201 + OC0.6.2+ 11beta4）
有时会断，不是必断，
有时开机(睡眠)也能自动连上，
下次断再发日志

williambj1, 2020-10-15:
路由器拒绝了好多包，换个试试？

CodeApePro, 2020-10-15:
> 路由器拒绝了好多包，换个试试？

好的,公司办公网,路由器是换不了了,我试试用手机热点wifi之类的吧.这个wifi是2.4和5G混合的,甚至有可能是多个网混合的,因为一大片办公区都在用这个wifi.可能因为这个吧.