OpenIntelWireless/itlwm #344: [AirportItlwm]MojaveV1.2.0有反应，但打不开Wi-Fi，Mojave使用有什么注意事项吗？

Deng-Xian-Sheng, 2020-10-12:
您阅读我们的文件了吗： 阅读了

您正在报告错误吗：是的

环境：OpenCore6.1

Kext版本：AirportItlwm-Mojave-v1.2.0-DEBUG-alpha-f809ee4
WiFi卡型号：AC9565
macOS版本：10.14.6
描述:Wi-Fi在系统设置--网络中有显示，但打不开Wi-Fi也无法扫描热点，检查过安装问题，对Mojace安装上有什么不同吗？
Kext下载源：您的github

williambj1, 2020-10-12:
Logs?

Deng-Xian-Sheng, 2020-10-12:
WiFi doesn't work properly, and there is no button to generate logs. I don't know how to extract logs. It seems that there are not many users on 10.14.6.

williambj1, 2020-10-12:
https://docs.oiw.workers.dev/itlwm/Troubleshooting.html#getting-kext-logs-for-runtime-issues

Please follow those instructions.

didi07, 2020-10-12:
hello friends I have installed Intel Bluetooth Firmware & itlwm & IntelBluetoothInjector.kext
the wifi works very well but not the bluetooth please help me

mehmetyukcell, 2020-10-12:
> hello friends I have installed Intel Bluetooth Firmware & itlwm & IntelBluetoothInjector.kext
> 
> the wifi works very well but not the bluetooth please help me

Remap USB ports with Hackintool

zxystd, 2020-10-13:
try the latest one, it may works but only on open wifi. will fix soon.

nlg96, 2020-10-13:
Beta Catania can work on Mojave

didi07, 2020-10-14:
the wifi works big beast but not the bluetooth

didi07, 2020-10-14:
yes I have installed the latest kext

zxystd, 2020-10-15:
@Deng-Xian-Sheng 1.2.0 alpha测试结果如何？

closed 2020-10-18 by zxystd