OpenIntelWireless/itlwm #345: itlwm disconnected every a few minutes

dengdaibian, 2020-10-12:
**Have You Read Our Docs**
yes

**Are You Reporting A Bug**
Yes 

**Environment**

- Kext Version: Itlwm 1.2.0
- WiFi Card Model: ax200
- Product ID: 0x2723
- macOS Version: 10.15.6

**Description**

wifi always disconnectd every a few minutes
i have to off and on it manually to connect to the wifi again.

**Bug Report Archive**

[bugreport_6147.zip](https://github.com/OpenIntelWireless/itlwm/files/5365579/bugreport_6147.zip)

**Kext Download Source**

from official OpenIntelWireless github

williambj1, 2020-10-13:
The logs provided can't describe your issue. Try to capture the logs after it disconnects.

dengdaibian, 2020-10-13:
but the logs are captured after it disconnectd and reconnected

dengdaibian, 2020-10-13:
I'll try again tonight,thank you

dengdaibian, 2020-10-13:
> The logs provided can't describe your issue. Try to capture the logs after it disconnects.

But the logs were captured after wifi disconnected and reconnected,i will try again tonight
thank you

zxystd, 2020-10-13:
According to the log, the wifi disconnect by you manually click "close", the driver seems works fine, maybe is your wifi network environment related issue? thanks.

closed 2020-10-13 by dengdaibian