OpenIntelWireless/itlwm #350: WiFi nor Bluetooth working with open core (new to OC, please be nice)

huwusky, 2020-10-15:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes, I am reporting an issue, potentially user error as I am new to OpenCore

**Environment**

- Kext Version: 1.1.0
- WiFi Card Model: Intel Dual Band Wireless 3165
- Product ID: 0x3165
- macOS Version: 10.15.7

**Description**
Upon booting with AirportItlwm in /EFI/OC/Kexts, I lose Bluetooth connectivity, WiFi also doesn't work upon doing this also (I'm really not sure how to force a kext to load with opencore as that may be the issue, I am very very new to this, just converted to opencore today after being a long time clover user).

**Bug Report Archive**
[Log_2020-10-15_07-27-12.log](https://github.com/OpenIntelWireless/itlwm/files/5384436/Log_2020-10-15_07-27-12.log)

**Kext Download Source**
Github Releases (1.1.0) 
[AirportItlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/5384451/AirportItlwm.kext.zip)

huwusky, 2020-10-15:
After clearing NVRAM I have regained BT Connectivity, still no clue how to go about having wifi working

williambj1, 2020-10-15:
I'm sorry but GitHub Issue is a place to trace software bugs. You should consider going to other places such as forums, Reddit or, Discord for personal help.

closed 2020-10-15 by williambj1