OpenIntelWireless/itlwm #351: Airportitlwm exclamation mark, like before

Silfalion, 2020-10-15:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 1.2.0
- WiFi Card Model:  intel 7260
- Product ID: 
- macOS Version: 10.15.6

**Description**
Like earlier versions of itlwm, it still returns an exclamation mark on my network wall plug wifi (works fine on router), tried changing the packet frequency, the password, all the possible parameters on the AP, reseting Macos network files, still nothing.

**Bug Report Archive**
[AirportItlwm.log](https://github.com/OpenIntelWireless/itlwm/files/5385051/AirportItlwm.log)

**Kext Download Source**
Release page, alpha Airportitlwm 1.2.0

zxystd, 2020-10-16:
So I can confirm it is a router compatible problem(relate to DHCP), because in version 1.2.0, we have change the Authentication component to Apple's native RSNSupplicant, And I can see it works fine according to the log.

Silfalion, 2020-10-16:
I tried so more. It works now, it could be added to the troubleshooting section. Delete the files doesn't suffice, the bin has to be emptied and in my case I delete all the files beside the boot one.

I tried this before but it didn't work, so it may be the change you mentioned. I'm getting around 1.6 Mb, just to be sure, is this the normal value?

williambj1, 2020-10-16:
> I tried so more. It works now, it could be added to the troubleshooting section. Delete the files doesn't suffice, the bin has to be emptied and in my case I delete all the files beside the boot [...]

Sorry but I really don't understand your comment which contains no subject at all.

williambj1, 2020-10-16:
Assume user mistake solved.

closed 2020-10-16 by williambj1

Silfalion, 2020-10-16:
It wasn't a user mistake. There is no specification of emptying the bin on the troubleshoot page of itlwm nor is there mention of other files to delete to reset the network settings. Doing both of those actions got the package working without doing that there was no connection. 

Please ask questions before closing, another person who similarly took a long time(3 months) to get the package working may benefit from this. 

For anyone who has this problem:  

1.delete all the ports on the network manager in settings

2.Open the system configuration file as instructed in the github help page.

3.Delete all the files except the one ending in boot. Bare in mind this will also delete all your saved network preferences(proxy, dns...).

4.VERY IMPORTANT: empty the macos bin.

5.Reboot and test, turn off and on the wifi if you can't see the network you are looking for or it won't connect.

williambj1, 2020-10-16:
https://openintelwireless.github.io/itlwm/Troubleshooting.html#resetting-system-preferences-network

TBH, the real problem based on your description is you did not empty the bin, which is clearly a user mistake.
All of the procedures you mentioned above are all present in our docs; userspace issues are not bugs of the kext itself as well.
In fact, emptying the bin makes no sense, please do some research about how trash works in macOS before commenting.

Silfalion, 2020-10-16:
I'd redirect the advice to you. Macos Bin keeps a placeholder for the deleted files that'd be the reason the Macos "remembers" details of your network even after restart and deletion. I did, as mentioned before, empty the bin on numerous occasions on the previous versions and it did nothing, I also emptied it with this version and it did nothing. 

It wasn't until I deleted other files not mentioned in the troubleshooting page AND emptied the bin that I solved the problem. If you're not going to take the time to understand my words, I'd advice you re read at a your leisure time or you simply ask for clarifications. I'm not trying to pick up a fight here but trying to pin on me anger or frustration build up from previous potentially misdirected questions is really intolerable.

Silfalion, 2020-10-16:
Don't get me wrong, I'm very thankful for all you did. This is awesome and tiring work. Just please don't try to generalize.

williambj1, 2020-10-16:
Glad that it's now working for you, have a nice day and enjoy.

Silfalion, 2020-10-16:
Thank you, very nice day to you too.

stevezhengshiqi, 2020-10-16:
> It wasn't until I deleted other files not mentioned in the troubleshooting page AND emptied the bin that I solved the problem. If you're not going to take the time to understand my words, I'd advice you re read at a your leisure time or you simply ask for clarifications. 

Apparently,  https://openintelwireless.github.io/itlwm/Troubleshooting.html#resetting-system-preferences-network has included everything you need to delete. 
[image]

Just follow the instruction line by line and you will be good. The developers are tired of getting around with all these user-side misbehaviors. What you suggest "Delete all the files except the one ending in boot" will surely break other account settings since com.apple.accounts.exists.plist and so on are unrelated to network settings but important to icloud account.

Silfalion, 2020-10-16:
Look. If I hadn't already read line by line the help page, each time a new versions comes out for the passed 3 months I wouldn't bother the developers. I relate to their anger about endless, time wasted asking. I get unnerved when I read it on the gitter page too. This is a new issue and that's why I bothered to mention it. 

Thank for the file deletion problem, I didn't know since I didn't look it up that much.