OpenIntelWireless/itlwm #352: [AirportItlwm][Bug Report] "Wi-Fi Recommendations" shows wrongly

EdenJohnson2006, 2020-10-15:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 1.2.0Alpha
- WiFi Card Model: 8265AC
- Product ID: 0x24FD
- macOS Version: macOS 10.15.7

**Description**
(The bug appears intermittently)
When connecting to a Wi-Fi network, "Wi-Fi Recommendations" will appear in the Wi-Fi menu, which is to change Wi-Fi to broadcast SSID, but the connected network is already a broadcast network
[image]
[image]
[image]

**Bug Report Archive**
[AirportItlwm-Log-2020-10-15-22-52.txt](https://github.com/OpenIntelWireless/itlwm/files/5385769/AirportItlwm-Log-2020-10-15-22-52.txt)

**Kext Download Source**
Action Release 1.2.0 Alpha Catalina [Here](https://github.com/OpenIntelWireless/itlwm/releases/tag/v1.2.0-alpha)

Stevemoretz, 2020-11-08:
For me it says you're using hidden SSID, I was using hidden SSID but since it didn't work with this kext I changed it to exposed SSID and now it works but it says under wifi recommendation that you're using hidden SSID please change it for blah blah blah.

Overc1ocker, 2022-02-21:
As of the 2.1.0 release, this seems to no longer be an issue. I dont know where it got fixed.
Monterey 12.2, AC7265 rev B

Eden7Ba23, 2022-02-22:
> As of the 2.1.0 release, this seems to no longer be an issue. I dont know where it got fixed. Monterey 12.2, AC7265 rev B

im sorry about that. i no longer use macOS so i dont know how it be now.