OpenIntelWireless/itlwm #353: Slow boot caused by the Kext (a lot of Verbose releated)

EndLineTech03, 2020-10-15:
I'm testing Airportltlwm_v1.0_Beta_Catalina on my desktop Hackintosh running macOS Catalina 10.15.7. My wifi card is Intel Wireless AC 9260. I don't know if it's normal or if it's possible to improve something, but I realized that this Build increases a lot the boot time (60/70 %). Can you work on this problem? Or, am I using the Kext incorrectly?

seungjunProgramming, 2020-10-16:
Me too...
It takes about 20~30secs to include itlwmAirport + intelBluetooth Firmware
(Both downloaded at your GitHub)

igorkulman, 2020-10-16:
I solved this by building both Airportitlwm and IntelBluetoothFirmware myself, removing firmware for all the cards except mine first. This results in much smaller kexts that load faster.

williambj1, 2020-10-16:
Boot time increase should be expected but I really doubt your speculation. Because you opened this issue in Itlwm's repo, I assume you are talking about itlwm instead of IntelBluetoothFirmware.

In fact, the increase of boot time is more likely related to IO80211 wanting the system to be already connected to the Internet before the system enters userspace, based on my observation. You could try itlwm without Wi-Fi config in its plist and see if there's any difference.

Booting macOS requires reading quite a lot of data on the hard drive, injecting a 10MB Kext into the system is really not a big deal. What's more, using PCI DMA, uploading Wi-Fi firmware into the card is almost instantaneous. I've tried simplifying the kext before and it really made no difference.

Bluetooth is another story, sometimes it takes a significant amount of time to wait for the card's USB part to respond. Transfering firmware through USB is not as fast as through PCI as well.

EndLineTech03, 2020-10-16:
Thanks for the responses. I will try also to debug the Kext, build it manually and I will try the other Kext.

EndLineTech03, 2020-10-16:
> Boot time increase should be expected but I really doubt your speculation. Because you opened this issue in Itlwm's repo, I assume you are talking about itlwm instead of IntelBluetoothFirmware.
> 
> 
> 
> In fact, the increase of boot time is more likely related to IO80211 wanting the system to be already connected to the Internet before the system enters userspace, based on my observation. You could try itlwm without Wi-Fi config in its plist and see if there's any difference.
> 
> 
> 
> Verbose logs are essential ATM since itlwm is a kext that runs all the time. If people experienced a random runtime issue, they can quickly pull out the logs and submit them to us. However, it would be a nightmare to capture those random bugs not to mention wasting a lot of our time dealing with newbies if logs are removed.
> 
> 
> 
> Booting macOS requires reading quite a lot of data on the hard drive, injecting a 10MB Kext into the system is really not a big deal. What's more, using PCI DMA, uploading Wi-Fi firmware into the card is almost instantaneous. I've tried simplifying the kext before and it really made no difference.
> 
> 
> 
> Bluetooth is another story, sometimes it takes a significant amount of time to wait for the card's USB part to respond. Uploading firmware through USB is not as fast as through PCI as well.

What changes between Airportltlwm and itlwm?

seungjunProgramming, 2020-10-16:
@williambj1 

> Boot time increase should be expected but I really doubt your speculation. Because you opened this issue in Itlwm's repo, I assume you are talking about itlwm instead of IntelBluetoothFirmware.
>
> In [...]

Then can you please make the kext without verbose logs please?
You can publish it together with the original version(with the verbose log) for newbies or etc to use.
Some users like me need those options...
(Or maybe a boot flag (ex: itlwmnoverbose=1) option)

EndLineTech03, 2020-10-16:
> @williambj1
>
>
>
>
>
>
>
>
>
>
>
>
>
>
> Then can you please make the kext without verbose logs please?
>
> You can publish it together with the original version(with the verbose log) for newbies or etc to use. [...]

I hope they will do this. Because it's horrible to see all the logs when I boot in verbose mode. A non-debug version would be better.

williambj1, 2020-10-16:
@seungjunProgramming @EndLineTech03 No we won’t, and I’ve already explained why. If you really hate those logs, you can delete them and build a custom version.

seungjunProgramming, 2020-10-17:
@williambj1 How can I delete the verbose log??? Can you tell me the command which prints the verbose???
(Ex: in C printing in kernel is printf)

EndLineTech03, 2020-10-17:
> @seungjunProgramming @EndLineTech03 No we won’t, and I’ve already explained why. If you really hate those logs, you can delete them and build a custom version.

Ok thanks. But I can't find the command related to the logs in the source code. And I can find only the source code of itlwm and not of AirportItlwm.

LongCX, 2020-10-18:
> I solved this by building both Airportitlwm and IntelBluetoothFirmware myself, removing firmware for all the cards except mine first. This results in much smaller kexts that load faster.

Please share me kext affter build

EndLineTech03, 2020-10-19:
Where is AirportItlwm source code???

phu54321, 2020-10-23:
Verbose log doesnt really affect boot time. This is a hidden bootarg of macOS Darwin Kernel (not related to itlwn), but try adding `pci=1` to boot-args. A lot of new texts will appear, but it wont affect the boot time too much.

racka98, 2020-10-23:
> I solved this by building both Airportitlwm and IntelBluetoothFirmware myself, removing firmware for all the cards except mine first. This results in much smaller kexts that load faster.

How did you go about this?? I tried removing the firmware and build failed showing a bunch of errors and missing files

johnwtt, 2020-11-07:
> I solved this by building both Airportitlwm and IntelBluetoothFirmware myself, removing firmware for all the cards except mine first. This results in much smaller kexts that load faster.

Can you teach me how to do that ?

closed 2020-11-07 by williambj1

racka98, 2020-11-14:
> > I solved this by building both Airportitlwm and IntelBluetoothFirmware myself, removing firmware for all the cards except mine first. This results in much smaller kexts that load faster.
> 
> Can you teach me how to do that ?

I finally got it to work. 
1. Clone the repo: `git clone https://github.com/OpenIntelWireless/itlwm.git`

2. Open the folder where it's cloned to

3. Open Xcode, press File -> New -> File.. on the Search bar/Filter type `shell` and choose to create a new shell script file

4. Copy this code below into it;

```shell
#!/bin/bash

# remove all local changes
git reset --hard HEAD
rm -rf build

# pull latest code
git pull

# remove generated firmware
rm include/FwBinary.cpp

# remove firmware for other wifi cards - DELETE OR CHANGE TO YOUR CARD
find itlwm/firmware/ -type f ! -name 'iwm-7265-*' -delete

# generate firmware
xcodebuild -project itlwm.xcodeproj -target fw_gen -configuration Release -sdk macosx

# build the kexts
## 1. itlwm.kext
xcodebuild -project itlwm.xcodeproj -target itlwm -configuration Release -sdk macosx

## 2. AirportItlwm Catalina
xcodebuild -project itlwm.xcodeproj -target AirportItlwm-Mojave -configuration Release -sdk macosx

## 3. AirportItlwm Catalina
xcodebuild -project itlwm.xcodeproj -target AirportItlwm-Catalina -configuration Release -sdk macosx

## 4. AirportItlwm Big Sur
xcodebuild -project itlwm.xcodeproj -target AirportItlwm-Big\ Sur -configuration Release -sdk macosx

# Location of Kexts
echo "You kexts are in build/Release!!"
echo " "
```

5. Change line 14: `find itlwm/firmware/ -type f ! -name 'iwm-7265-*' -delete` change `iwm-7265` to your firmware name and save the file.

6. Place the file in the root directory of the cloned itlwm folder.

7. Clone MacKernelSDK `git clone https://github.com/acidanthera/MacKernelSDK.git` and place it's folder inside itlwm folder

8. Run the script with sh command.
Ex: `sh script-name.sh` where 'script-name' is the name of the shell script you made.

Done, you'll find your kexts under build/Release

EndLineTech03, 2020-11-14:
> I finally got it to work.
>
> 1. Clone the repo: `git clone https://github.com/OpenIntelWireless/itlwm.git`
>
>
>
> 2. Open the folder where it's cloned to
>
>
>
> 3. Open Xcode, press File -> New -> File.. on [...]

Thanks a lot!! Is it possible also to remove boot logs?

racka98, 2020-11-14:
> Thanks a lot!! Is it possible also to remove boot logs?

As william said. It won't change anything. Every kexts has log, some are hidden but they are still there running. Disabling them won't change anything. Further delay may be caused by the network controller itself

mhl221135, 2020-11-30:
> > > I solved this by building both Airportitlwm and IntelBluetoothFirmware myself, removing firmware for all the cards except mine first. This results in much smaller kexts that load faster.
> > 
> > Can you teach me how to do that ?
> 
> I finally got it to work. 
> 1. Clone the repo: `git clone https://github.com/OpenIntelWireless/itlwm.git`
> 
> 2. Open the folder where it's cloned to
> 
> 3. Open Xcode, press File -> New -> File.. on the Search bar/Filter type `shell` and choose to create a new shell script file
> 
> 4. Copy this code below into it;
> 
> ```shell
> #!/bin/bash
> 
> # remove all local changes
> git reset --hard HEAD
> rm -rf build
> 
> # pull latest code
> git pull
> 
> # remove generated firmware
> rm include/FwBinary.cpp
> 
> # remove firmware for other wifi cards - DELETE OR CHANGE TO YOUR CARD
> find itlwm/firmware/ -type f ! -name 'iwm-7265-*' -delete
> 
> 
> # generate firmware
> xcodebuild -project itlwm.xcodeproj -target fw_gen -configuration Release -sdk macosx
> 
> # build the kexts
> ## 1. itlwm.kext
> xcodebuild -project itlwm.xcodeproj -target itlwm -configuration Release -sdk macosx
> 
> ## 2. AirportItlwm Mojave
> xcodebuild -project itlwm.xcodeproj -target AirportItlwm-Mojave -configuration Release -sdk macosx
> 
> ## 3. AirportItlwm Catalina
> xcodebuild -project itlwm.xcodeproj -target AirportItlwm-Catalina -configuration Release -sdk macosx
> 
> ## 4. AirportItlwm Big Sur
> xcodebuild -project itlwm.xcodeproj -target AirportItlwm-Big\ Sur -configuration Release -sdk macosx
> 
> # Location of Kexts
> echo "You kexts are in build/Release!!"
> echo " "
> ```
> 
> 5. Change line 14: `find itlwm/firmware/ -type f ! -name 'iwm-7265-*' -delete` change `iwm-7265` to your firmware name and save the file.
> 
> 6. Place the file in the root directory of the cloned itlwm folder.
> 
> 7. Clone MacKernelSDK `git clone https://github.com/acidanthera/MacKernelSDK.git` and place it's folder inside itlwm folder
> 
> 8. Run the script with sh command.
> Ex: `sh script-name.sh` where 'script-name' is the name of the shell script you made.
> 
> Done, you'll find your kexts under build/Release

Thank yoy for your manual, it is work for me =) Maybe you also know how to remove other firmwares for bluetooth firmware kext ?

racka98, 2020-12-03:
> > > > I solved this by building both Airportitlwm and IntelBluetoothFirmware myself, removing firmware for all the cards except mine first. This results in much smaller kexts that load faster.
> > > 
> > > 
> > > Can you teach me how to do that ?
> > 
> > 
> > I finally got it to work.
> > 
> > 1. Clone the repo: `git clone https://github.com/OpenIntelWireless/itlwm.git`
> > 2. Open the folder where it's cloned to
> > 3. Open Xcode, press File -> New -> File.. on the Search bar/Filter type `shell` and choose to create a new shell script file
> > 4. Copy this code below into it;
> > 
> > ```shell
> > #!/bin/bash
> > 
> > # remove all local changes
> > git reset --hard HEAD
> > rm -rf build
> > 
> > # pull latest code
> > git pull
> > 
> > # remove generated firmware
> > rm include/FwBinary.cpp
> > 
> > # remove firmware for other wifi cards - DELETE OR CHANGE TO YOUR CARD
> > find itlwm/firmware/ -type f ! -name 'iwm-7265-*' -delete
> > 
> > 
> > # generate firmware
> > xcodebuild -project itlwm.xcodeproj -target fw_gen -configuration Release -sdk macosx
> > 
> > # build the kexts
> > ## 1. itlwm.kext
> > xcodebuild -project itlwm.xcodeproj -target itlwm -configuration Release -sdk macosx
> > 
> > ## 2. AirportItlwm Mojave
> > xcodebuild -project itlwm.xcodeproj -target AirportItlwm-Mojave -configuration Release -sdk macosx
> > 
> > ## 3. AirportItlwm Catalina
> > xcodebuild -project itlwm.xcodeproj -target AirportItlwm-Catalina -configuration Release -sdk macosx
> > 
> > ## 4. AirportItlwm Big Sur
> > xcodebuild -project itlwm.xcodeproj -target AirportItlwm-Big\ Sur -configuration Release -sdk macosx
> > 
> > # Location of Kexts
> > echo "You kexts are in build/Release!!"
> > echo " "
> > ```
> > 
> > 
> > 
> > 1. Change line 14: `find itlwm/firmware/ -type f ! -name 'iwm-7265-*' -delete` change `iwm-7265` to your firmware name and save the file.
> > 2. Place the file in the root directory of the cloned itlwm folder.
> > 3. Clone MacKernelSDK `git clone https://github.com/acidanthera/MacKernelSDK.git` and place it's folder inside itlwm folder
> > 4. Run the script with sh command.
> >    Ex: `sh script-name.sh` where 'script-name' is the name of the shell script you made.
> > 
> > Done, you'll find your kexts under build/Release
> 
> Thank yoy for your manual, it is work for me =) Maybe you also know how to remove other firmwares for bluetooth firmware kext ?

The problem is for the bluetooth firmware the names aren't very specific. You need to know exactly what the firmware of you card is and they are not prefixed by your card name like the wifi ones.

williambj1, 2020-12-03:
The firmware name is present in IOReg.

racka98, 2020-12-03:
> The firmware name is present in IOReg.

Oh yeah. I see it now

cristoiroberto, 2021-01-12:
Which firmware shall I keep for intel ax201? Thanks

racka98, 2021-01-12:
> Which firmware shall I keep for intel ax201? Thanks

> The firmware name is present in IOReg.

cristoiroberto, 2021-01-12:
> > Which firmware shall I keep for intel ax201? Thanks
> 
> > The firmware name is present in IOReg.

Where? I can't find IOReg.

racka98, 2021-01-12:
> > > Which firmware shall I keep for intel ax201? Thanks
> > 
> > > The firmware name is present in IOReg.
> 
> Where? I can't find IOReg.
Here:
https://github.com/khronokernel/IORegistryClone/blob/master/ioreg-210.zip

cristoiroberto, 2021-01-12:
> > > > Which firmware shall I keep for intel ax201? Thanks
> > > 
> > > 
> > > > The firmware name is present in IOReg.
> > 
> > 
> > Where? I can't find IOReg.
> 
> Here:
> https://github.com/khronokernel/IORegistryClone/blob/master/ioreg-210.zip

Could you please point me exactly to where is the firmware name? Thanks

racka98, 2021-01-12:
> > > > > Which firmware shall I keep for intel ax201? Thanks
> > > > 
> > > > 
> > > > > The firmware name is present in IOReg.
> > > 
> > > 
> > > Where? I can't find IOReg.
> > 
> > 
> > Here:
> > https://github.com/khronokernel/IORegistryClone/blob/master/ioreg-210.zip
> 
> Could you please point me exactly to where is the firmware name? Thanks

Open the IORegistryExplorer I linked, on the search bar at the top right type bluetooth then click on intelbluetoothfirmware and you can see fw_name there
[image]

cristoiroberto, 2021-01-12:
@racka98 Thank you, you pointed me to bluetooth firmware. I want for wifi card to find which firmware to keep(build airportitlwm)

racka98, 2021-01-12:
> @racka98 Thank you, you pointed me to bluetooth firmware. I want for wifi card to find which firmware to keep(build airportitlwm)

Search for "airportitlwm" or "itlwm" if you are using itlwm.kext then click on it. Look for IOModel you should see the name
[image]

cristoiroberto, 2021-01-12:
@racka98 

I can see "Intel Wireless Card" at IoModel.

cristoiroberto, 2021-01-14:
@racka98  Could you please provide the script you used to build Bluetooth kext?

aspenluxxxy, 2021-03-15:
@racka98 your method of using IOReg only works with AirportItlwm, not itlwm it seems.