OpenIntelWireless/itlwm #359: [AirportItlwm] doesn't connect

alessiomatricardi, 2020-10-17:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 1.2.0 7ae2ecd
- WiFi Card Model: AC7265
- Product ID: 0x095A
- macOS Version: 10.15.7

**Description**
Now driver doesn't connect to any WiFi network. If i click on icon, no Wifi networks are provided.

**Bug Report Archive**
[logs.zip](https://github.com/OpenIntelWireless/itlwm/files/5396063/logs.zip)

**Kext Download Source**
github actions last commit

alessiomatricardi, 2020-10-17:
Aditional informations: I'm back to 324e157 commit and now everything works

williambj1, 2020-10-17:
FW crashed.

CC @zxystd

chriseg93, 2020-10-18:
I confirm, the latest debug version crashed AirportItlwm

zxystd, 2020-10-18:
@alessiomatricardi have you tried click off and then click on?

alessiomatricardi, 2020-10-18:
@zxystd yes, i tried 1/2 times to switch off/on but no networks appear. should be saved also into logs

rediffusion, 2020-10-19:
Hello beautiful people ⭐ can someone share with me the `invite code` for the [PC Beta](http://www.pcbeta.com/) forum.
I'll be appreciated❗

[image]

_...sorry for offtopic._

williambj1, 2020-10-21:
Confirmed, bug fix WIP.

zxystd, 2020-10-21:
fix in 204c965, please help to confirm. thanks.

alessiomatricardi, 2020-10-21:
@zxystd fixed. I'll close the issue. Thank you 💯 

[logs.zip](https://github.com/OpenIntelWireless/itlwm/files/5417084/logs.zip)

closed 2020-10-21 by alessiomatricardi