OpenIntelWireless/itlwm #360: Itlwm is not running on Mojave_6 Clover_5102

Omanuar, 2020-10-19:
**Have You Read Our Docs**
Y
**Are You Reporting A Bug**
N

**Environment**

- Kext Version: 1.1.0
- WiFi Card Model: AC9560
- Product ID: A370
- macOS Version: 10.14.6
-Clover 2.5k_r5102-UEFI
**Description**
1.I put itlwm(1.1.0) to Efi/clover/kext/other.
2.heliport to apps/ & autouboot.
3.booting...heliport=tilwm is not running / kextsate | grep -i "itlwm"=nothing.
4.Is there something to add to config.plist?

**Bug Report Archive**
[bugreport_1629.zip](https://github.com/OpenIntelWireless/itlwm/files/5399730/bugreport_1629.zip)

**Kext Download Source**
Download here

Basstorm, 2020-10-19:
you should load it as a IO80211Family plugin

zxystd, 2020-10-19:
just only put the itlwm.kext into kext folder and then reboot, if it don't works, maybe you should try to debug your EFI and check the BIOS setting, detail information please go to gitter for help. Mojave is confirmed to work. thanks.

closed 2020-10-19 by zxystd