OpenIntelWireless/itlwm #361: No Networks Found (but wifi hardware / bluetooth detected.

tlefko, 2020-10-20:
Hey guys, I tried to install this using clover forcing IO802 to load from the System, as well as tried placing in the clover folder itself along with airportitwlm. itlwm works well, but when trying to use the airportitwlm it is unable to find any networks. I'm not really sure what I'm doing wrong, below I've attached a zip (with overkill) all the system information you may need to help diagnose the problem. I'm not really familiar with OpenCore and struggled deeply to get it setup so that's why I'm on clover. I currently have the Matebook X pro 2018 model. Regardless, I'll upload the files (sorry if format isn't good)

[Archive.zip](https://github.com/OpenIntelWireless/itlwm/files/5406336/Archive.zip)
[image]
[image]

mehmetyukcell, 2020-10-20:
Delete IO80211family, again tried. Also, what is your wifi model?

tlefko, 2020-10-20:
> Delete IO80211family, again tried. Also, what is your wifi model?

Deleted it, retried but no luck. Same result with it removed. The model is the Intel ac 8265 @mehmetyukcell

williambj1, 2020-10-20:
@tlefko Force load IO80211Family, and ask about the details in Reddit or Insanlymac.

Sorry but I must close this issue since this is not a bug report of our product.

closed 2020-10-20 by williambj1

mehmetyukcell, 2020-10-20:
İf You using opencore, should be show the kext with opencore configurator in the kernel kexts

tlefko, 2020-10-20:
> @tlefko Force load IO80211Family, and ask about the details in Reddit or Insanelymac.
> 
> Sorry but I must close this issue since this is not a bug report of our product.

hey, thanks for that I appreciate you going out of your way for the suggestion. I'll take that advice. Awesome work guys