OpenIntelWireless/itlwm #363: Airport Itlwm 切换网络时不会自动进行 DHCP 续租

the-eric-kwok, 2020-10-21:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: v1.2.0-7ae2ecd
- WiFi Card Model: 8260AC
- Product ID: pci8086,24f3
- macOS Version: Big Sur

**Description**
我有两个常用的Wi-Fi，一个是宿舍Wi-Fi，一个是校园网Wi-Fi。宿舍Wi-Fi的IP网段为 192.168.1.\*，校园网Wi-Fi的网段为10.34.\*.\*，因此我在两个Wi-Fi之间切换的时候需要在设置里面手动点击“DHCP续租”才能联网。

**Bug Report Archive**

**Kext Download Source**
GitHub Release page

williambj1, 2020-10-21:
有切换网络时的日志吗

the-eric-kwok, 2020-10-21:
切换网络发生在20:11
[AirportItlwm.log.zip](https://github.com/OpenIntelWireless/itlwm/files/5415736/AirportItlwm.log.zip)

zxystd, 2020-10-21:
> 切换网络发生在20:11
> [AirportItlwm.log.zip](https://github.com/OpenIntelWireless/itlwm/files/5415736/AirportItlwm.log.zip)

日志不全，应该包含itlwm的日志

the-eric-kwok, 2020-10-21:
[归档.zip](https://github.com/OpenIntelWireless/itlwm/files/5416418/default.zip)
这是新的日志，切换发生在21:57左右

zxystd, 2020-10-22:
日志是正常的，试用一下最新版？之前的系统版本会有这个问题吗，还是beta10才有？

the-eric-kwok, 2020-10-22:
尝试了 74c2cdf 的自动编译kext，这次甚至无法获取IP地址了
[归档.zip](https://github.com/OpenIntelWireless/itlwm/files/5423144/default.zip)

更新：重连WiFi后似乎可以获取IP地址了，我试试切换网络
[归档 2.zip](https://github.com/OpenIntelWireless/itlwm/files/5423149/2.zip)

依然无法自动DHCP
[归档.zip](https://github.com/OpenIntelWireless/itlwm/files/5423160/default.zip)

zxystd, 2020-10-22:
以前版本的系统会不会，你从哪个版本的系统开始用这个驱动的？

lsvih, 2020-10-23:
同样有这个问题，之前在 Catalina 配 itlwmx 没出现过，后来在 BigSur b5 到 b10 用 airportitlwm 都会偶尔出现需要手动续租 DHCP 的情况，最新的 alpha 同样会出现这种情况。

网卡：9560 二代 02f0

the-eric-kwok, 2020-10-23:
> 以前版本的系统会不会，你从哪个版本的系统开始用这个驱动的？

Catalina 不会，当时也是用的 AirportItlwm，后来直接升级到Big Sur b9 似乎就出现了这个问题。

刚刚回退到了 324e157 依然有这个问题

the-eric-kwok, 2020-10-23:
回退到[v1.1.0](https://github.com/OpenIntelWireless/itlwm/releases/tag/v1.1.0)，问题修复

williambj1, 2020-10-23:
@the-eric-kwok 能分别测试一下 4b03070 和 324e157 吗? 谢谢

stillcoolme, 2020-10-25:
@the-eric-kwok Could you tell me how to get the itlwm log file，or the log file path?  I also have a similar problem.

the-eric-kwok, 2020-10-25:
It’s in the Troubleshooting page of the official document. 

https://openintelwireless.github.io/itlwm/Troubleshooting.html

the-eric-kwok, 2020-10-26:
> @the-eric-kwok 能分别测试一下 [01d0912](https://github.com/OpenIntelWireless/itlwm/commit/01d09125072beb55d9aea6fc28d719169122f759) 和 [324e157](https://github.com/OpenIntelWireless/itlwm/commit/324e157c33b7aac9c1f01403647c3bb4c72c1fd3) 吗? 谢谢

当然可以，但是我刚刚尝试自行编译，却是无法启用。不知道是不是编译的方式不对

itol1001, 2020-10-28:
Big Sur同样存在这个问题，离Wi-Fi信号源稍远一些就断网了，并且尝试了几款软件都不能科学上网。不知道是不是我不会用的问题

the-eric-kwok, 2020-10-29:
> @the-eric-kwok 能分别测试一下 [01d0912](https://github.com/OpenIntelWireless/itlwm/commit/01d09125072beb55d9aea6fc28d719169122f759) 和 [324e157](https://github.com/OpenIntelWireless/itlwm/commit/324e157c33b7aac9c1f01403647c3bb4c72c1fd3) 吗? 谢谢

@williambj1 大神你好，我不知道应该怎么编译itlwm的源代码，我把 debug 版本的 Lilu 放到 clone 下来的工程目录下，点击 Product -> Archive， 01d0912 我能编译出来，但是放到EFI中无法连接Wi-Fi，而 324e157 则是无法编译，报的头文件IOKit/IOLib.h缺失，如果大神百忙中能帮忙编译一下这两个kext，我是非常愿意继续测试的！

zxystd, 2020-10-29:
git clone --depth=1 https://github.com/OpenIntelWireless/itlwm.git
cd itlwm
git clone --depth=1 https://github.com/acidanthera/MacKernelSDK.git
xcodebuild -scheme "AirportItlwm (all)" -configuration Debug

the-eric-kwok, 2020-10-29:
@zxystd 成功编译了

@williambj1 在 01d0912 下能自动DHCP，表现为切换网络后等了一阵子才出现新的IP地址；但有时会出现无法获取IP地址的情况，此时IP地址为空，Wi-Fi图标上出现感叹号。
在 324e157 下不会自动DHCP，表现为切换网络后IP地址无变化。

测试的系统版本为：macOS 11.0 developer beta 9 (20A5384c)

以下为日志
[Log_324e157_切换后不自动进行DHCP.zip](https://github.com/OpenIntelWireless/itlwm/files/5456053/Log_324e157_.DHCP.zip)
[Log_01d0912_切换后无法获得IP.zip](https://github.com/OpenIntelWireless/itlwm/files/5456054/Log_01d0912_.IP.zip)
[Log_01d0912_切换后成功获得IP.zip](https://github.com/OpenIntelWireless/itlwm/files/5456055/Log_01d0912_.IP.zip)

这是我编译的kext
[AirportItlwm.zip](https://github.com/OpenIntelWireless/itlwm/files/5456073/AirportItlwm.zip)

the-eric-kwok, 2020-10-29:
补充：在 01d0912 下，从睡眠中唤醒时有可能会导致Wi-Fi无法取得IP，与release v1.1.0的表现一致
以下为日志，唤醒大概发生在 17:50 左右
[归档.zip](https://github.com/OpenIntelWireless/itlwm/files/5457513/default.zip)

zxystd, 2020-11-23:
@the-eric-kwok  我这几天装了BigSur测试了确实有不切换Wi-Fi的问题，根据测试群里的童鞋反馈m1 macbook也存在切换网络不切换ip的问题，看起来这像是BigSur的bug，看看官方是否会修复吧。

the-eric-kwok, 2020-12-06:
@meanwill https://openintelwireless.github.io/itlwm/Troubleshooting.html#getting-kext-logs-for-loading-failures

DeniJsonC, 2020-12-07:
> @meanwill https://openintelwireless.github.io/itlwm/Troubleshooting.html#getting-kext-logs-for-loading-failures

老哥 你现在的解决办法是什么 切换到1.1就行了 同样的问题，宿舍ok 到公司后 自动获取的ip段都不对，出现感叹号，明明10段，获取的是169段 手动分配的话 会出现连接几秒就断开的问题 反复秒连秒断

kyleh356, 2020-12-07:
我把地区代码改成CN就很少获取不到ip了 你可以试试 boot-args 添加 itlwm_cc=CN

the-eric-kwok, 2020-12-07:
@DeniJsonC 目前用的是1.2版本，需要手动dhcp续租，但是拿到IP的成功率似乎更高一些

lsvih, 2020-12-29:
现在自动获取 dhcp 概率高了不少

myselfghost, 2021-01-28:
Intel(R) Dual Band Wireless-AC 8275
卡类型：	AirPort Extreme  (0x8086, 0x3E02)
固件版本：	itlwm: 1.2.0 fw: 34.0.1
macOS Version:   11.1  

一样的问题

nemoshimo, 2021-02-03:
> > @zxystd 成功编译了
> > @williambj1 在 [01d0912](https://github.com/OpenIntelWireless/itlwm/commit/01d09125072beb55d9aea6fc28d719169122f759) 下能自动DHCP，表现为切换网络后等了一阵子才出现新的IP地址；但有时会出现无法获取IP地址的情况，此时IP地址为空，Wi-Fi图标上出现感叹号。
> > 在 [324e157](https://github.com/OpenIntelWireless/itlwm/commit/324e157c33b7aac9c1f01403647c3bb4c72c1fd3) 下不会自动DHCP，表现为切换网络后IP地址无变化。
> > 测试的系统版本为：macOS 11.0 developer beta 9 (20A5384c)
> > 以下为日志
> > [Log_324e157_切换后不自动进行DHCP.zip](https://github.com/OpenIntelWireless/itlwm/files/5456053/Log_324e157_.DHCP.zip)
> > [Log_01d0912_切换后无法获得IP.zip](https://github.com/OpenIntelWireless/itlwm/files/5456054/Log_01d0912_.IP.zip)
> > [Log_01d0912_切换后成功获得IP.zip](https://github.com/OpenIntelWireless/itlwm/files/5456055/Log_01d0912_.IP.zip)
> > 这是我编译的kext
> > [AirportItlwm.zip](https://github.com/OpenIntelWireless/itlwm/files/5456073/AirportItlwm.zip)
> 
> 类似的问题,我的系统是mac os11.0.1
> 使用的Airport Itlwm 1.1
> 网卡是ac9462
> 只有头几次连接wifi是能够正常获取到ip并上网,但之后连接就会显示连接成功但是wifi上会有一个感叹号
> [image]
> 如果需要日志,麻烦教我一下如何生成

一样的，公司内网外网切换之后 重新切换回内网 内网连接上了也没办法打开公司网站 只能重启电脑

r33int, 2021-03-03:
Can confirm I am experiencing this issue as well. Still need debug logs?

In1t0, 2021-03-15:
目前只能手动续租DHCP
```bash
sudo ipconfig set en1 DHCP
```

linshixiong, 2021-05-17:
Big Sur 11.3.1, v1.3.0及v2.0.0 均存在相同问题

williambj1, 2021-06-27:
@the-eric-kwok 试一下最新的测试版

the-eric-kwok, 2021-06-27:
@williambj1 确认修复！进行了几次切换网络的测试都没有出现未续租的问题！包括禁用和重新启用Wi-Fi，均未出现问题！

the-eric-kwok, 2021-06-27:
修复于 3e6323070

closed 2021-06-27 by the-eric-kwok