OpenIntelWireless/itlwm #364: Switching from iPhone hotspot to other wifi networks and disconnecting from the network

antoniomcr96, 2020-10-22:
**Have You Read Our Docs**
Yes
**Are You Reporting A Bug**
Probably

**Environment**

- Kext Version: v. 1.2.0 alpha commit 1ec95a8
- WiFi Card Model: AC-9560
- Product ID: 0x9DF0
- macOS Version: Big Sur Beta 9

**Description**
If I'm connected to my home wifi network and I try to connect to my iPhone hotspot it happens that the preferences report that the network is connected, but with my old local IP address (the one of my home wifi network) and internet doesn't work. iPhone correctly reports one device connected.
The same happens if I'm connected to my iPhone hotspot and I try to connect to my home wifi network: the connection sometimes fails, but when it works system preferences reports my old IP address (the one that was associated with my iPhone) and Internet doesn't work.
(This is usually solved by turning off the wifi and turning it back on, then connecting to the desired network.)

When it happens the old network still reports one device connected, even if Mac OS reports to be connected to the new network. My iPhone reports that it has still a device connected even if I manually disconnect from it.

williambj1, 2020-10-23:
Duplicate of #363

williambj1, 2020-10-23:
Leaving open for a valid English version

antoniomcr96, 2020-10-23:
Ok, I’m sorry I didn’t notice. Let me know if I can test other versions or I simply have to wait.
Thanks for all your work!

williambj1, 2020-10-23:
I’m going to test this during the weekend. Some people told me this issue only occurs on Big Sur but not on Catalina.

antoniomcr96, 2020-11-10:
I'm writing just to say that last commit has speeded up incredibly my AC9560 that now works stably even upsteirs (before the connection wasn't stable). The problems remains, though. DHCP needs to be renewed for almost every network switch

williambj1, 2021-10-02:
@antoniomcr96 Should be fixed in newer version, if that's not the case feel free to comment below, thanks.

closed 2021-10-02 by williambj1