OpenIntelWireless/itlwm #365: Internet not working when connecting to a wifi network

mastercreeper66, 2020-10-23:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes
**Environment**

- Kext Version: 1.1.0 Stable
- WiFi Card Model: Intel AC 3160
- Product ID: 0x08b4
- macOS Version: 10.14.6

**Description**

I installed a new router (Fritz!Box 7530) but when i connect to the network i get a (!) on the wi-fi icon

**Bug Report Archive**

[bugreport_12808.zip](https://github.com/OpenIntelWireless/itlwm/files/5431995/bugreport_12808.zip)

**Kext Download Source**
-compiled release
downloaded from pre

zxystd, 2020-10-23:
everything seems to works fine in the log, have you check the wifi password is correct? or can you try to press the DHCP button in the network configurstion panel?

mastercreeper66, 2020-10-24:
Yes, the password is correct. I tried to press "Renew DHCP Lease" but nothing changes

phucnguyen2411, 2020-10-29:
please try to set ip manually, that should works.

mastercreeper66, 2020-10-30:
Setting the IP manually works

closed 2020-11-16 by zxystd