OpenIntelWireless/itlwm #366: AC9560使用AirportItlwm-Catalina开机无法连接WiFi，睡眠后开机才能连接

scdzkj, 2020-10-25:
- Kext Version: AirportItlwm-Catalina-1.2.0
- WiFi Card Model: AC9560
- Product ID: 0xa370
- macOS Version: 10.5.7

AC9560使用AirportItlwm-Catalina
每次开机开机都无法连接WiFi，自动连接保存WiFi和手动连接WiFi都不行，开机点睡眠再开机发现自动连上保存WiFi了，屡试不爽
换了1.1.0 1.2.0版本都一样，连接后非常完美，感谢大佬，能帮忙解决下这个问题吗

williambj1, 2020-10-25:
关闭 Wi-Fi 再次打开即可

scdzkj, 2020-10-25:
> 关闭 Wi-Fi 再次打开即可

试过了，关闭打开也不行，连不上，每次都必须点睡眠后再开机才能连接。。

williambj1, 2020-10-25:
请提供日志

scdzkj, 2020-10-25:
[itlwm_logs.log](https://github.com/OpenIntelWireless/itlwm/files/5434815/itlwm_logs.log)
这是开机无法连接WiFi的日志
[itlwm_logs.log](https://github.com/OpenIntelWireless/itlwm/files/5434817/itlwm_logs.log)
这是睡眠后开机自动连接WiFi的日志

zxystd, 2020-10-26:
第一个日志不完整，后面应该还有

scdzkj, 2020-10-26:
> 第一个日志不完整，后面应该还有

[itlwm_logs.log](https://github.com/OpenIntelWireless/itlwm/files/5436451/itlwm_logs.log)
大佬看看这个日志对不

zncs520, 2020-10-26:
重建缓存就好了,我是NUC8原装版!

scdzkj, 2020-10-26:
> 重建缓存就好了,我是NUC8原装版!

刚刚试了，重建缓存还是一样

scdzkj, 2020-10-27:
> 第一个日志不完整，后面应该还有

大佬，日志看得出来什么问题吗

zxystd, 2020-10-27:
[AirportItlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/5446170/AirportItlwm.kext.zip)
@scdzkj 你试试这个

scdzkj, 2020-10-28:
@zxystd 还是一样的问题，这是用这个驱动后生成的日志
[itlwm_logs.log](https://github.com/OpenIntelWireless/itlwm/files/5448811/itlwm_logs.log)

scdzkj, 2020-10-28:
@zxystd 刚刚突然发现，先关闭显示器开机，完成开机后再开显示器，WiFi正常连接；如果开着显示器开机就不行，只有睡眠再开机就能用。我用的是RX470免驱显卡，应该不是显卡的问题吧。。

zxystd, 2020-10-28:
@scdzkj 根据日志发现两个问题，
1. 认证的时候路由器未回应。这个不知道原因。
2. 认证失败，移除上下文的时候固件崩溃导致重新初始化，初始化的时候失败，导致无法继续下去。这个问题你看看下面这个是否可以工作。
[AirportItlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/5449708/AirportItlwm.kext.zip)

如果上面那个还是无法开机就连上，试试itlwm看看是否可以工作。
[itlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/5449705/itlwm.kext.zip)

测试请都提供log，谢谢。

scdzkj, 2020-10-28:
@最后发的AirportItlwm正常工作了，谢谢大神，请问后续驱动会加入这个修复吗

scdzkj, 2020-10-28:
@zxystd 最后发的AirportItlwm正常工作了，谢谢大神，请问后续驱动会加入这个修复吗

zxystd, 2020-10-28:
> @zxystd 最后发的AirportItlwm正常工作了，谢谢大神，请问后续驱动会加入这个修复吗

会，所以需要你确认是否修复，另外现在工作了也发个日志吧。

scdzkj, 2020-10-28:
> > @zxystd 最后发的AirportItlwm正常工作了，谢谢大神，请问后续驱动会加入这个修复吗
> 
> 会，所以需要你确认是否修复，另外现在工作了也发个日志吧。

确认修复了，这是正常工作的日志
[itlwm_logs.log](https://github.com/OpenIntelWireless/itlwm/files/5450187/itlwm_logs.log)

zxystd, 2020-10-28:
好的。可以多试几次。

scdzkj, 2020-10-28:
> 好的。可以多试几次。

不好意思，大神，刚刚试了几次，还是有这个问题。。
[itlwm_logs.log](https://github.com/OpenIntelWireless/itlwm/files/5451875/itlwm_logs.log)

scdzkj, 2020-10-28:
多试了几次itlwm确定是正常的
[itlwm_logs.log](https://github.com/OpenIntelWireless/itlwm/files/5451953/itlwm_logs.log)
airportitlwm还是有这个问题
[itlwm_logs.log](https://github.com/OpenIntelWireless/itlwm/files/5451875/itlwm_logs.log)

zxystd, 2020-10-28:
@scdzkj 1.1.0版本有没有这个问题？这次貌似是多个线程竞争了。。。

scdzkj, 2020-10-28:
> @scdzkj 1.1.0版本有没有这个问题？这次貌似是多个线程竞争了。。。

1.1.0也有这个问题，这是日志
[itlwm_logs.log](https://github.com/OpenIntelWireless/itlwm/files/5452191/itlwm_logs.log)

zxystd, 2020-10-28:
1.1.0跟这个不是同一个问题，不过都有关联失败这个问题，我也不知道为什么，你确定itlwm是正常的？关联这部分是由公共部分做的，itlwm的也是一样的。

scdzkj, 2020-10-28:
> 1.1.0跟这个不是同一个问题，不过都有关联失败这个问题，我也不知道为什么，你确定itlwm是正常的？关联这部分是由公共部分做的，itlwm的也是一样的。

已经试过几次了，确定itlwm是正常的，只有airportitlwm有问题，不知道为什么睡眠一下再开机又是正常的

AndresGarciaSobrado91, 2020-10-30:
@zxystd has v1.2.0-alpha this fix so I can continue updating the kext from there? Thank u

tommyvct, 2020-10-31:
我ax200也有这个问题，冷启动不行，从Windows重启行

scdzkj, 2020-11-01:
@zxystd 我把显示器换成HDMI接口就正常了，换回DP又不行，470免驱显卡，难道和显卡有冲突？

zxystd, 2020-11-01:
@scdzkj 关于黑苹果的东西我并不是很懂。。。对了，最新的alpha版本你试了的没？

zxystd, 2020-11-01:
@AndresGarciaSobrado91 alpha version is always the latest build, you can download there.

scdzkj, 2020-11-01:
> @scdzkj 关于黑苹果的东西我并不是很懂。。。对了，最新的alpha版本你试了的没？

最新的alpha版本还是没解决，我再研究试试big sur有没有问题

scdzkj, 2020-11-04:
@zxystd 试了下big sur是正常的，虽然系统报告的状态还是显示未关联，日志也显示Kext not loaded，但是wifi正常连接使用

sail9708, 2020-11-16:
我和你的问题正好相反，重启后正常，睡眠唤醒后，无法连接

Creky, 2020-11-20:
使用 [Hackintool](https://github.com/headkaze/Hackintool) 清理一下缓存就好了，我的原先也是无法打开wifi连网

Creky, 2020-11-20:
> 使用 [Hackintool](https://github.com/headkaze/Hackintool) 清理一下缓存就好了，我的原先也是无法打开wifi连网

这工具别乱用，直接用工具菜单下，倒数第二排最后一个图标，鼠标移上去会提示重建缓存的

win1010525, 2021-02-26:
> 我和你的问题正好相反，重启后正常，睡眠唤醒后，无法连接

> 使用 [Hackintool](https://github.com/headkaze/Hackintool) 清理一下缓存就好了，我的原先也是无法打开wifi连网

> > 使用 [Hackintool](https://github.com/headkaze/Hackintool) 清理一下缓存就好了，我的原先也是无法打开wifi连网
> 
> 这工具别乱用，直接用工具菜单下，倒数第二排最后一个图标，鼠标移上去会提示重建缓存的

Off-topic issue. Have you tried to change the ASPM？

win1010525, 2021-02-26:
@williambj1

closed 2021-02-26 by williambj1