OpenIntelWireless/itlwm #369: kextstat displays itlwm loaded but heliport confirms itlwm is not loaded

Tushar-Harsora, 2020-10-30:
**Have You Read Our Docs**
YES

**Are You Reporting A Bug**
Yes

**Environment**
- Kext Version: v1.1.0 stable
- WiFi Card Model: Intel® Wi-Fi 6 AX200
- macOS Version: Catalina 10.15.7 (19H2)

**Description**
running `kextstat | grep itlwm` in terminal displays itlwm as loaded kext.
but when opening heliport it reports error saying itlwm is not running.

additionally system preferences -> network also doesn't list wifi card as wifi or ethernet device.

**Bug Report Archive**
[image]
[bugreport_3253.zip](https://github.com/OpenIntelWireless/itlwm/files/5463690/bugreport_3253.zip)

**Kext Download Source**
https://github.com/OpenIntelWireless/itlwm/releases/download/v1.1.0/itlwm_v1.1.0_Stable.kext.zip

ErrorErrorError, 2020-10-30:
Looking at the log files the firmware failed to load too the wifi card. Try restarting your computer and see if it works.

XEmpty, 2020-10-31:
I had similar problem yesterday. But in some reason I use function keys on keyboard (fn + wifi / bluetooth). Anyway, resolve that problem login to Live USB Linux distro (pop-os_). Then system automatic reenable my wifi/bt . I read that you shouldn't use these functional keys in some cases. 

BTW : Speed of those drivers are so bad... even signal are poor. from 150Mbs(Linux, Windows) to 20(OSX) on WIFI when on linux I had 140+ in room where ITLWM doesn't even work... and it's doesn't matter if I using wifi near router or not. My speed it's all the time the same.

ErrorErrorError, 2020-10-31:
> I had similar problem yesterday. But in some reason I use function keys on keyboard (fn + wifi / bluetooth). Anyway, resolve that problem login to Live USB Linux distro (pop-os_). Then system [...]

Speeds are expected as per FAQ. https://openintelwireless.github.io/itlwm/FAQ.html

williambj1, 2020-10-31:
@XEmpty Feel free to use alternative solutions, no one is forcing you to use itlwm.

williambj1, 2020-10-31:
@Tushar-Harsora Looks like itlwm did not load properly, could you please check whether itlwm has attached to your WiFi card in IOReg? The log file for itlwm is empty so we really don't know what's going on. I would suggest loading itlwm manually in terminal (instructions are in our docs) and see if there is any difference.

closed 2020-11-02 by Tushar-Harsora