OpenIntelWireless/itlwm #371: Random disconnects from SSID and cannot reconnect

devksingh4, 2020-10-31:
Hello, 
Using OpenCore 0.6.2, AirportItlwm v1.0 Beta on macOS Catalina 10.15.7. I am using Intel AC 8260 on an AMD Hackintosh. The wifi will disconnect about once every few hours, and requires a reboot to get reconnected. If I do not reboot, it will do one of the following: 

*  not find any wifi networks
*  find the network and seems like its trying to connect, but it does not connect.
*  actually attempts to connect, but then loses all wifi networks and does not connect

win1010525, 2020-11-01:
Please send us log.

win1010525, 2020-11-01:
No log is provided, we don’t know either.

devksingh4, 2020-11-01:
My apologies, I will get that to you ASAP.

zxystd, 2020-11-01:
@devksingh4 try to get the log. and then try the 1.2.0 alpha version, if problem persist, also collect the log. thanks.

win1010525, 2020-11-01:
You can get the log by typing in this in Terminal: log show --last boot --predicate 'sender="AirportItlwm"' >~/Desktop/itlwm.log

devksingh4, 2020-11-01:
Hi @zxystd ,
Is there a precompiled kext somewhere?

win1010525, 2020-11-01:
No, the newest next is 1.2.0-alpha. It is updating constantly.

devksingh4, 2020-11-01:
Here's a copy of after boot and switching wifi networks a few times. Lots of OUTPUT_ERROR status=142, status=131. I will also take a log of when it happens by itself and upload it here. 
[Uploading itlwm.log…]()

win1010525, 2020-11-01:
Ok thanks.

devksingh4, 2020-11-01:
@win1010525  are there compile instructions anywhere? I am not very experienced with macOS development so not sure what the toolchain process is.

win1010525, 2020-11-01:
If you compile the code on the master branch, you’ll get a 100% same one as 1.2.0-alpha. Prereleases are always the newest one.

devksingh4, 2020-11-01:
Sorry, I mean instructions on how to and what to compile for AirportItlwm. Is it just "press build in xcode after cloning"?

win1010525, 2020-11-01:
Yes, all you need is to choose which kext to compile and then compile it.

zxystd, 2020-11-01:
`iwm_rx_tx_cmd_single 701 OUTPUT_ERROR status=131` this is a router compatible problem, will not concern in short. you had better try another router.

devksingh4, 2020-11-01:
I have a Netgear Nighthawk r7900p, with the SSIDs split into 2.4 and 5ghz. What is the issue with the router?

win1010525, 2020-11-01:
Sorry but I dont have one.🙁

zxystd, 2020-11-01:
@devksingh4 `131` means Network card send the packet to the router but the router refused because th packet size is too large, is happening in some router, not confirm why. you can use your cellphone hotspot and try whether it is the same behavior.

devksingh4, 2020-11-01:
so why does this not happen in linux or windows then? 

I have implicit beamforming and MU-MIMO enabled if that help any.

on compiling it, I keep getting the error: IOKit/IOLib.h not found, and sys/param.h not found, "In file included from itlwm/itl80211/...". Is the master branch currently buildable or do I have something setup incorrectly in Xcode?

zxystd, 2020-11-01:
@devksingh4 because macOS didn't have the 802.11 stack to use, so we use the openbsd's which is partial implemented the 802.11 standards.

you should download macKerSDK, here are the full build commands:
```
git clone --depth=1 https://github.com/OpenIntelWireless/itlwm.git
cd itlwm
git clone --depth=1 https://github.com/acidanthera/MacKernelSDK.git
xcodebuild -scheme "AirportItlwm (all)" -configuration Debug
```

zxystd, 2020-11-01:
@devksingh4 Also another way may caused the problem is your MTU setting, the MTU should be the same or lower with you router/ISP setting.

devksingh4, 2020-11-01:
This seems to be the issue - mac os automatic was trying 1500 as far as I can tell, but the router one for LAN seems to be set to 1472 based off of ping test. Once I can verify that this was the issue, I'll close this issue.

devksingh4, 2020-11-01:
The issue now manifests differently with 1.2.0 alpha - It now says that any discovered SSID is not available, so I don't think the problem lies with my router. "Request Unhandled IOCTL WOW_PARAMETERS (69)" and a few other Unhandled Requests

[itlwm2.log](https://github.com/OpenIntelWireless/itlwm/files/5471125/itlwm2.log)
[itlwm3.log](https://github.com/OpenIntelWireless/itlwm/files/5471126/itlwm3.log)

zxystd, 2020-11-01:
it is another problem.

devksingh4, 2020-11-01:
Should I file a different bug?

zxystd, 2020-11-01:
already exists #366 , i can not cinfirm whether it us fixed because we never meet this problem...

devksingh4, 2020-11-01:
Got it, thanks. Will close this issue as the original issue is fixed in 1.2.0

closed 2020-11-01 by devksingh4