OpenIntelWireless/itlwm #372: 没有日志是要我给你算一卦么？

hsmliu, 2020-11-02:
没有日志是要我给你算一卦么？

给我算一卦吧

Timestamp                       Thread     Type        Activity             PID    TTL  
2020-11-02 08:35:29.709696+0800 0x3b4      Default     0x0                  0      0    kernel: (IO80211Family) IO80211Controller::dataLinkLayerAttachComplete():  adding AppleEFINVRAM notification
2020-11-02 08:35:29.709712+0800 0x3b4      Default     0x0                  0      0    kernel: (IO80211Family) IO80211Interface::efiNVRAMPublished():
2020-11-02 08:35:29.709851+0800 0x3b4      Default     0x0                  0      0    kernel: (IO80211Family) initting Scan Manager controller 0xe858afed7b7375bb ioservice 0xe858afed7b8e95bb interface 0xe858afed7b97d68b ioservce 0xe858afed7b7375bb
2020-11-02 08:35:29.710034+0800 0x3b4      Default     0x0                  0      0    kernel: (IO80211Family) bpfAttach len 72 dlt 12
2020-11-02 08:35:29.710388+0800 0x3b4      Default     0x0                  0      0    kernel: (IO80211Family) Not adding subscription due to missing boot-arg wlan.iorsubscriptions
2020-11-02 08:35:32.135464+0800 0x636      Default     0x0                  0      0    kernel: (IO80211Family) AirportItlwm::init AirPortFamily_kexts-1838 "AirPortFamily_kexts-1838" Oct  7 2020 21:04:24
2020-11-02 08:35:32.189228+0800 0x638      Default     0x0                  0      0    kernel: (IO80211Family) AirportItlwm::init AirPortFamily_kexts-1838 "AirPortFamily_kexts-1838" Oct  7 2020 21:04:24
2020-11-02 08:35:32.189606+0800 0x62f      Default     0x0                  0      0    kernel: (IO80211Family) AirportItlwm::init AirPortFamily_kexts-1838 "AirPortFamily_kexts-1838" Oct  7 2020 21:04:24
2020-11-02 08:35:32.245485+0800 0x63e      Default     0x0                  0      0    kernel: (IO80211Family) AirportItlwm::init AirPortFamily_kexts-1838 "AirPortFamily_kexts-1838" Oct  7 2020 21:04:24
2020-11-02 08:35:32.264181+0800 0x62b      Default     0x0                  0      0    kernel: (IO80211Family) AirportItlwm::init AirPortFamily_kexts-1838 "AirPortFamily_kexts-1838" Oct  7 2020 21:04:24
2020-11-02 08:35:32.691459+0800 0x487      Default     0x0                  0      0    kernel: (IO80211Family) IO80211VirtualInterface::IO80211P2PInterface::init name <p2p0> role 1
2020-11-02 08:35:32.691476+0800 0x487      Default     0x0                  0      0    kernel: (IO80211Family) IO80211VirtualInterface::init _controller is 0xe858afed7b7375bb, provider is 0xe858afed7b8e95bb role is 1
2020-11-02 08:35:32.691788+0800 0x487      Default     0x0                  0      0    kernel: (IO80211Family) Not adding subscription due to missing boot-arg wlan.iorsubscriptions
2020-11-02 08:35:32.691793+0800 0x487      Default     0x0                  0      0    kernel: (IO80211Family) IO80211VirtualInterface::init SUCCESS _controller is 0xe858afed7b7375bb, provider is 0xe858afed7b8e95bb
2020-11-02 08:35:32.692306+0800 0x487      Default     0x0                  0      0    kernel: (IO80211Family) IO80211VirtualInterface::IO80211P2PInterface::init name <awdl0> role 4
2020-11-02 08:35:32.692321+0800 0x487      Default     0x0                  0      0    kernel: (IO80211Family) IO80211P2PInterface::peerToPeerAttachToBpf name <awdl0> role 4 successful attach to bpf type 147
2020-11-02 08:35:32.692334+0800 0x487      Default     0x0                  0      0    kernel: (IO80211Family) IO80211VirtualInterface::init _controller is 0xe858afed7b7375bb, provider is 0xe858afed7b8e95bb role is 4
2020-11-02 08:35:32.692976+0800 0x487      Default     0x0                  0      0    kernel: (IO80211Family) com.apple.p2p: setPeerManagerLogFlags[5752] logflags: 0
2020-11-02 08:35:32.692977+0800 0x487      Default     0x0                  0      0    kernel: (IO80211Family) com.apple.p2p: setPeerManagerLogFlags[5760] fPcapLogStream is NULL
2020-11-02 08:35:32.692979+0800 0x487      Default     0x0                  0      0    kernel: (IO80211Family) com.apple.p2p: setPeerManagerLogFlags[5752] logflags: 0
2020-11-02 08:35:32.692980+0800 0x487      Default     0x0                  0      0    kernel: (IO80211Family) com.apple.p2p: setPeerManagerLogFlags[5760] fPcapLogStream is NULL
2020-11-02 08:35:32.692981+0800 0x487      Default     0x0                  0      0    kernel: (IO80211Family) Finding _faultReporter
2020-11-02 08:35:32.692999+0800 0x487      Default     0x0                  0      0    kernel: (IO80211Family) com.apple.p2p: initializeParams:265 [BssSteering] Resetting BSS Steering Ctxt
2020-11-02 08:35:32.693000+0800 0x487      Default     0x0                  0      0    kernel: (IO80211Family) com.apple.p2p: addAllEventSources:313 [BssSteering] Adding Event sources
2020-11-02 08:35:32.693009+0800 0x487      Default     0x0                  0      0    kernel: (IO80211Family) com.apple.p2p: setRealTime[21987] :0
2020-11-02 08:35:32.693242+0800 0x487      Default     0x0                  0      0    kernel: (IO80211Family) com.apple.p2p: setPeerManagerLogFlags[5752] logflags: 0
2020-11-02 08:35:32.693245+0800 0x487      Default     0x0                  0      0    kernel: (IO80211Family) com.apple.p2p: setPeerManagerLogFlags[5752] logflags: 0
2020-11-02 08:35:32.693472+0800 0x487      Default     0x0                  0      0    kernel: (IO80211Family) Not adding subscription due to missing boot-arg wlan.iorsubscriptions
2020-11-02 08:35:32.693476+0800 0x487      Default     0x0                  0      0    kernel: (IO80211Family)  Creating all peerManager reporters
2020-11-02 08:35:32.693521+0800 0x487      Default     0x0                  0      0    kernel: (IO80211Family) IO80211VirtualInterface::init SUCCESS _controller is 0xe858afed7b7375bb, provider is 0xe858afed7b8e95bb
2020-11-02 08:35:32.693528+0800 0x487      Default     0x0                  0      0    kernel: (IO80211Family) com.apple.p2p: IO80211AWDLPeerManager::interfaceStateChange Interface link is down, nothing to do. Source = SetInfraChannel
2020-11-02 08:35:32.701023+0800 0x793      Default     0x0                  0      0    kernel: (IO80211Family) com.apple.p2p: IO80211AWDLPeerManager::setAwdlPeerTrafficRegistration assisted discovery requested: channel 0 realTime 0 active 0 forcedMaster 00:00:00:00:00:00
2020-11-02 08:35:32.701048+0800 0x793      Default     0x0                  0      0    kernel: (IO80211Family) scanControl: AWDL is controlling scan, ePNOScan ON, roamScan ON, limitedRoam OFF, abortScan 0
2020-11-02 08:35:32.701054+0800 0x793      Default     0x0                  0      0    kernel: (IO80211Family) scanControlCommand: Sending scan control command, ePNOScan ON (0x0), roamScan ON (0x0), limitedRoam OFF (0x0), abortScan 0
2020-11-02 08:35:32.701167+0800 0x793      Default     0x0                  0      0    kernel: (IO80211Family) com.apple.p2p: IO80211AWDLPeerManager::updateLocalRequests: Deleting PTR:._wifid._tcp.local. service PTR browse from airportd (pid 211)
2020-11-02 08:35:32.701193+0800 0x793      Default     0x0                  0      0    kernel: (IO80211Family) com.apple.p2p: IO80211AWDLPeerManager::updateLocalRequests Received service request but not configured (_stateFlags:0x0)! bailing..
2020-11-02 08:35:32.711015+0800 0x792      Default     0x0                  0      0    kernel: (IO80211Family) en2: setting diversity to: -1
2020-11-02 08:35:32.711020+0800 0x792      Default     0x0                  0      0    kernel: (IO80211Family) en2: Error configuring antenna diversity (index = -1).
2020-11-02 08:35:32.711028+0800 0x792      Default     0x0                  0      0    kernel: (IO80211Family) en2: setting tx antenna: -1
2020-11-02 08:35:32.711031+0800 0x792      Default     0x0                  0      0    kernel: (IO80211Family) en2: Error configuring transmit antenna (index = -1).
2020-11-02 08:35:32.719595+0800 0x487      Default     0x0                  0      0    kernel: (IO80211Family) Setting BTCoex Profile: band:8
2020-11-02 08:35:32.719602+0800 0x487      Default     0x0                  0      0    kernel: (IO80211Family) Profile[0]: mode:7; desense:0; desense_level:0; chain_power_offset:
2020-11-02 08:35:32.719605+0800 0x487      Default     0x0                  0      0    kernel: (IO80211Family) 0,
2020-11-02 08:35:32.719606+0800 0x487      Default     0x0                  0      0    kernel: (IO80211Family) 0,
2020-11-02 08:35:32.719607+0800 0x487      Default     0x0                  0      0    kernel: (IO80211Family) 0,
[... cut ...]
2020-11-02 08:42:42.541109+0800 0x487      Default     0x0                  0      0    kernel: (IO80211Family) 0,
2020-11-02 08:42:42.541110+0800 0x487      Default     0x0                  0      0    kernel: (IO80211Family) 0,
2020-11-02 08:42:42.541111+0800 0x487      Default     0x0                  0      0    kernel: (IO80211Family) 0,
2020-11-02 08:42:42.541112+0800 0x487      Default     0x0                  0      0    kernel: (IO80211Family) 0,
2020-11-02 08:42:42.541113+0800 0x487      Default     0x0                  0      0    kernel: (IO80211Family)

hsmliu, 2020-11-02:
我这个是连不上网，重启后导出的日志，可能不行

zxystd, 2020-11-02:
你这两个日志都是连上了。另外你的日志不对，看不到itlwm的日志。

hsmliu, 2020-11-02:
> 你这两个日志都是连上了。另外你的日志不对，看不到itlwm的日志。

我用的是AirportItlwm，然后kext目录里面也放了itlwm，我把itlwm设置为未启用，如果AirportItlwm和itlwm同时启用就卡开机界面

zxystd, 2020-11-02:
我指的就是`AirportItlwm `的日志，你这个其他日志都有，就是没包含`AirportItlwm `的日志

hsmliu, 2020-11-02:
> 我指的就是`AirportItlwm `的日志，你这个其他日志都有，就是没包含`AirportItlwm `的日志

可是明明能上网   真奇怪，改怎么用这个驱动才对呢

win1010525, 2020-11-03:
@zxystd 他是Big Sur没日志...

win1010525, 2020-11-03:
> > 你这两个日志都是连上了。另外你的日志不对，看不到itlwm的日志。
> 
> 我用的是AirportItlwm，然后kext目录里面也放了itlwm，我把itlwm设置为未启用，如果AirportItlwm和itlwm同时启用就卡开机界面

同时启用会冲突，这两个只能启用一个

closed 2020-11-08 by zxystd