OpenIntelWireless/itlwm #374: Airportitlwm有时会自己断线，要等一会才会恢复连线

alancheng49, 2020-11-02:
如题目叙述，使用airportitlwm时，wifi有时或自行断开，需等待一阵子或手动重启才可恢复连线
我的网卡是AX201，使用airportitlwm版本是alpha 1.2.0，macOS版本是10.15.7
[itlwm.log](https://github.com/OpenIntelWireless/itlwm/files/5475997/itlwm.log)

devksingh4, 2020-11-02:
I have this same issue with Intel AC 8265.

zxystd, 2020-11-02:
日志里没有看到断开连接，但是看到上行一直是满的，这样确实会造成断流，目前这个驱动还扛不住很强的上行压力，当然，也跟路由器有关。

alancheng49, 2020-11-02:
所以解法是不要一次跑那么多流量吗？还是可以透过设定某个东西来改善问题

devksingh4, 2020-11-02:
Sorry, I don't speak Mandarin so I'm using Google Translate. However: 
I have also noticed that the driver cannot handle high downstream traffic either. Is there a way to limit the upload and download to get more stability?

holoto, 2020-11-03:
> 如题目叙述，使用airportitlwm时，wifi有时或自行断开，需等待一阵子或手动重启才可恢复连线
> 我的网卡是AX201，使用airportitlwm版本是alpha 1.2.0，macOS版本是10.15.7
> [itlwm.log](https://github.com/OpenIntelWireless/itlwm/files/5475997/itlwm.log)

我有同样的问题 在ax200上
需要手动dhcp下
推测是网卡睡眠了(硬件都是有睡眠模式 一段时间会自动关闭)

holoto, 2020-11-03:
> 所以解法是不要一次跑那么多流量吗？还是可以透过设定某个东西来改善问题

大概率睡眠了 这种故障是 间歇性的 
等驱动修复吧

win1010525, 2020-11-03:
> Sorry, I don't speak Mandarin so I'm using Google Translate. However:
> I have also noticed that the driver cannot handle high downstream traffic either. Is there a way to limit the upload and download to get more stability?

Well you can try to limit upload speed in your router settings (if supported)

holoto, 2020-11-04:
`
root@MacBook-Pro ~ # ping qq.com
PING qq.com (58.250.137.36): 56 data bytes
64 bytes from 58.250.137.36: icmp_seq=0 ttl=53 time=24.806 ms
64 bytes from 58.250.137.36: icmp_seq=1 ttl=53 time=14.636 ms
^C
--- qq.com ping statistics ---
2 packets transmitted, 2 packets received, 0.0% packet loss
round-trip min/avg/max/stddev = 14.636/19.721/24.806/5.085 ms
root@MacBook-Pro ~ # ping 192.168.1.1
PING 192.168.1.1 (192.168.1.1): 56 data bytes
Request timeout for icmp_seq 0
Request timeout for icmp_seq 1
64 bytes from 192.168.1.1: icmp_seq=0 ttl=64 time=2560.441 ms
64 bytes from 192.168.1.1: icmp_seq=1 ttl=64 time=1555.613 ms
64 bytes from 192.168.1.1: icmp_seq=2 ttl=64 time=551.976 ms
64 bytes from 192.168.1.1: icmp_seq=3 ttl=64 time=7.846 ms
64 bytes from 192.168.1.1: icmp_seq=4 ttl=64 time=2.944 ms
64 bytes from 192.168.1.1: icmp_seq=5 ttl=64 time=1.451 ms
64 bytes from 192.168.1.1: icmp_seq=6 ttl=64 time=2.276 ms
64 bytes from 192.168.1.1: icmp_seq=7 ttl=64 time=17.312 ms
64 bytes from 192.168.1.1: icmp_seq=8 ttl=64 time=2.208 ms
64 bytes from 192.168.1.1: icmp_seq=9 ttl=64 time=2.455 ms
64 bytes from 192.168.1.1: icmp_seq=10 ttl=64 time=1.796 ms
64 bytes from 192.168.1.1: icmp_seq=11 ttl=64 time=1.776 ms
64 bytes from 192.168.1.1: icmp_seq=12 ttl=64 time=2.292 ms
64 bytes from 192.168.1.1: icmp_seq=13 ttl=64 time=5.311 ms
64 bytes from 192.168.1.1: icmp_seq=14 ttl=64 time=4.352 ms
64 bytes from 192.168.1.1: icmp_seq=15 ttl=64 time=2.480 ms
64 bytes from 192.168.1.1: icmp_seq=16 ttl=64 time=1.798 ms
64 bytes from 192.168.1.1: icmp_seq=17 ttl=64 time=3.727 ms
o64 bytes from 192.168.1.1: icmp_seq=18 ttl=64 time=29.969 ms
64 bytes from 192.168.1.1: icmp_seq=19 ttl=64 time=1.768 ms
64 bytes from 192.168.1.1: icmp_seq=20 ttl=64 time=17.741 ms
64 bytes from 192.168.1.1: icmp_seq=21 ttl=64 time=15.070 ms
64 bytes from 192.168.1.1: icmp_seq=22 ttl=64 time=1.758 ms
64 bytes from 192.168.1.1: icmp_seq=23 ttl=64 time=11.787 ms
64 bytes from 192.168.1.1: icmp_seq=24 ttl=64 time=14.922 ms
64 bytes from 192.168.1.1: icmp_seq=25 ttl=64 time=1.835 ms
64 bytes from 192.168.1.1: icmp_seq=26 ttl=64 time=1.896 ms
64 bytes from 192.168.1.1: icmp_seq=27 ttl=64 time=4.252 ms
64 bytes from 192.168.1.1: icmp_seq=28 ttl=64 time=4.935 ms
64 bytes from 192.168.1.1: icmp_seq=29 ttl=64 time=15.869 ms
64 bytes from 192.168.1.1: icmp_seq=30 ttl=64 time=4.068 ms
64 bytes from 192.168.1.1: icmp_seq=31 ttl=64 time=1.743 ms
64 bytes from 192.168.1.1: icmp_seq=32 ttl=64 time=1.790 ms
64 bytes from 192.168.1.1: icmp_seq=33 ttl=64 time=12.946 ms
64 bytes from 192.168.1.1: icmp_seq=34 ttl=64 time=2.602 ms
64 bytes from 192.168.1.1: icmp_seq=35 ttl=64 time=1.811 ms
64 bytes from 192.168.1.1: icmp_seq=36 ttl=64 time=1.728 ms
64 bytes from 192.168.1.1: icmp_seq=37 ttl=64 time=2.007 ms
64 bytes from 192.168.1.1: icmp_seq=38 ttl=64 time=3.976 ms
64 bytes from 192.168.1.1: icmp_seq=39 ttl=64 time=3.726 ms
64 bytes from 192.168.1.1: icmp_seq=40 ttl=64 time=2.974 ms
64 bytes from 192.168.1.1: icmp_seq=41 ttl=64 time=204.260 ms
64 bytes from 192.168.1.1: icmp_seq=42 ttl=64 time=485.960 ms
64 bytes from 192.168.1.1: icmp_seq=43 ttl=64 time=397.483 ms
64 bytes from 192.168.1.1: icmp_seq=44 ttl=64 time=78.842 ms
64 bytes from 192.168.1.1: icmp_seq=45 ttl=64 time=414.953 ms
64 bytes from 192.168.1.1: icmp_seq=46 ttl=64 time=132.992 ms
64 bytes from 192.168.1.1: icmp_seq=47 ttl=64 time=1.848 ms
64 bytes from 192.168.1.1: icmp_seq=48 ttl=64 time=2.026 ms
64 bytes from 192.168.1.1: icmp_seq=49 ttl=64 time=2.292 ms
64 bytes from 192.168.1.1: icmp_seq=50 ttl=64 time=32.580 ms
64 bytes from 192.168.1.1: icmp_seq=51 ttl=64 time=52.129 ms
64 bytes from 192.168.1.1: icmp_seq=52 ttl=64 time=2.598 ms
64 bytes from 192.168.1.1: icmp_seq=53 ttl=64 time=4.229 ms
64 bytes from 192.168.1.1: icmp_seq=54 ttl=64 time=2.529 ms
64 bytes from 192.168.1.1: icmp_seq=55 ttl=64 time=3.413 ms
64 bytes from 192.168.1.1: icmp_seq=56 ttl=64 time=2.564 ms
[... 716 lines cut ...]
64 bytes from 192.168.1.1: icmp_seq=772 ttl=64 time=8.273 ms
64 bytes from 192.168.1.1: icmp_seq=773 ttl=64 time=365.705 ms
64 bytes from 192.168.1.1: icmp_seq=774 ttl=64 time=3.210 ms
64 bytes from 192.168.1.1: icmp_seq=775 ttl=64 time=65.203 ms
64 bytes from 192.168.1.1: icmp_seq=776 ttl=64 time=476.777 ms
64 bytes from 192.168.1.1: icmp_seq=777 ttl=64 time=1.840 ms
64 bytes from 192.168.1.1: icmp_seq=778 ttl=64 time=2.263 ms
64 bytes from 192.168.1.1: icmp_seq=779 ttl=64 time=1.480 ms
64 bytes from 192.168.1.1: icmp_seq=780 ttl=64 time=6.856 ms
64 bytes from 192.168.1.1: icmp_seq=781 ttl=64 time=5.491 ms
64 bytes from 192.168.1.1: icmp_seq=782 ttl=64 time=2.529 ms
64 bytes from 192.168.1.1: icmp_seq=783 ttl=64 time=3.743 ms
64 bytes from 192.168.1.1: icmp_seq=784 ttl=64 time=2.302 ms
64 bytes from 192.168.1.1: icmp_seq=785 ttl=64 time=35.345 ms
64 bytes from 192.168.1.1: icmp_seq=786 ttl=64 time=18.788 ms
64 bytes from 192.168.1.1: icmp_seq=787 ttl=64 time=1.703 ms
64 bytes from 192.168.1.1: icmp_seq=788 ttl=64 time=10.458 ms
64 bytes from 192.168.1.1: icmp_seq=789 ttl=64 time=6.780 ms
64 bytes from 192.168.1.1: icmp_seq=790 ttl=64 time=6.566 ms

i
    
`

我测试了下 如果一直有个ping程序 工作 ,网卡就不会自动休眠 断线

holoto, 2020-11-04:
[image]
修改mtu 貌似有作用

closed 2020-11-08 by zxystd