OpenIntelWireless/itlwm #375: HeliPort not turning on WiFi

dvrkoo, 2020-11-02:
**Have You Read Our Docs**
yes
**Are You Reporting A Bug**
yes
**Environment**

- Kext Version: itlwm 1.1.0
- WiFi Card Model: intel ac 7260
- Product ID: MacBook Air 13 mid 2013
- macOS Version: Catalina

**Description**
Long story short I put my pc in sleep mode earlier today and after turning it back on HeliPort simply isn't turning wifi on I tried to reboot to force a shut down and boot the pc, I tried uninstalling and reinstalling the kext , I cannot replicate the issue as I can't get itlwm back working
**Bug Report Archive**
[bugreport_15349.zip](https://github.com/OpenIntelWireless/itlwm/files/5476842/bugreport_15349.zip)

**Kext Download Source**
the GitHub and no it's not a modified version

williambj1, 2020-11-02:
Looks like your card is not responding. Try to restart to Linux or Windows, see if there is any difference.

dvrkoo, 2020-11-03:
> Looks like your card is not responding. Try to restart to Linux or Windows, see if there is any difference.

Tried booting into Linux from usb and WiFi is working as it should

zxystd, 2020-11-04:
> > Looks like your card is not responding. Try to restart to Linux or Windows, see if there is any difference.
> 
> Tried booting into Linux from usb and WiFi is working as it should

other systems didn't use the EFI that your Hackintosh using.

dvrkoo, 2020-11-04:
> > > Looks like your card is not responding. Try to restart to Linux or Windows, see if there is any difference.
> 
> > 
> 
> > Tried booting into Linux from usb and WiFi is working as it should
> 
> 
> 
> other systems didn't use the EFI that your Hackintosh using.

Yeah but you guys are not helping me answering every 10 hours without actually pointing me in the right direction,I understood it was a EFI problem but how can I check in my config.plist if I f.ed something up, if I open hackintool it actually shows the WiFi card and its shown as in L1 powerstate

williambj1, 2020-11-04:
Sorry but we are not responsible for personal configs. We haven't seen any evidence showing your issue as a bug of our code.

zxystd, 2020-11-04:
@dvrkoo First of all, You are asking about "Hackintosh" problems, sorry I don't know too much about this, and it is out of our control. I understand you want help and want to report the problem(if it really be). Thank you but I suggest you check your configuration by yourself(sleep/wake/hibernatemode side), rather than said other systems works fine.

closed 2020-11-08 by zxystd