OpenIntelWireless/itlwm #376: [Airportitlwm] 1.2最新内测版本,大文件上传时频繁断开重连

CodeApePro, 2020-11-03:
硬件 nuc8i5beh
驱动版本 1.2版本
引导版本 oc 0.6.2 
系统型号 10.15.7
网卡型号 Intel® Wireless-AC 9560 + Bluetooth 5.0
在上传大文件跑高上行时,频繁断线重连,最高时甚至不到一分钟就会断开重连一次,日志如下.辛苦各位大神.
[itlwm_log.txt](https://github.com/OpenIntelWireless/itlwm/files/5480470/itlwm_log.txt)

williambj1, 2020-11-03:
固件频繁崩溃 @zxystd

zxystd, 2020-11-03:
如果你的路由器还可以的话，可以做大流量上传，但是很明显你的路由器不让，上行一直占用满的时候，路由器直接发了断连请求。固件崩溃也是断连造成的，并不是造成断连的原因。

CodeApePro, 2020-11-03:
> 如果你的路由器还可以的话，可以做大流量上传，但是很明显你的路由器不让，上行一直占用满的时候，路由器直接发了断连请求。固件崩溃也是断连造成的，并不是造成断连的原因。

好的,了解了,环境是办公网,可能是路由器上做了什么限制.我打算找一台别的环境设备或者装win10下在同一个路由环境下做对比测试,看看别的系统或硬件环境下是否也会被路由器强制断开上传.

CodeApePro, 2020-11-05:
> 如果你的路由器还可以的话，可以做大流量上传，但是很明显你的路由器不让，上行一直占用满的时候，路由器直接发了断连请求。固件崩溃也是断连造成的，并不是造成断连的原因。

经测试路由器应该确实有带宽限制,不过其他的系统和产品触达限制时并没有发现断网重连的情况.给您这边反馈下测试结果以供您这边后期调整.感谢您的付出.

meanwill, 2020-11-30:
我也是.上传占满几秒就可能会掉线

williambj1, 2020-12-25:
Should be fixed in recent commits.

closed 2020-12-25 by williambj1