OpenIntelWireless/itlwm #377: Handoff has some problems that shouldn't be

Theredmc, 2020-11-03:
**Have You Read Our Docs**
YES

**Are You Reporting A Bug**
YES

**Environment**

- Kext Version: newest available alpha 
- WiFi Card Model: 3165
- Product ID: 
- macOS Version: 10.15.7

**Description**
I am germany so sorry for my English. 
Handoff works except when you right click on the desktop and what to take a picture with your iPhone it always time out and WFI hotspot doesn't work neither it times out , but the other stuff is working

**Bug Report Archive**
I am not using heliport there is no need if the system stuff is working.

**Kext Download Source**
this website

williambj1, 2020-11-03:
Other features are not advertised as "Handoff". You are probably looking for something else. Read our docs and the following URL for more info.

https://support.apple.com/en-us/HT204681

closed 2020-11-03 by williambj1

williambj1, 2020-11-03:
BTW, we are working on supporting more features; stay tuned.

Theredmc, 2020-11-03:
Bruh like bruh it’s working for everyone else and it’s a handoff thing

williambj1, 2020-11-03:
You did not understand the difference between "Continuity" and "Handoff".

Theredmc, 2020-11-04:
It should work, it worked for everyone else, heck it’s even working for my brother

williambj1, 2020-11-04:
Those features are *not* supported right now. Please consider alternative solutions if you really need them.

zxystd, 2020-11-04:
AirportItlwm is not a fake or patch driver, you should read our Docs. Thanks.