OpenIntelWireless/itlwm #379: [Enhancement] Dependency fix for macOS Sierra 10.12.3

VoltaVX, 2020-11-04:
After some headaches, I finally got ITLWM to work on my macOS Sierra 10.12.3. There were some errors regarding the kext needing some libraries in iokit and kpi to be at least 16.7. However, in the info.plist, I changed the versions to 16.4 which were the version bundled in the operating system, and it worked perfectly. So yeah, just sharing my results here for the possibility of the dependency version needed for the kext to work to be lowered.

	<key>OSBundleLibraries</key>
	<dict>
		<key>com.apple.iokit.IONetworkingFamily</key>
		<string>3.2</string>
		<key>com.apple.iokit.IOPCIFamily</key>
		<string>2.9</string>
		<key>com.apple.kpi.bsd</key>
		<string>16.4</string>
		<key>com.apple.kpi.iokit</key>
		<string>16.4</string>
		<key>com.apple.kpi.libkern</key>
		<string>16.4</string>
		<key>com.apple.kpi.mach</key>
		<string>16.4</string>
	</dict>

williambj1, 2020-11-05:
@VoltaVX 

Thanks for your report, could please you attach the output from `uname -a` and `kextlibs -undef-symbols itlwm.kext` as well? Thanks.

VoltaVX, 2020-11-06:
@williambj1 Hi, unfortunately, I don't have macOS Sierra 10.12.3 since I managed to get macOS high sierra 10.13.6 working with full support. Will getting the info you'd like in the macOS Sierra installer work? From memory, I remember that when I ran the kextlibs command on the original ITLWM, I got that "com.apple.kpi.bsd, com.apple.kpi.iokit, com.apple.kpi.libkern, com.apple.kpi.mach" weren't compatible with my machine's libraries, that I needed to have at least versions 16.7.

zxystd, 2020-11-23:
@VoltaVX I think it is fix now.

closed 2020-11-24 by zxystd

zcream, 2021-06-19:
For reference, which of the versions can be used on Sierra ? There is a version for High Sierra, Monterey and Big Sur.