OpenIntelWireless/itlwm #380: AirportItlwm related boot kernel panic log

kingo132, 2020-11-05:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Maybe, it happens occasionally.

**Environment**

- Kext Version: AirportItlwm_v1.0_Beta_Catalina.kext
- WiFi Card Model: Ax200
- Product ID: 8086:2723
- macOS Version: 10.15.7

**Description**
I followed the installation guide, set SecureBootModel to Default, force loaded IO80211Family.kext. Everything seems OK. Then this panic happens. However, it just happens once until now. 
panic(cpu 4 caller 0xffffff80082469aa): Kernel trap at 0xffffff80087f91a9, type 13=general protection, registers:
CR0: 0x000000008001003b, CR2: 0xffffff81f9fd1000, CR3: 0x000000002048c000, CR4: 0x00000000003626e0
RAX: 0x787c1bf994667b9f, RBX: 0xffffff803c4349c0, RCX: 0x0000000002000000, RDX: 0x0000000003000000
RSP: 0xffffff81faa63d20, RBP: 0xffffff81faa63d40, RSI: 0xffffff803bb1e760, RDI: 0xffffff803c430000
R8:  0xffffff81faa63b04, R9:  0xffffff803c436600, R10: 0xaaaaaaaaaaaaaaab, R11: 0x0000000000000002
R12: 0xffffff803c3bea40, R13: 0xffffff803c41a800, R14: 0xffffff803bb1e760, R15: 0xffffff803c35ba40
RFL: 0x0000000000010246, RIP: 0xffffff80087f91a9, CS:  0x0000000000000008, SS:  0x0000000000000000
Fault CR2: 0xffffff81f9fd1000, Error code: 0x0000000000000000, Fault CPU: 0x4, PL: 0, VF: 0

Backtrace (CPU 4), Frame : Return Address
0xffffff8007f53220 : 0xffffff800811a65d mach_kernel : _handle_debugger_trap + 0x49d
0xffffff8007f53270 : 0xffffff8008254a75 mach_kernel : _kdp_i386_trap + 0x155
0xffffff8007f532b0 : 0xffffff80082465fe mach_kernel : _kernel_trap + 0x4ee
0xffffff8007f53300 : 0xffffff80080c0a40 mach_kernel : _return_from_trap + 0xe0
0xffffff8007f53320 : 0xffffff8008119d27 mach_kernel : _DebuggerTrapWithState + 0x17
0xffffff8007f53420 : 0xffffff800811a117 mach_kernel : _panic_trap_to_debugger + 0x227
0xffffff8007f53470 : 0xffffff80088c1a6c mach_kernel : _panic + 0x54
0xffffff8007f534e0 : 0xffffff80082469aa mach_kernel : _sync_iss_to_iks + 0x2aa
0xffffff8007f53660 : 0xffffff80082466a8 mach_kernel : _kernel_trap + 0x598
0xffffff8007f536b0 : 0xffffff80080c0a40 mach_kernel : _return_from_trap + 0xe0
0xffffff8007f536d0 : 0xffffff80087f91a9 mach_kernel : __ZNK15IORegistryEntry11getPropertyEPK8OSSymbol + 0x49
0xffffff81faa63d40 : 0xffffff80087f8bab mach_kernel : __ZN15IORegistryEntry4initEP12OSDictionary + 0x17b
0xffffff81faa63d70 : 0xffffff80087fdee2 mach_kernel : __ZN9IOService4initEP12OSDictionary + 0x12
0xffffff81faa63da0 : 0xffffff7f89b18480 com.apple.iokit.IO80211Family : __ZN28IO80211VirtualInterfaceNamer4initEv + 0x18
0xffffff81faa63dc0 : 0xffffff7f89b18801 com.apple.iokit.IO80211Family : __Z34IO80211VirtualInterfaceNamerRetainv + 0x46
0xffffff81faa63dd0 : 0xffffff7f89ade56b com.apple.iokit.IO80211Family : __ZN17IO80211Controller4initEP12OSDictionary + 0xf5
0xffffff81faa63e10 : 0xffffff7f8d012f4b com.zxystd.AirportItlwm : __ZN12AirportItlwm4initEP12OSDictionary + 0x2b
0xffffff81faa63e40 : 0xffffff8008806c41 mach_kernel : __ZN9IOService15probeCandidatesEP12OSOrderedSet + 0x661
0xffffff81faa63ef0 : 0xffffff800880644e mach_kernel : __ZN9IOService14doServiceMatchEj + 0x2de
0xffffff81faa63f50 : 0xffffff8008808c06 mach_kernel : __ZN15_IOConfigThread4mainEPvi + 0x186
0xffffff81faa63fa0 : 0xffffff80080c013e mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         com.apple.iokit.IO80211Family(1200.12.2b1)[D6EB9626-5A80-3FE4-A1F3-4F455A48C64A]@0xffffff7f89a6c000->0xffffff7f89bc7fff
            dependency: com.apple.driver.corecapture(1.0.4)[AF55F663-DDC4-39F0-970E-F45E63E2AB86]@0xffffff7f89581000
            dependency: com.apple.driver.AppleMobileFileIntegrity(1.0.5)[FF23E772-067A-3614-B922-33C82CEC0CBF]@0xffffff7f8922e000
            dependency: com.apple.kec.corecrypto(1.0)[1984287F-ABE7-3770-9FD9-61565B55E332]@0xffffff7f89146000
            dependency: com.apple.iokit.IOSkywalkFamily(1)[CEB3CDA2-1D41-3238-BFBB-7097C2636DF5]@0xffffff7f895ff000
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[26FE14A5-825D-35E4-BD06-C5B8A1AE1FD9]@0xffffff7f88eb0000
         com.zxystd.AirportItlwm(1.0)[D4493512-24B9-3B2C-89CE-98C65E75A04F]@0xffffff7f8cf88000->0xffffff7f8d9abfff
            dependency: com.apple.iokit.IO80211Family(1200.12.2b1)[D6EB9626-5A80-3FE4-A1F3-4F455A48C64A]@0xffffff7f89a6c000
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[26FE14A5-825D-35E4-BD06-C5B8A1AE1FD9]@0xffffff7f88eb0000
            dependency: com.apple.iokit.IOPCIFamily(2.9)[44472E6F-8DA0-3B46-ADEF-AFF76EC6C6DB]@0xffffff7f88b12000

BSD process name corresponding to current thread: Unknown
Boot args: -v -igfxcdc -igfxdvmt keepsyms=1 debug=0x100 darkwake=0 -noDC9 chunklist-security-epoch=0 -chunklist-no-rev2-dev

Mac OS version:
Not yet set

Kernel version:
Darwin Kernel Version 19.6.0: Mon Aug 31 22:12:52 PDT 2020; root:xnu-6153.141.2~1/RELEASE_X86_64
Kernel UUID: 05D51A3D-3A87-3FF0-98C3-9CF3827A3EDD
Kernel slide:     0x0000000007e00000
Kernel text base: 0xffffff8008000000
__HIB  text base: 0xffffff8007f00000
System model name: MacBookAir9,1 (Mac-0CFF9C7C2B63DF8D)
System shutdown begun: NO
Panic diags file unavailable, panic occurred prior to initialization

System uptime in nanoseconds: 4624912581

**Bug Report Archive**
[AirItlDebug_2020-11-05_21-41-50.zip](https://github.com/OpenIntelWireless/itlwm/files/5494760/AirItlDebug_2020-11-05_21-41-50.zip)

**Kext Download Source**
Download from https://github.com/OpenIntelWireless/itlwm/releases/tag/v1.1.0.

kingo132, 2020-11-09:
Today and yesterday, I encountered another two itlwm panic. Here is the panic log.

`
panic(cpu 3 caller 0xffffff8008991dde): "OSObject::refcount: Attempting to retain a freed object"@/AppleInternal/BuildRoot/Library/Caches/com.apple.xbs/Sources/xnu/xnu-6153.141.2.2/libkern/c++/OSObject.cpp:155
Backtrace (CPU 3), Frame : Return Address
0xffffff81faef3af0 : 0xffffff800831868d mach_kernel : _handle_debugger_trap + 0x49d
0xffffff81faef3b40 : 0xffffff8008452ab5 mach_kernel : _kdp_i386_trap + 0x155
0xffffff81faef3b80 : 0xffffff800844463e mach_kernel : _kernel_trap + 0x4ee
0xffffff81faef3bd0 : 0xffffff80082bea40 mach_kernel : _return_from_trap + 0xe0
0xffffff81faef3bf0 : 0xffffff8008317d57 mach_kernel : _DebuggerTrapWithState + 0x17
0xffffff81faef3cf0 : 0xffffff8008318147 mach_kernel : _panic_trap_to_debugger + 0x227
0xffffff81faef3d40 : 0xffffff8008abf2bc mach_kernel : _panic + 0x54
0xffffff81faef3db0 : 0xffffff8008991dde mach_kernel : __ZNK8OSObject12taggedRetainEPKv + 0x4e
0xffffff81faef3dc0 : 0xffffff7f89d187d1 com.apple.iokit.IO80211Family : __Z34IO80211VirtualInterfaceNamerRetainv + 0x16
0xffffff81faef3dd0 : 0xffffff7f89cde56b com.apple.iokit.IO80211Family : __ZN17IO80211Controller4initEP12OSDictionary + 0xf5
0xffffff81faef3e10 : 0xffffff7f8d212f4b com.zxystd.AirportItlwm : __ZN12AirportItlwm4initEP12OSDictionary + 0x2b
0xffffff81faef3e40 : 0xffffff8008a04491 mach_kernel : __ZN9IOService15probeCandidatesEP12OSOrderedSet + 0x661
0xffffff81faef3ef0 : 0xffffff8008a03c9e mach_kernel : __ZN9IOService14doServiceMatchEj + 0x2de
0xffffff81faef3f50 : 0xffffff8008a06456 mach_kernel : __ZN15_IOConfigThread4mainEPvi + 0x186
0xffffff81faef3fa0 : 0xffffff80082be13e mach_kernel : _call_continuation + 0x2e
      Kernel Extensions in backtrace:
         com.apple.iokit.IO80211Family(1200.12.2b1)[3454B540-61EE-3ECC-8F8E-D894B246CBDE]@0xffffff7f89c6c000->0xffffff7f89dc7fff
            dependency: com.apple.driver.corecapture(1.0.4)[AF55F663-DDC4-39F0-970E-F45E63E2AB86]@0xffffff7f89781000
            dependency: com.apple.driver.AppleMobileFileIntegrity(1.0.5)[4159DFFE-7746-3327-9752-C161DC295828]@0xffffff7f8942e000
            dependency: com.apple.kec.corecrypto(1.0)[1984287F-ABE7-3770-9FD9-61565B55E332]@0xffffff7f89346000
            dependency: com.apple.iokit.IOSkywalkFamily(1)[CEB3CDA2-1D41-3238-BFBB-7097C2636DF5]@0xffffff7f897ff000
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[26FE14A5-825D-35E4-BD06-C5B8A1AE1FD9]@0xffffff7f890b0000
         com.zxystd.AirportItlwm(1.0)[D4493512-24B9-3B2C-89CE-98C65E75A04F]@0xffffff7f8d188000->0xffffff7f8dbabfff
            dependency: com.apple.iokit.IO80211Family(1200.12.2b1)[3454B540-61EE-3ECC-8F8E-D894B246CBDE]@0xffffff7f89c6c000
            dependency: com.apple.iokit.IONetworkingFamily(3.4)[26FE14A5-825D-35E4-BD06-C5B8A1AE1FD9]@0xffffff7f890b0000
            dependency: com.apple.iokit.IOPCIFamily(2.9)[2F37AE58-E6B9-3B18-9092-3B80D34C334B]@0xffffff7f88d12000

BSD process name corresponding to current thread: Unknown
Boot args: -v -igfxcdc -igfxdvmt keepsyms=1 debug=0x100 darkwake=0 -noDC9 chunklist-security-epoch=0 -chunklist-no-rev2-dev chunklist-security-epoch=0 -chunklist-no-rev2-dev

Mac OS version:
Not yet set

Kernel version:
Darwin Kernel Version 19.6.0: Thu Oct 29 22:56:45 PDT 2020; root:xnu-6153.141.2.2~1/RELEASE_X86_64
Kernel UUID: 9B5A7191-5B84-3990-8710-D9BD9273A8E5
Kernel slide:     0x0000000008000000
Kernel text base: 0xffffff8008200000
__HIB  text base: 0xffffff8008100000
System model name: MacBookAir9,1 (Mac-0CFF9C7C2B63DF8D)
System shutdown begun: NO
Panic diags file unavailable, panic occurred prior to initialization

System uptime in nanoseconds: 4991966619
`

`panic(cpu 0 caller 0xffffff80044449ea): Kernel trap at 0xffffff7f89aaee94, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x0000000000000000, CR3: 0x000000001bc5c000, CR4: 0x00000000003626e0
RAX: 0x0000000000000000, RBX: 0xffffff803e6d2520, RCX: 0x0000000000000003, RDX: 0x0000000000000003
RSP: 0xffffff81f627bda0, RBP: 0xffffff81f627bdf0, RSI: 0xffffff81b6cc2060, RDI: 0xffffff81b6cc2240
R8:  0xffffff81b6cc31ef, R9:  0x0000000000000002, R10: 0xffffff8004df5280, R11: 0xffffff8004df5280
R12: 0xffffff7f89a46490, R13: 0xffffff7f8a417118, R14: 0x0000000000000000, R15: 0xffffff7f8a417118
RFL: 0x0000000000010296, RIP: 0xffffff7f89aaee94, CS:  0x0000000000000008, SS:  0x0000000000000010
Fault CR2: 0x0000000000000000, Error code: 0x0000000000000000, Fault CPU: 0x0, PL: 0, VF: 1

Backtrace (CPU 0), Frame : Return Address
0xffffff81f627b700 : 0xffffff800431868d mach_kernel : _handle_debugger_trap + 0x49d
0xffffff81f627b750 : 0xffffff8004452ab5 mach_kernel : _kdp_i386_trap + 0x155
0xffffff81f627b790 : 0xffffff800444463e mach_kernel : _kernel_trap + 0x4ee
0xffffff81f627b7e0 : 0xffffff7f88e536f4 as.vit9696.VirtualSMC : __ZN18VirtualSMCProvider10kernelTrapI22x86_saved_state_1010_tEEvPT_Pm + 0x454
0xffffff81f627b860 : 0xffffff80042bea40 mach_kernel : _return_from_trap + 0xe0
0xffffff81f627b880 : 0xffffff8004317d57 mach_kernel : _DebuggerTrapWithState + 0x17
0xffffff81f627b980 : 0xffffff8004318147 mach_kernel : _panic_trap_to_debugger + 0x227
0xffffff81f627b9d0 : 0xffffff8004abf2bc mach_kernel : _panic + 0x54
0xffffff81f627ba40 : 0xffffff80044449ea mach_kernel : _sync_iss_to_iks + 0x2aa
[... 10 lines cut ...]
            dependency: com.apple.iokit.IOACPIFamily(1.4)[2956198D-24F2-3790-A9B2-1EAB9434B906]@0xffffff7f84d09000
[... cut ...]
Boot args: -v -igfxcdc -igfxdvmt keepsyms=1 debug=0x100 darkwake=0 -noDC9 chunklist-security-epoch=0 -chunklist-no-rev2-dev

Mac OS version:
19H15

Kernel version:
Darwin Kernel Version 19.6.0: Thu Oct 29 22:56:45 PDT 2020; root:xnu-6153.141.2.2~1/RELEASE_X86_64
Kernel UUID: 9B5A7191-5B84-3990-8710-D9BD9273A8E5
Kernel slide:     0x0000000004000000
Kernel text base: 0xffffff8004200000
__HIB  text base: 0xffffff8004100000
System model name: MacBookAir9,1 (Mac-0CFF9C7C2B63DF8D)
System shutdown begun: NO
Panic diags file available: YES (0x0)

System uptime in nanoseconds: 144435382670

`

zxystd, 2020-11-09:
The first panic is the same as you mention in this issue, maybe it is not a itlwm bug but related to Apple's Airport, I don't know what is going on, the Apple Airport framework crash on init state.
how do you trigger the second panic?

kingo132, 2020-11-09:
> The first panic is the same as you mention in this issue, maybe it is not a itlwm bug but related to Apple's Airport, I don't know what is going on, the Apple Airport framework crash on init state.
> how do you trigger the second panic?

The second panic happens when I try to load AirportItlwm.kext manually after login. Because AirportItlwm.kext doesn't load firmware at startup. So I disabled it in Opencore and try to load it manually.
Here is my load scripts:
```
sudo kextunload -b com.zxystd.AirportItlwm
sudo chown -R root:wheel /Users/jason/soft/AirportItlwm.kext
sudo chmod -R 755 /Users/jason/soft/AirportItlwm.kext
sudo kextutil -v 6 /Users/jason/soft/AirportItlwm.kext
```

zxystd, 2020-11-09:
@kingo132 different with itlwm, AirportItlwm can not be unload, that's why you caused panic. Don't try to do that.

kingo132, 2020-11-09:
> @kingo132 different with itlwm, AirportItlwm can not be unload, that's why you trigger panic. Don't try to do that.

I see. Thank you.

zxystd, 2020-11-23:
[AirportItlwm_Catalina.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/5582055/AirportItlwm_Catalina.kext.zip)
@kingo132 Can you help to check if this is fixed the panic? Or I will try another way.

VergeDX, 2021-01-14:
> [AirportItlwm.kext.zip](https://github.com/OpenIntelWireless/itlwm/files/5582070/AirportItlwm.kext.zip)
> updated.
> 
> @kingo132 Can you help to check if this is fixed the panic? Or I will try another way.

@zxystd 
I meet this issues today, and this kext is good for me. But seems this kext set my maximum brightness level, so my screen looks so dark now.
If any log needed, please let me know : )

VergeDX, 2021-01-15:
@zxystd 
And today, I meet this issues again. It back to normal after clean nvram.

zxystd, 2021-01-15:
> @zxystd
> And today, I meet this issues again. It back to normal after clean nvram.

which version of AirportItlwm you are using? what's the kernel panic log? can you upload it or take a picture when it happened?

VergeDX, 2021-01-15:
> > @zxystd
> > And today, I meet this issues again. It back to normal after clean nvram.
> 
> which version of AirportItlwm you are using? what's the kernel panic log? can you upload it or take a picture when it happened?

I used the version of your comment, it just appears once so I have no log about this...
I noticed 1.2.0 stable version released, I will try it and give feedback : )

zxystd, 2021-01-15:
@VergeDX Well, wait for your feedback.

VergeDX, 2021-01-15:
> @VergeDX Well, wait for your feedback.

@zxystd 
I test on my Catalina 10.15.7 with iPad Air 2 (iPad OS 14.2), All permission function works fine! (Hand off, Universal clipboard, wifi connect). And I'm not meet the panic issue.
Thanks for the perfect work!

williambj1, 2021-01-16:
Should be fixed in the latest version.

closed 2021-01-16 by williambj1