OpenIntelWireless/itlwm #381: Bluetooth does not work with latest AirportItlwm commit installed.

lgariv, 2020-11-06:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**
- Kext Version: [`07cb669`](https://github.com/OpenIntelWireless/itlwm/commit/07cb669b5ecd6e069c9cff15fa76a1a80ab65563), AirportItlwm
- WiFi Card Model: AC-9462
- Product ID: `0x42A4`
- macOS Version: macOS Catalina 10.15.7 (before supplemental update)

**Description**
This commit broke Bluetooth completely - firmware is not uploading. reverting back to the previous commit resolves the issue for me.

**Bug Report Archive**
[AirItlDebug_2020-11-06_17-21-41.zip](https://github.com/OpenIntelWireless/itlwm/files/5501978/AirItlDebug_2020-11-06_17-21-41.zip)

**Kext Download Source**
Releases, latest compiled pre-release.

win1010525, 2020-11-07:
Works fine for me. Same card

lgariv, 2020-11-07:
> Works fine for me. Same card

Same product ID? Which boot loader do you use?

zxystd, 2020-11-08:
@lgariv I don't think they have relationship with the commit... have you cleaned the recycle bin after you update the kext and before reboot?

lgariv, 2020-11-12:
> @lgariv I don't think they have relationship with the commit... have you cleaned the recycle bin after you update the kext and before reboot?

I always clear it as soon as I'm done doing something. 
Anyway, just updated to the latest alpha and now everything works.

lgariv, 2020-11-15:
Apparently, I am not the only one [having this problem](https://github.com/OpenIntelWireless/IntelBluetoothFirmware/issues/212
). Also, on the [latest commit](https://github.com/OpenIntelWireless/itlwm/commit/219b6178e40b45565d67fe1cc8e72fc16e783a11)

dpatrongomez, 2020-11-16:
I have the same problem with Intel Wireless 8265

luuknieuwdorp, 2020-11-17:
Same issue here with Intel Wireless 8265. After upgrading I can't connect to my remembered bluetooth device. After forgetting it to try and repair it no devices seem to show up in the bluetooth list at all when using the latest alpha.

luuknieuwdorp, 2020-11-23:
Update: bluetooth is working for me again since updating to the latest alpha, built 7 hours ago (https://github.com/OpenIntelWireless/itlwm/commit/d9bd8d435b80c009c928156ef805e14b2a527fd7).

dpatrongomez, 2020-11-23:
> Update: bluetooth is working for me again since updating to the latest alpha, built 7 hours ago ([d9bd8d4](https://github.com/OpenIntelWireless/itlwm/commit/d9bd8d435b80c009c928156ef805e14b2a527fd7)).

Well, even with that version it lets me turn off the bluetooth

MenschLennart, 2020-11-26:
I do have the AC - 9260 and Bluetooth doesn't work at all. Already tried resetting Bluetooth completely with SHIFT-Option-LeftClick on Bluetooth icon in menu bar. But nothing helps. Wether Magic Mouse 2 nor Magic Keyboard is working.

closed 2020-12-25 by williambj1