OpenIntelWireless/itlwm #386: itlwm loaded with error: unknown adapter type

blackberry4311, 2020-11-08:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: 1.1.0, itlwm
- WiFi Card Model: Intel Wifi AX201
- Product ID: 0x06F0
- macOS Version: 10.15.7

**Description**
itlwm loaded with error: unknown adapter type,
tried auto load via EFI inject with OC and manual via terminal follow instruction at https://openintelwireless.github.io/itlwm/Troubleshooting.html#getting-kext-logs-for-loading-failures

**Bug Report Archive**
[bugreport_24931.zip](https://github.com/OpenIntelWireless/itlwm/files/5507098/bugreport_24931.zip)

**Kext Download Source**
stable release of 1.1.0

williambj1, 2020-11-08:
Does manually loading the kext after booting into the system give you the same result?

zxystd, 2020-11-09:
@blackberry4311 can you upload your pcie info with Hackintool? like this:
[image]

blackberry4311, 2020-11-09:
@williambj1 I  got above log when manual load the kext, and it gives me the same result
@zxystd here you are
[image]

zxystd, 2020-11-09:
same card and same driver? now it shows unknown adapter type? @blackberry4311

blackberry4311, 2020-11-09:
yes, it worked 2 days ago, I change nothing and it show the log like I uploaded yesterday.

zxystd, 2020-11-09:
subdevice id is 0x0000, so weird, that should not work. do you have linux? can you check if the card can work on linux now?
I will compile a kext which will output more logs what I needed to you soon.
[image]

blackberry4311, 2020-11-09:
I don't have linux currently but I can install it for further checking.Is Ubuntu latest version ok?

williambj1, 2020-11-09:
@blackberry4311 A Ubuntu 20 Live USB would be enough.

blackberry4311, 2020-11-09:
work well on ubuntu 20
[image]
[image]

blackberry4311, 2020-11-09:
Wifi back and working like charm, after I'm boot to Ubuntu,
[image]
Sub Device change like in this image

Hmm, I don't know the reason why but let me try to make some test on this scenario first.

closed 2020-11-11 by blackberry4311