OpenIntelWireless/itlwm #388: itlwm无法分配ip，也无法忘记网络。

HKMV, 2020-11-12:
**Have You Read Our Docs**
Yes

**Are You Reporting A Bug**
Yes

**Environment**

- Kext Version: itlwm-v1.2.0-DEBUG-alpha-65dc639.zip
- WiFi Card Model: AC 8265
- Product ID: 0x24fd
- macOS Version: 10.15.7

**Description**
之前我是通过写入修改源码，把ssid和密码替换成我的Serenity进行连接的，
后来支持了Info.plist中配值，都是修改后编译或者编译后修改，然后注入到 四叶草 中使用。
再后来有了HeliPort，github上itlwm也定期自动编译，我也就没在自己编译过了；
但是不知道为何我最初使用的这个ssid连接上后一直无法进行dhcp分配ip，然后过将近一分钟就会断开自动重连；
修改ssid名称就可以连接上，也能正常上网。

我想的是忘记这条记录重新进行 连接，但是现在正在使用的Info.plist我没进行修改是默认的，不知道为啥开机他还会自动连接Serenity，从HeliPort的已保存网络列表中也没找到，通过文档中的 [重置系统首选项 - 网络](https://docs.oiw.workers.dev/itlwm/Troubleshooting.html#resetting-system-preferences-network)，也无法忘记这个网络。

[image]

日志晚上回去补上。

zxystd, 2020-11-12:
进钥匙串删除密码记录。没有ip是因为密码错误。

closed 2020-11-12 by zxystd

HKMV, 2020-11-12:
> 进钥匙串删除密码记录。没有ip是因为密码错误。

感谢大佬的付出，我回去试下。

HKMV, 2020-11-12:
> 进钥匙串删除密码记录。没有ip是因为密码错误。

已解决，感谢师傅的付出。